openwrt/luci #5848: [bug report] openwrt luci web should support ipv6 address for firewall port forwarding

xiedeacc, 2022-06-22:
I turned on ipv6 nat, and want exporsure a port to public, but now web can only input a ipv4 address, so I have to modify /etc/config/firewall directly, luci should support select and input ipv6 address
https://user-images.githubusercontent.com/8072296/174958070-699dbc50-e22a-4fb4-b87d-a9b19665bf0a.png
https://user-images.githubusercontent.com/8072296/174958262-7cb07d62-f1e6-4adc-9315-347e707591b6.png

xiedeacc, 2022-06-24:
openwrt version: 22.03.rc4

echo094, 2023-03-02:
> What is your use case? NAT is not needed with IPv6 in general, see https://en.wikipedia.org/wiki/Network_address_translation#NAT_in_IPv6 Instead one should create a Firewall Traffic Rule, e.g: [image]

In my school, a device can only be allocated a single IPv6 address dynamically, and each device has to complete a web authentication before accessing the internet. Besides, there is a limit to the number of devices associated with an account. However, my NAS cannot complete the authentication. Thus, I have to add a router keep a stable public ipv6 address and use [NAT66](https://openwrt.org/docs/guide-user/network/ipv6/ipv6.nat6) to make my device reachable via ipv6. In this case, a DNAT rule is required to expose a port of a sub device instead of a traffic rule. 

This feature request has been posted several years ago in issue #925, when masq6 has not been supported by default. In Openwrt 22.03, ipv6 masquerading and redirect have already been supported by nftables and can be enabled by editing the config file. However, we still cannot set the redirect rules via the luci. Besides, the existing rules for ipv6 cannot be displayed correctly, as shown in the top floor. 

A general redirect rule for ipv6 is listed in this issue: openwrt/openwrt#9624. According to the sample, two modifications should be made:
* A `family` option is required to choose between ipv4 and ipv6. 
* The address checking should be modified to support ipv6 address.

systemcrash, 2024-05-01:
Supported since 9c55500fe8efa309d55f34c21d5ae2bf69fabf06

closed 2024-05-01 by systemcrash