openwrt/luci #5850: luci-mod-network: device page - bonding glitches

plappermaul, 2022-06-23:
The network device page has two glitches if we use bonding interfaces.

1. A device bonding_masters is shown. Nevertheless the node /sys/class/net/bonding_masters is no device at all
2. The bond devices are shown as normal interfaces. They should be given some meaningful type like VLAN interfaces

Steps to reproduce

- Load Bonding driver: insmod bonding
- Create new bonding interface: echo +bond16 > /sys/class/net/bonding_masters
- Look into luci network device page

Environment

cat /etc/openwrt_release
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='SNAPSHOT'
DISTRIB_REVISION='r19862-1d42af720c'
DISTRIB_TARGET='realtek/rtl838x'
DISTRIB_ARCH='mips_4kec'
DISTRIB_DESCRIPTION='OpenWrt SNAPSHOT r19862-1d42af720c'
DISTRIB_TAINTS=''

plappermaul, 2022-06-23:
Looking around we might need this for bonding_masters in luci/libs/rpcd-mod-luci/src/luci.c

```
rpc_luci_get_network_devices()
{
  ...
  if (strcmp(e->d_name, ".") && strcmp(e->d_name, "..") && strcmp(e->d_name, "bonding_masters"))
    rpc_luci_parse_network_device_sys(e->d_name, ifaddr);
  ...
}
```

plappermaul, 2022-06-23:
For showing up bonding devices the proper way fix could be in luci/modules/luci-mod-network/htdocs/luci-static/resources/view/network/interfaces.js:

```
function getDevTypeDesc(section_id)
{
  ...
  case 'bond':
    return _('LAG device');
  ...
}
```

jow-, 2022-06-23:
Please provide a diff with the suggested changes.

plappermaul, 2022-06-24:
I started taking care about the bonding_masters listing. Build environment is set up. Modification and compilation finished successfully. Afterwars replaced /usr/lib/rpcd/luci.so on the router and restarted rpcd and router. Nevertheless bonding_masters interface still shows up. Am I on the right track?

```
ubus -v call luci-rpc getNetworkDevices
...
	"bonding_masters": {
		"name": "bonding_masters",
...

```

plappermaul, 2022-06-26:
memo to myself: NEVER keep old library below /usr/lib/rpcd

plappermaul, 2022-06-26:
Fix 1:https://github.com/openwrt/luci/pull/5855

closed 2024-12-19 by systemcrash