openwrt/luci #5854: luci-app-statistics: use SVG for graphs

CAFxX, 2022-06-25:
note: this is a cosmetic enhancement request

## Steps to reproduce:

1. open the graphs page on a high-DPI display

## Actual behavior:

1. notice that the graphs are not displayed at native resolution and are lacking subpixel AA, as they are raster images

[image]

## Expected behavior:

Graphs are displayed at native screen resolution, and with text rendered using the appropriate antialiasing.

[rrdtools supports exporting graphs as SVG](https://oss.oetiker.ch/rrdtool/doc/rrdgraph.en.html#Miscellaneous) (in addition to PNG), and [every supported browser supports displaying SVG images](https://caniuse.com/svg). While there is a point to be made that simply switching from PNG to SVG would likely work for virtually everyone, it may be safer to maintain a PNG fallback path for ancient browsers (or a config option to choose between PNG and SVG). 

## Additional Information:
OpenWrt version information from system `/etc/openwrt_release`

```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='21.02.3'
DISTRIB_REVISION='r16554-1d4dea6d4f'
DISTRIB_TARGET='ath79/generic'
DISTRIB_ARCH='mips_24kc'
DISTRIB_DESCRIPTION='OpenWrt 21.02.3 r16554-1d4dea6d4f'
DISTRIB_TAINTS=''
```

jow-, 2022-06-28:
Not supportable

closed 2022-06-28 by jow-

CAFxX, 2022-06-28:
Just a small note in case someone stumbles upon this: the current version of rrdtool does support SVG, but it would require [cairo](https://www.cairographics.org/) to be available at build-time and run-time.