openwrt/luci #5860: luci-i18n-*-en: Several Portuguese translations in English language files

raineth, 2022-06-29:
A weblate user contributed several Portuguese translations to the English localizations that were merged in commit 62ee3fbfbf91e41d1260ea7023145fbdd4eb0d86.  It appears that all submissions from viniribeirossa in that commit should be rolled back. Those changes should probably be fixed in Weblate as well -- I don't know whether they have any admin tools to help mass-correct actions from confused users.

feckert, 2022-06-29:
I've noticed that too!
@hnyman I think we have to fix that in the weblate, otherwise there will be merge conflicts.

castillofrancodamian, 2022-06-29:
The English "translation" should be removed.

hnyman, 2022-06-29:
@jow-  already pondered about removing the English translation from the built packages. It serves no function as the untranslated strings are already in English.

Any language (including English) can't be easily disabled in weblate, as it allows casual users to initiate new translation languages. 

Ps. I can't easily do fixes myself in the next few weeks as I am travelling.

jellybean13, 2022-07-02:
Some "translations" are still wrong despite OpenWrt 22.03-SNAPSHOT r19507-158a5af801 installed.

Language: English
[image]

hnyman, 2022-07-03:
I think that I managed to remove the erroneous Portuguese strings from English .po files.
(The translator had apparently tried to translate luci-app-attended-sysupgrade, and weblate had then propagated the same strings into other packages.)

I backported the fixes from master to 22.03 and 21.02.

closed 2022-07-03 by hnyman