openwrt/luci #5875: LuCi cannot delete Devices

QA-Manager, 2022-07-10:
## Steps to reproduce:

1. go to: Network → Interfaces → Devices
2. I have somehow created 4 instances of br-lan.13 (I was attempting to set up a VLAN); all four instances appear on the Devices screen

## Actual behavior:
 - There isn't any way to delete a Device

## Expected behavior:
 - There should be a way to delete Devices (and Interfaces), if for no other reason than you may create one and decide that you don't want it.

## Additional Information:
OpenWrt version information from system `/etc/openwrt_release`

```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='21.02.3'
DISTRIB_REVISION='r16554-1d4dea6d4f'
DISTRIB_TARGET='mvebu/cortexa9'
DISTRIB_ARCH='arm_cortex-a9_vfpv3-d16'
DISTRIB_DESCRIPTION='OpenWrt 21.02.3 r16554-1d4dea6d4f'
DISTRIB_TAINTS=''
```

jow-, 2022-07-11:
You can't delete devices, only unconfigure them. Netifd will eventually garbage collect ephemeral devices.

closed 2022-07-11 by jow-

QA-Manager, 2022-07-11:
If I create a VLAN 13 and a br-lan.13, and later reconfigure my network and decide that I don't need VLAN 13, LuCi shouldn't force me to leave it in the system as though it is still being used.

Arndorferd, 2024-03-15:
This makes no sense, please reopen, it shouldnt be forced to stay in the system

QA-Manager, 2024-03-16:
@Arndorferd, are you asking me to write up this defect again, or are you asking @jow- to reopen it?

jow-, 2024-03-16:
It is the way it works, you unconfigure the device and it gets garbage collected by the system once it becomes unused. You can reboot or restart the network service to force immediate removal of the kernel device.