openwrt/luci #5876: LUCI - Won't save changes & won't explain why

QA-Manager, 2022-07-10:
## Steps to reproduce:

1) Select Network/Interfaces/Devices
2) Click 'Configure' on the 'br-lan' line
3) Click the 'Bridge VLAN Filtering' tab
4) Click 'Enable VLAN Filtering'
5) Click 'Add'
6) Click 'Save'

## Actual behavior:
  Nothing happens. It doesn't save, and it doesn't tell you why it won't save; your only option is 'Dismiss'.

## Expected behavior:
We should prevent the user from making this mistake (?Remove the 'Add' button?). If we can't prevent it, then clicking 'Save' should produce a message explaining the user's mistake, and how to correct it. DSA is not yet well documented; DSA with VLANs is even less well documented, so we should make the experience as friendly as possible.

## Additional Information:
OpenWrt version information from system `/etc/openwrt_release`

```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='21.02.3'
DISTRIB_REVISION='r16554-1d4dea6d4f'
DISTRIB_TARGET='mvebu/cortexa9'
DISTRIB_ARCH='arm_cortex-a9_vfpv3-d16'
DISTRIB_DESCRIPTION='OpenWrt 21.02.3 r16554-1d4dea6d4f'
DISTRIB_TAINTS=''
```

jow-, 2022-07-11:
Works fine here. Please open your browser debug console (F12) and look for errors being reported.

QA-Manager, 2022-07-11:
I can reproduce it every time. Note that I am starting with no VLANs defined and VLAN filtering turned off.

The console shows this:
This page uses the non standard property “zoom”. Consider using calc() in the relevant property values, or using “transform” along with “transform-origin: 0 0”. network
RPCError: RPC call to uci/set failed with ubus code 4: Resource not found
  at handleCallReply (https://192.168.1.1/luci-static/resources/rpc.js?v=git-22.083.69138-0a0ce2a:15:3)
    raise https://192.168.1.1/luci-static/resources/luci.js?v=git-22.083.69138-0a0ce2a:155
    handleCallReply https://192.168.1.1/luci-static/resources/rpc.js?v=git-22.083.69138-0a0ce2a:15
luci.js:161:9
RPCError: RPC call to uci/set failed with ubus code 4: Resource not found
  at handleCallReply (https://192.168.1.1/luci-static/resources/rpc.js?v=git-22.083.69138-0a0ce2a:15:3)
    raise https://192.168.1.1/luci-static/resources/luci.js?v=git-22.083.69138-0a0ce2a:155
    handleCallReply https://192.168.1.1/luci-static/resources/rpc.js?v=git-22.083.69138-0a0ce2a:15
luci.js:161:9
RPCError: RPC call to uci/set failed with ubus code 4: Resource not found
  at handleCallReply (https://192.168.1.1/luci-static/resources/rpc.js?v=git-22.083.69138-0a0ce2a:15:3)
  at promise callback*parseCallReply (https://192.168.1.1/luci-static/resources/rpc.js?v=git-22.083.69138-0a0ce2a:9:70)
  at promise callback*call (https://192.168.1.1/luci-static/resources/rpc.js?v=git-22.083.69138-0a0ce2a:5:110)
  at declare/</< (https://192.168.1.1/luci-static/resources/rpc.js?v=git-22.083.69138-0a0ce2a:23:320)
  at declare/< (https://192.168.1.1/luci-static/resources/rpc.js?v=git-22.083.69138-0a0ce2a:20:436)
  at save (https://192.168.1.1/luci-static/resources/uci.js?v=git-22.083.69138-0a0ce2a:89:17)
  at promise callback*save (https://192.168.1.1/luci-static/resources/form.js?v=git-22.083.69138-0a0ce2a:47:140)
  at handleModalSave (https://192.168.1.1/luci-static/resources/form.js?v=git-22.083.69138-0a0ce2a:206:176)
  at super (https://192.168.1.1/luci-static/resources/luci.js?v=git-22.083.69138-0a0ce2a:22:42)
  at handleModalSave (https://192.168.1.1/luci-static/resources/form.js?v=git-22.083.69138-0a0ce2a:212:886)
  at createHandlerFn/< (https://192.168.1.1/luci-static/resources/ui.js?v=git-22.083.69138-0a0ce2a:306:29)
    raise https://192.168.1.1/luci-static/resources/luci.js?v=git-22.083.69138-0a0ce2a:155
    handleCallReply https://192.168.1.1/luci-static/resources/rpc.js?v=git-22.083.69138-0a0ce2a:15
    promise callback*parseCallReply https://192.168.1.1/luci-static/resources/rpc.js?v=git-22.083.69138-0a0ce2a:9
    promise callback*call https://192.168.1.1/luci-static/resources/rpc.js?v=git-22.083.69138-0a0ce2a:5
    declare https://192.168.1.1/luci-static/resources/rpc.js?v=git-22.083.69138-0a0ce2a:23
    declare https://192.168.1.1/luci-static/resources/rpc.js?v=git-22.083.69138-0a0ce2a:20
    save https://192.168.1.1/luci-static/resources/uci.js?v=git-22.083.69138-0a0ce2a:89
    promise callback*save https://192.168.1.1/luci-static/resources/form.js?v=git-22.083.69138-0a0ce2a:47
    handleModalSave https://192.168.1.1/luci-static/resources/form.js?v=git-22.083.69138-0a0ce2a:206
    super https://192.168.1.1/luci-static/resources/luci.js?v=git-22.083.69138-0a0ce2a:22
    handleModalSave https://192.168.1.1/luci-static/resources/form.js?v=git-22.083.69138-0a0ce2a:212
    createHandlerFn https://192.168.1.1/luci-static/resources/ui.js?v=git-22.083.69138-0a0ce2a:306
luci.js:161:9
RPCError: RPC call to uci/set failed with ubus code 4: Resource not found
  at handleCallReply (https://192.168.1.1/luci-static/resources/rpc.js?v=git-22.083.69138-0a0ce2a:15:3)
  at promise callback*parseCallReply (https://192.168.1.1/luci-static/resources/rpc.js?v=git-22.083.69138-0a0ce2a:9:70)
  at promise callback*call (https://192.168.1.1/luci-static/resources/rpc.js?v=git-22.083.69138-0a0ce2a:5:110)
  at declare/</< (https://192.168.1.1/luci-static/resources/rpc.js?v=git-22.083.69138-0a0ce2a:23:320)
  at declare/< (https://192.168.1.1/luci-static/resources/rpc.js?v=git-22.083.69138-0a0ce2a:20:436)
  at save (https://192.168.1.1/luci-static/resources/uci.js?v=git-22.083.69138-0a0ce2a:89:17)
  at promise callback*save (https://192.168.1.1/luci-static/resources/form.js?v=git-22.083.69138-0a0ce2a:47:140)
  at handleModalSave (https://192.168.1.1/luci-static/resources/form.js?v=git-22.083.69138-0a0ce2a:206:176)
  at super (https://192.168.1.1/luci-static/resources/luci.js?v=git-22.083.69138-0a0ce2a:22:42)
  at handleModalSave (https://192.168.1.1/luci-static/resources/form.js?v=git-22.083.69138-0a0ce2a:212:886)
  at createHandlerFn/< (https://192.168.1.1/luci-static/resources/ui.js?v=git-22.083.69138-0a0ce2a:306:29)
    raise https://192.168.1.1/luci-static/resources/luci.js?v=git-22.083.69138-0a0ce2a:155
    handleCallReply https://192.168.1.1/luci-static/resources/rpc.js?v=git-22.083.69138-0a0ce2a:15
    promise callback*parseCallReply https://192.168.1.1/luci-static/resources/rpc.js?v=git-22.083.69138-0a0ce2a:9
    promise callback*call https://192.168.1.1/luci-static/resources/rpc.js?v=git-22.083.69138-0a0ce2a:5
    declare https://192.168.1.1/luci-static/resources/rpc.js?v=git-22.083.69138-0a0ce2a:23
    declare https://192.168.1.1/luci-static/resources/rpc.js?v=git-22.083.69138-0a0ce2a:20
    save https://192.168.1.1/luci-static/resources/uci.js?v=git-22.083.69138-0a0ce2a:89
    promise callback*save https://192.168.1.1/luci-static/resources/form.js?v=git-22.083.69138-0a0ce2a:47
    handleModalSave https://192.168.1.1/luci-static/resources/form.js?v=git-22.083.69138-0a0ce2a:206
    super https://192.168.1.1/luci-static/resources/luci.js?v=git-22.083.69138-0a0ce2a:22
    handleModalSave https://192.168.1.1/luci-static/resources/form.js?v=git-22.083.69138-0a0ce2a:212
    createHandlerFn https://192.168.1.1/luci-static/resources/ui.js?v=git-22.083.69138-0a0ce2a:306
luci.js:161:9

commit 25dcd0b7 (2022-07-22): luci-base: uci.js: prevent sending empty uci set operations

Under certain circumstances, a staged uci option value might get unset
again before saving the config, leaving an empty section change set
behind, causing the save call to send an empty uci set request via rpc,
triggering an ubus code 4 (Resource not found) error.

In particular this prevented bridge VLANs from getting saved properly.

Fixes: #5876
commit df15ea39 (2022-07-22): luci-base: uci.js: prevent sending empty uci set operations

Under certain circumstances, a staged uci option value might get unset
again before saving the config, leaving an empty section change set
behind, causing the save call to send an empty uci set request via rpc,
triggering an ubus code 4 (Resource not found) error.

In particular this prevented bridge VLANs from getting saved properly.

Fixes: #5876

(cherry picked from commit 25dcd0b77c4450b83cd8b8307397e58c34ebfcd1)
closed 2022-07-22 by jow-