openwrt/luci #5877: Luci-proto-ncm

breenstorm, 2022-07-11:
## Steps to reproduce:
**Example: Replace the following lines and remove this**

1. go to: Network → Interfaces → WWAN0 (the interface used for NCM)
2. Click edit

## Actual behavior:

1. Service type is displayed with pulldown showing lte/preferlte/umts/etc
2. This gets added as 'option service '<servicetype>' to /etc/config/network

## Expected behavior:

1. Network mode is displayed with pulldown showing lte/preferlte/umts/etc
2. This gets added as 'option mode '<servicetype>' to /etc/config/network

## Additional Information:
Package luci-proto-ncm depends on gcom-ncm. Gcom-ncm uses the option called 'mode' to set the radiomode of the modem. There is no option 'service' in the gcom-ncm scripts.

breenstorm, 2022-07-11:
Pull request #5879

jow-, 2022-07-23:
Fixed by 4d6642b636a9feaa173c568c99f4129c440ed8b8

closed 2022-07-23 by jow-