openwrt/luci #5886: LUCI_LANG_ur： Warning notices

zxlhhyccc, 2022-07-16:
[image]
[image]

ptpt52, 2022-07-17:
same issue here:
```
tmp/.config-package.in:59941:warning: leading whitespace ignored
```

feckert, 2022-07-21:
So I don't get this warning!
I have seen that there are more and more translations that are not listed in the luci.mk.
Maybe one of you @ptpt52 @zxlhhyccc can sync this and add the missing information to luci.mk.
So far, these translations are not built because they do not have an entry in luci.mk.

hnyman, 2022-07-21:
> So far, these translations are not built because they do not have an entry in luci.mk.

Actually, they are being built.
E.g. the ur / urdu has zero strings translated, but the base.po has been created in weblate.
No translation files for other apps have been started.

The base.po download exists quite normally, e.g.
```
luci-i18n-base-ur_git-22.200.34529-5d7af48_all.ipk | 1.1 KB | Thu Jul 21 12:11:2022
https://downloads.openwrt.org/snapshots/packages/arm_cortex-a15_neon-vfpv4/luci/luci-i18n-base-ur_git-22.200.34529-5d7af48_all.ipk
```

The warning is cosmetic and due to the missing language name, which leads into leading space from the parsing in
https://github.com/openwrt/luci/blob/master/luci.mk#L288

ptpt52, 2022-07-22:
how to fix it?

commit 5650e163 (2022-07-22): luci.mk: ignore language directories not explicitly mentioned

Only create translation packages for languages mentioned in luci.mk

Fixes: #5886
commit cd970917 (2022-07-22): luci.mk: ignore language directories not explicitly mentioned

Only create translation packages for languages mentioned in luci.mk

Fixes: #5886
closed 2022-07-22 by jow-