openwrt/luci #5892: luci-app-mwan3 - Adding rule in 22.03.0-rc5 gets deleted again

unregist, 2022-07-21:
## Steps to reproduce:
go to: Network → Multiwan → rule tab
Enter a rule name at the bottom and click "add"
Click save in the rule creation window (content of the rule doesn't matter)
Note the "unsaved changes" now will report the correct UCI commands
Click save
The "unsaved changes" now adds an "uci del" command undoing the rule
Using "Save & Apply" without doing save first also results in the same issue.

[image]

## Actual behaviour:
Rule does not get created, or rather, gets created and deleted in the same go

## Expected behaviour:
Rule to be created and not deleted

## Workaround:
Execute the UCI commands listed in the "unsaved changes" page from the CLI instead (without the "uci del" one obviously)
follow it with "uci commit"

## Additional Information:
Device/Config was upgraded from 21.02

```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='22.03.0-rc5'
DISTRIB_REVISION='r19523-bfd070e7fa'
DISTRIB_TARGET='mvebu/cortexa9'
DISTRIB_ARCH='arm_cortex-a9_vfpv3-d16'
DISTRIB_DESCRIPTION='OpenWrt 22.03.0-rc5 r19523-bfd070e7fa'
DISTRIB_TAINTS=''
```
Imagebuilder image from following command:
```
make image PROFILE=linksys_wrt1200ac PACKAGES="luci luci-ssl luci-app-advanced-reboot luci-app-uhttpd luci-app-sqm luci-app-openvpn openvpn-openssl openssl-util curl htop tcpdump screen block-mount e2fsprogs kmod-usb-storage kmod-usb2 kmod-usb3 kmod-fs-ext4 fdisk zabbix-agentd zabbix-extra-mac80211 zabbix-extra-network zabbix-extra-wifi wireguard-tools luci-app-wireguard bind-dig luci-app-mwan3 mwan3 luci-app-https-dns-proxy https-dns-proxy avahi-daemon" 
```

feckert, 2022-07-21:
That is the same problem https://github.com/openwrt/luci/issues/5760

jow-, 2022-07-21:
Duplicate of #5760

closed 2022-07-21 by jow-

unregist, 2022-07-21:
I did see that one, but didn't recognise it as the same issue. 
Apologies everyone :)