openwrt/luci #5899: luci-mod-network: multi-port devices (>9) incorrectly sorted

oliv3r, 2022-07-27:
Any page featuring device ports will mess up the ordering (due to using digital sorting instead of natural sorting, e.g. 0, 1, 10, 11, 2, 3)

## Steps to reproduce:
Running luci master on a device with more then 9 ports

1. go to: Network → Interfaces → Devices

## Actual behavior:
[image]

also on bridge ports:
[image]

and most importantly also on the vlan page!
[image]

However, dismissing the dialog, and re-opening it, will sort it properly! (this does not apply to the other cases listed above.
[image]

## Expected behavior:
As a user, having the ports ordered logically, is far more user-friendly, and makes it cognitively easier to follow.

## Additional Information:
OpenWrt version information from system `/etc/openwrt_release`

```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='SNAPSHOT'
DISTRIB_REVISION='r20181+4-a436923133'
DISTRIB_TARGET='realtek/rtl930x'
DISTRIB_ARCH='mips_24kc'
DISTRIB_DESCRIPTION='OpenWrt SNAPSHOT r20181+4-a436923133'
DISTRIB_TAINTS='no-all busybox'
```

jow-, 2022-07-27:
Should be fixed with 8199b2ce9a, 37422e891a, 801d4513b2

closed 2022-07-27 by jow-

oliv3r, 2022-07-29:
Confirmed closed: while you already merged; but you can have my tested by :)