openwrt/luci #5900: luci-mod-network: New vlan filtering options are 'confusing'

oliv3r, 2022-07-27:
## Steps to reproduce:
1. go to: Network → Interfaces → Devices (switch interface) → Bridge vlan filtering tab → IPv6 RA settings tab → filter drop down box

## Actual behavior:
[image]
The listed options, 'Do not participate', 'Egress unagged', 'Egress tagged', 'primary VLAN ID'; while surly logical (and better, offering more flexibility) are not 'common'. E.g. compared to a 'off the shell HP switch uses the 'Tag, Untag, Exclude' or D-Link which uses tagged, untagged and 'not a member' Previous versions of OpenWRT had, Off, Untagged and Tagged.

HP 
[image]
or D-Link
[image]

## Expected behavior:
As there's more options then what is common, as seen above, even so slightly. Or maybe it's just the fourth option that is not common, Primary VLAN ID for example is 'new and different'. The fact that tagged/untagged (the primary action) is prefixed (logical) with `Egress`. Maybe call it 'Tag egress' which is not only shorter, but has the focus on Tag?
'Do not participate' is also quite long, off was nice and short, but not completely accurate, because it implies the port is off of course.

Maybe the most important bit that is mssing is a (short) description (like HP offers). Didn't luci have this int he past?

## Additional Information:
OpenWrt version information from system `/etc/openwrt_release`

**Example: Replace the following lines and remove this**
```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='SNAPSHOT'
DISTRIB_REVISION='r20181+4-a436923133'
DISTRIB_TARGET='realtek/rtl930x'
DISTRIB_ARCH='mips_24kc'
DISTRIB_DESCRIPTION='OpenWrt SNAPSHOT r20181+4-a436923133'
DISTRIB_TAINTS='no-all busybox'
```

jow-, 2022-07-27:
So `Tag egress`, `Untag egress`, `Not member`?

`Local` means that the bridge itself can see/receive traffic on this VLAN. Without it, traffic is purely forwarded among ports but never seen by the host system netdev (bridge.vid)

oliv3r, 2022-07-27:
> So `Tag egress`, `Untag egress`, `Not member`?

I think that makes it much easier to relate (Not _a_ member maybe)

> 
> `Local` means that the bridge itself can see/receive traffic on this VLAN. Without it, traffic is purely forwarded among ports but never seen by the host system netdev (bridge.vid)
Yeah, but it does that via the connection that is eth0, e.g. port.28 (in this particiular example) so lan1 -> port1 -> switch-chip -> port28 -> eth0 -> linux

But what if I have 2 ports connected? realtek chips offer this (as probably many others do to, either for bandwith or management reasons (eth0: lan, eth1: Wan; but not connected directly to a port, but routed via the chip)

jow-, 2022-07-27:
I don't understand the question, sorry. As far as I know, the `local` option for Linux bridge VLANs simply controls whether the host system itself (or rather bridge netdev itself) is added as "port" to a VLAN. If unset, the kernel will forward frames among the ports in that bridge VLAN but the bridge netdev itself will not get those frames delivered by the kernel.

This should be independent from whether you have one, two or 10 ports in a bridge VLAN. It also has nothing to do with any externally facing ports.

oliv3r, 2022-07-28:
So DSA is completly new to me, and I'm trying to wrap my head around it. Documentation seems to be quite scarse. The kernel wiki has a good [howto](https://www.kernel.org/doc/html/latest/networking/dsa/configuration.html), however I've not managed to actually apply it to my switch :), which got me to use luci to see if I can graphically set it up.

I have been using openwrt since white-russian :) and currently use 21.03 (swconfig) on my 3 accesspoints (they are not routing) that have vlan's configured. So that is my reference point, standard linux vlan, with 21.03 swconfig based openwrt.

In luci on 21.03 I have the 'CPU' port (eth0) that _also_ connects to the switch chip on a specific port. So I assumed/thought that this was the 'local' flag.

so your simple statement:
```
 bridge VLANs simply controls whether the host system itself (or rather bridge netdev itself) is added as "port" to a VLAN.
```
doesn't make sense to me, i simply don't understand it :( in the past, one would connect eth0 of the SoC to the switch chip via a specific port.

But lets not get too much into `local` flag, I'll learn/figure that out; this ticket was mostly about the confusing naming :)

jow-, 2022-07-28:
`local` is what the CPU port used to be in swconfig. On swconfig you could also setup VLANs without including the CPU port, in this case the switch backplane would forward traffic among the ports but not to the CPU (host system).

The same could be done using bridge VLANs by omitting the "local" flag.

With DSA/bridge VLAN filtering the "CPU ports" are handled transparently by the kernel, they're not exposed in VLAN configuration anymore. There's just the abstract `local` flag to tell whether VLAN port group traffic shall be connected to the SoC or not (= including the CPU port(s) or not in swconfig terms)

oliv3r, 2022-07-29:
Ah, but then it's what I asked before; right now, `local` refers to (I assume) the only CPU port (port 28 on the switch) connecting to eth0 (in my particular case).

So what if port 27 on the switch, internally connects to eth1.

Does `local` affect just the one? both?

Btw, on topic, I think `U` looks better then `u`, but that's of course personal taste :)

jow-, 2022-07-29:
`local` does affect all CPU ports, it's handled transparently by the kernel.

The `u`, `t` and `*` letters follow uci configuration syntax, I prefer to keep them in sync with what the ui shows.

oliv3r, 2022-08-01:
> The `u`, `t` and `*` letters follow uci configuration syntax, I prefer to keep them in sync with what the ui shows.

While that makes perfect sense to a developer/cli-user. Much less for an end-user that'll only click through a UI. What if in the future graphics would be used for that (icons). What about translations? Non-english UI's? Tagged could be called (literally translated) 'labeld' but the UI would nowl show 'lt'? (notice how much less readable that becomes).

Also from a readability POV capitals are slightly easier to read of course. So for _display_ purposes i'd "upper" them by default, and let .po files handle the details :) that internally within luci it is called `u` and `t` of course is perfectly sensible.

jow-, 2022-08-01:
The `u`, `t` and `*` letters are not supposed to get translated.

oliv3r, 2022-08-01:
I understand, but from a UX point of view, that's not an ideal concept :) The letters where chosen, to match the English language. So if your UI is in thai, you all of a sudden have these symbols that make no sense and are hard to give context :)

Just some food for thought ...

jow-, 2022-10-13:
Closing this as it is unlikely to get changed.

closed 2022-10-13 by jow-