openwrt/luci #5907: luci-mod-dashboard: 4 Features and additions for each panel...

aphorise, 2022-07-29:
Some customisation or configuration would be nice for a few features which I'll strive to draw using the attached image as well.

1. Wireless DHCP devices set via alternative bridge devices / connection and not wireless interfaces on OpenWRT device itself. Allow for selecting Wireless devices based on complete connection pool.

2. Show WAN IP via `dig` / other lookup if not in WAN DHCP mode and already with a Public IP. There could be an @ symbol next to the IPv6 & IPv4 addresses to the signify look up vs without meaning direct WAN DHCP IP.

3. Allow for device specific logo or renaming to better distinguish larger networks.

4. Additional system stats like RAM, Storage & Last down / since Online as well as total SLA / uptime overall in %.

It would be nice if there's a configuration page where some of these could be enabled and set.
[image]

## Actual behavior:
Does not correctly populate Wireless area with devices not directly set via OpenWRT / Wireless NIC on OpenWRT (related to: [#4472)](https://github.com/openwrt/luci/issues/4472))

## Expected behavior:
Allow for user customisation and further option as detailed already.

jow-, 2022-07-29:
Please provide minimal proof of concepts on command line, e.g. on how you think how "actual wifi devices" can be distinguished from other ARP neighbors if the router itself has no wireless interface. Likewise describe, ideally in terms of cli commands, the "dig" feature you propose. It is totally unclear what is meant with it.

@ZoZhang - maybe you could look into those as you did the dashboard implementation?

aphorise, 2022-07-29:
> .. how you think how "actual wifi devices" can be distinguished from other ARP neighbors if the router itself has no wireless interface.

Leave the distinction to user by way of a configuration page where known devices (by MAC) can be selected via a checkbox next to each known session. 

What's would be good is also to allow for `disable wifi panel` stats as per the related request (#4472).

It should also be possible to rename devices on the same listed device configuration page which will help where devices are actually listed as `?` (as per my image). Ideally it would also be possible to set a small logo for that particular device (irrespective of whether its WiFi or not).

> `dig`

to determine WAN IPs if lookup is required as per:
```bash
# // if DHCP address is not in private range:
# // 10.0.0.0/8, 192.168.0.0/16, 172.16.0.0/16, 169.254.0.0/16
# // then perform dig to find WAN
WANv4_IP=$(dig +short @1.1.1.1 ch txt whoami.cloudflare |  cut -d'"' -f2) ;
echo "WANv4 IP == @${WANv4_IP}" ;
# // ^^^ repeat with IPv6
WANv6_IP=$(dig +short @2606:4700:4700::1111 -6 ch txt whoami.cloudflare|  cut -d'"' -f2) ;
echo "WANv6 IP == @${WANv6_IP}" ;
# // ...

# do unconditional reverse DNS lookup:
echo "RDNS WAN IP == $(dig +short -x ${WANv4_IP})"
  # 77-66-55-44.cable.dynamic.v4.ziggo.nl.
echo "RDNS WAN IP == $(dig +short -x ${WANv6_IP})"
```

jow-, 2022-07-30:
> Leave the distinction to user by way of a configuration page where known devices (by MAC) can be selected via a checkbox next to each known session.

The wifi list is supposed to show details like signal strength. This information is not available in this case and would lead to incomplete records. Not sure if it is worth it.

> to determine WAN IPs if lookup is required as per:

The above might be an unexpected privacy violation (especially with IPv6) if internal hosts suddenly query public name servers without explicit user consent. It would also query hardcoded cloudflare nameservers which were not configured by the user. Far from ideal imho.

aphorise, 2022-07-30:
> The wifi list is supposed to show details like signal strength. This information is not available in this case and would lead to incomplete records. Not sure if it is worth it.

Sure - but in my case as you can see in the initial image I shared that there's no device distinction for those clients which I've highlighted that are DHCP set by OpenWRT but using other wireless devices. IMO - it would be better to have the panel populated with known and user selectable wireless devices than with nothing - as well just offering to disable this panel too.

> The above might be an unexpected privacy violation (especially with IPv6) if internal hosts suddenly query public name servers without explicit user consent.

I agree. However if the user is willing to opt in to it then why not? it surely cannot be worse than a HTTP / Web request for the same :-D - what's more it would only be conditional to those router / devices that are in NAT and not in Public WAN / IP ranges.

>  It would also query hardcoded cloudflare nameservers which were not configured by the user. Far from ideal imho.

Why hard coded and not user settable? - the Cloudflare could be a template suggestion when user enable it - and any others that the user may wish to have configure instead they could do so as part of that enable / save process.

ZoZhang, 2022-08-04:
Hello, I will be a bit busy because I have a new job recently. …

@aphorise maybe we can add a simply text stating that no device is currently connected to wifi ?

I agree to add more current router information on the "system" card. However, when it comes to more professional information, I may not be able to deal with it, because I have no router equipment available at the moment. I was under the impression that I could only get basic network data when I was developing...

@jow - this does probably take a while to process, especially once I need the job to stabilize...

aphorise, 2022-08-09:
Hey @ZoZhang I can contribute €400 for these features requested - would that help? :-) 

If some or 2 out of 4 of the issues can be addressed with some customisation or framework for it in the code then maybe I can help with the reset and it would give me a lead into Luci. Anyway happy to help and thank you for your efforts.

ZoZhang, 2022-08-10:
Hey @aphorise Thank you for your trust and help, I think it's really nice !

However, I am busy this month and I use the sharing connection with my phone.
I think if you can help me a router OpenWrt, that will be enough for me and I'm also glad that I can do useful things on Luci.
So, I will continue on this subject when I have more time than now.

Finally, thank you ! ^ ^

aphorise, 2022-08-10:
@ZoZhang sure! - can you pls reach out using the same handle at g-mail? - I'll be happy to get a [NanoPi R5S](https://www.aliexpress.com/item/1005004309166883.html?spm=a2g0o.productlist.0.0.56705735Z46aDs&algo_pvid=b1187e6f-084a-4498-8e12-fd1f86765154&algo_exp_id=b1187e6f-084a-4498-8e12-fd1f86765154-0&pdp_ext_f=%7B%22sku_id%22%3A%2212000028771167397%22%7D&pdp_npi=2%40dis%21EUR%21102.39%21102.39%21%21%21%21%21%4021362f1a16601602754672257e193e%2112000028771167397%21sea&curPageLogUid=PXiGdTnVsiRY) or [FastRhino R68s](https://www.aliexpress.com/item/1005004368978198.html?spm=a2g0o.productlist.0.0.cddd4353r5uVK6&algo_pvid=b887bfb8-2e62-4df9-922b-d363c7f7b861&algo_exp_id=b887bfb8-2e62-4df9-922b-d363c7f7b861-1&pdp_ext_f=%7B%22sku_id%22%3A%2212000028932165395%22%7D&pdp_npi=2%40dis%21EUR%21125.28%21125.28%21%21%21%21%21%402100bdde16601605502982398e2ab9%2112000028932165395%21sea&curPageLogUid=5Q77UU5wZOvq) to you or any other variant as you see fit - those also dont have wireless devices by default and in my case for example I use a few bridged devices that are driven by (Qualcomm IP4xxx) Powerline adaptors / via AC outlet connected to my OpenWRT router.

ZoZhang, 2023-02-14:
@aphorise Can you tell me how to do this? I don't seem to have used it like that. 
Thank you very much for your trust and support! 
But I have a very important thing recently, I will plan my trip later, if I can continue. I'll talk to you then.

aphorise, 2023-02-17:
@ZoZhang hey man - tell me what you're in need of? - are you after a few example devices? - you have my email right? same handle `at gmail dot com` :-)

ZoZhang, 2023-02-22:
@aphorise hey man, I sent an email to your Gmail last Saturday, but it didn't seem to work.
Actually, I've had some problems with reality lately and I need to take some time to adjust myself.
Sorry about the improvements on the router, I can only put it on hold for now. ;(

aphorise, 2023-08-02:
In the latest rebuild I did of an identical host / setup it does look as I'd expect where the WiFi panel is not present - maybe in my prior install I configured things differently?

[image]

aphorise, 2023-08-26:
After installing a new WiFi device the panel then appears - I still dont get what's different from my earlier install / device which does not have a WiFi module at all.

Anyway this may be another issue separate to the original features request.

[image]

Alkhalidi84, 2026-02-20:
> thanks for the some tips. modified my router. that's exactly what I was missing (memory status, signal, connected time of wifi client, more info in dhcp).
> <img alt="Image" width="713" height="374" src="https://private-user-images.githubusercontent.com/152337177/485150694-669a4fd2-3456-4540-8971-50f8703ed3d0.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NzE1NjI0NTMsIm5iZiI6MTc3MTU2MjE1MywicGF0aCI6Ii8xNTIzMzcxNzcvNDg1MTUwNjk0LTY2OWE0ZmQyLTM0NTYtNDU0MC04OTcxLTUwZjg3MDNlZDNkMC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjYwMjIwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI2MDIyMFQwNDM1NTNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT00NDdkZDAxNzI2YzZlZDMxNGUzMWY2Y2I3NGNkZGFiZGI3MWU2YzdjOThjNTEyMGZhYzcxYmMwMmQ0N2UyMmE0JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.N_tFPCd7L2fGeEOGtDm9M1BX5nTu4-YW4_zqXvCPmdk"> <img alt="Image" width="361" height="227" src="https://private-user-images.githubusercontent.com/152337177/485513805-40c896e2-fc16-4dd4-bab3-e23db82a4036.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NzE1NjI0NTMsIm5iZiI6MTc3MTU2MjE1MywicGF0aCI6Ii8xNTIzMzcxNzcvNDg1NTEzODA1LTQwYzg5NmUyLWZjMTYtNGRkNC1iYWIzLWUyM2RiODJhNDAzNi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjYwMjIwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI2MDIyMFQwNDM1NTNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1hMTJhODhiN2IxZDcxMjJhNjlkMWUyNzg2YTY2ZTJmMTVlMjBhZjQ5YTZmMzdiNjNkYjY0NDRjMDZjMWVmMjk4JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.eykeyiBoHriS16jQenxEbXIgERxoz1DIFPIBVi_1_iY">

how to do that can send the method

zeropoint-exception, 2026-10-19:
I freakin' love this!

The amount of data without overwhelming the user is amazing. Everything is properly sorted, color-coded, and displayed as graphs, without taking information away from the user. It shows everything you need to know, at a glance, with as little mental load as possible (or as regular people would say, its "easy to understand").

This is what a free-as-in-freedom Router should look like in 2026!

I propose to even make this the new standard for OpenWRT routers. Period.