openwrt/luci #5909: LuCI doesn't show all active connections

abajk, 2022-07-30:
The device is BT Home Hub 5A. An iperf was started between the LAN and WAN interfaces. On the router, active connections were checked using conntrack:
```
root@OpenWrt:/# conntrack -L
tcp      6 7439 ESTABLISHED src=192.168.1.10 dst=192.168.1.1 sport=44656 dport=80 packets=13 bytes=1610 src=192.168.1.1 dst=192.168.1.10 sport=80 dport=44656 packets=12 bytes=1004 [ASSURED] mark=0 use=1
tcp      6 7439 ESTABLISHED src=192.168.1.10 dst=192.168.1.1 sport=44650 dport=80 packets=13 bytes=1595 src=192.168.1.1 dst=192.168.1.10 sport=80 dport=44650 packets=12 bytes=1006 [ASSURED] mark=0 use=1
tcp      6 src=192.168.1.10 dst=10.0.1.1 sport=39816 dport=5201 packets=11700340 bytes=17550505701 src=10.0.1.1 dst=10.0.1.2 sport=5201 dport=39816 packets=965711 bytes=50270932 [OFFLOAD] mark=0 use=2
tcp      6 7439 ESTABLISHED src=192.168.1.10 dst=192.168.1.1 sport=44638 dport=80 packets=20 bytes=2757 src=192.168.1.1 dst=192.168.1.10 sport=80 dport=44638 packets=19 bytes=3061 [ASSURED] mark=0 use=1
tcp      6 113 TIME_WAIT src=192.168.1.10 dst=192.168.1.1 sport=44590 dport=80 packets=16 bytes=2176 src=192.168.1.1 dst=192.168.1.10 sport=80 dport=44590 packets=14 bytes=2620 [ASSURED] mark=0 use=1
tcp      6 7439 ESTABLISHED src=192.168.1.10 dst=192.168.1.1 sport=44606 dport=80 packets=40 bytes=5850 src=192.168.1.1 dst=192.168.1.10 sport=80 dport=44606 packets=36 bytes=11106 [ASSURED] mark=0 use=1
tcp      6 7223 ESTABLISHED src=192.168.1.10 dst=10.0.1.1 sport=39804 dport=5201 packets=9 bytes=620 src=10.0.1.1 dst=10.0.1.2 sport=5201 dport=39804 packets=8 bytes=428 [ASSURED] mark=0 use=1
tcp      6 7439 ESTABLISHED src=192.168.1.10 dst=192.168.1.1 sport=44622 dport=80 packets=19 bytes=2854 src=192.168.1.1 dst=192.168.1.10 sport=80 dport=44622 packets=18 bytes=1691 [ASSURED] mark=0 use=1
conntrack v1.4.6 (conntrack-tools): 8 flow entries have been shown.
```

LuCI does not show this connection for some reason:
[image]

Active connections on the host:
[image]

The most interesting thing is that conntrack shows 8 active connections. LuCI shows 7 TCP connections, and only 6 are listed below.

I was able to reproduce the same problem on my main router (TP-Link WDR4300) on OpenRT 22.03.

abajk, 2022-07-30:
Maybe this is due to the fact that software flow offloading is enabled on my devices. Offloaded connections do not have the ESTABLISHED flag.

jow-, 2022-07-30:
Please share the relevant `/proc/net/nf_conntrack` output, this is what is parsed.

abajk, 2022-07-31:
@jow- 
I updated the issue description with the contents of `/proc/net/nf_conntrack`. The connection is on this list, but has been filtered out for some reason by LuCI.

closed 2022-08-01 by jow-

abajk, 2022-08-06:
I confirm that the fix works, however, it incorrectly counts statistics when software flow offload is enabled ([Issue](https://github.com/openwrt/openwrt/issues/10399)).

commit abd9125f (2022-08-01): luci-base: sys.lua: make timeout optional when parsing conntrack entries

Fixes: #5909
commit d09fbe0b (2022-08-01): luci-base: sys.lua: make timeout optional when parsing conntrack entries

Fixes: #5909

(cherry picked from commit abd9125fbf5e2fadc0c5ae293496680df08e6e1b)
commit 87836cad (2022-08-01): luci-base: sys.lua: make timeout optional when parsing conntrack entries

Fixes: #5909

(cherry picked from commit abd9125fbf5e2fadc0c5ae293496680df08e6e1b)