openwrt/luci #5910: luci-network: TypeError while editing wireless network on second radio on gl-mt1300_v1

kevinsumner, 2022-07-31:
## Steps to reproduce:
1. Perform a fresh install or reset on hardware with two radios, such as GL-iNet MT1300. (This isn't required, but ensures starting from a clean install.)
2. Go to: Network → Wireless
3. Ensure one wireless network on each radio, adding wireless networks as needed (defaults for networks ok).
4. Edit first wireless network (`radio0.network1`), change SSID, then click Save.
5. Save & Apply.
6. Edit second wireless network (`radio1.network1`), change SSID, then click Save, then click Dismiss. (Save doesn't dismiss dialog like expected, possibly start of bugged behavior)
7. Save & Apply.
8. Click Edit button for the second wireless network.

## Actual behavior:
UI presents the error "TypeError chan.options[0] is undefined" and no edit dialog is shown.

## Expected behavior:
The dialog for editing wireless network is displayed.

## Additional Information:
```
root@OpenWrt:~# cat /etc/openwrt_release
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='21.02.3'
DISTRIB_REVISION='r16554-1d4dea6d4f'
DISTRIB_TARGET='ramips/mt7621'
DISTRIB_ARCH='mipsel_24kc'
DISTRIB_DESCRIPTION='OpenWrt 21.02.3 r16554-1d4dea6d4f'
DISTRIB_TAINTS=''
```

Some other notes:
- What exactly is edited on the wireless networks doesn't seem to matter, but SSID consistently reproduces.
- Adding a new wireless network on the second wireless interface (`radio1`) gives the same error.
- Editing wireless networks on the second wireless interface works as expected prior to editing the wireless network on the first radio.
- This works on a GL-AR750S with OpenWrt 19.07.3 and 21.02.3 (after installing kernel modules for hardware support). I'm not sure why the hardware would matter.

jow-, 2022-07-31:
This usually happens when iwinfo cannot identify the second radio phy. Please provide the output of `uci show wireless` (strip passwords is needed).

kevinsumner, 2022-07-31:
Triggering the behavior doesn't require a Save & Apply, just a Save on the dialog. Edited the issue description for that.

Fresh install info:

```
root@OpenWrt:~# uci show wireless
wireless.radio0=wifi-device
wireless.radio0.type='mac80211'
wireless.radio0.path='1e140000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0'
wireless.radio0.channel='36'
wireless.radio0.band='5g'
wireless.radio0.htmode='VHT80'
wireless.radio0.disabled='1'
wireless.default_radio0=wifi-iface
wireless.default_radio0.device='radio0'
wireless.default_radio0.network='lan'
wireless.default_radio0.mode='ap'
wireless.default_radio0.ssid='OpenWrt'
wireless.default_radio0.encryption='none'
wireless.radio1=wifi-device
wireless.radio1.type='mac80211'
wireless.radio1.path='1e140000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0+1'
wireless.radio1.channel='36'
wireless.radio1.band='5g'
wireless.radio1.htmode='VHT80'
wireless.radio1.disabled='1'
wireless.default_radio1=wifi-iface
wireless.default_radio1.device='radio1'
wireless.default_radio1.network='lan'
wireless.default_radio1.mode='ap'
wireless.default_radio1.ssid='OpenWrt'
wireless.default_radio1.encryption='none'
root@OpenWrt:~# iwinfo
wlan0     ESSID: unknown
          Access Point: 94:83:C4:0E:A1:E9
          Mode: Client  Channel: unknown (unknown)
          Center Channel 1: unknown 2: unknown
          Tx-Power: 3 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: unknown
          Bit Rate: unknown
          Encryption: unknown
          Type: nl80211  HW Mode(s): 802.11nac
          Hardware: 14C3:7615 7615:14C3 [MediaTek MT7615E]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy0

wlan1     ESSID: unknown
          Access Point: 16:83:C4:0E:A1:E9
          Mode: Client  Channel: unknown (unknown)
          Center Channel 1: unknown 2: unknown
          Tx-Power: 3 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: unknown
          Bit Rate: unknown
          Encryption: unknown
          Type: nl80211  HW Mode(s): 802.11nac
          Hardware: 14C3:7615 7615:14C3 [MediaTek MT7615E]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy1
```

After the update that triggers the behavior:

```
root@OpenWrt:~# uci show wireless ; iwinfo
wireless.radio0=wifi-device
wireless.radio0.type='mac80211'
wireless.radio0.path='1e140000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0'
wireless.radio0.channel='36'
wireless.radio0.band='5g'
wireless.radio0.htmode='VHT80'
wireless.radio0.disabled='1'
wireless.default_radio0=wifi-iface
wireless.default_radio0.device='radio0'
wireless.default_radio0.network='lan'
wireless.default_radio0.mode='ap'
wireless.default_radio0.ssid='OpenWrt'
wireless.default_radio0.encryption='none'
wireless.radio1=wifi-device
wireless.radio1.type='mac80211'
wireless.radio1.path='1e140000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0+1'
wireless.radio1.channel='36'
wireless.radio1.band='5g'
wireless.radio1.htmode='VHT80'
wireless.radio1.disabled='1'
wireless.default_radio1=wifi-iface
wireless.default_radio1.device='radio1'
wireless.default_radio1.network='lan'
wireless.default_radio1.mode='ap'
wireless.default_radio1.ssid='OpenWrt'
wireless.default_radio1.encryption='none'
wlan0     ESSID: unknown
          Access Point: 94:83:C4:0E:A1:E9
          Mode: Client  Channel: unknown (unknown)
          Center Channel 1: unknown 2: unknown
          Tx-Power: 3 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: unknown
          Bit Rate: unknown
          Encryption: unknown
          Type: nl80211  HW Mode(s): 802.11nac
          Hardware: 14C3:7615 7615:14C3 [MediaTek MT7615E]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy0

wlan1     ESSID: unknown
          Access Point: 16:83:C4:0E:A1:E9
          Mode: Client  Channel: unknown (unknown)
          Center Channel 1: unknown 2: unknown
          Tx-Power: 3 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: unknown
          Bit Rate: unknown
          Encryption: unknown
          Type: nl80211  HW Mode(s): 802.11nac
          Hardware: 14C3:7615 7615:14C3 [MediaTek MT7615E]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy1
```

I also noticed that LuCI also doesn't know some info about the second radio. Here's a screenshot.

[image]

jow-, 2022-08-01:
As expected. The 2nd radio uses the mutli-phy radio path syntax (`0000:01:00.0+1') which is not understood by the iwinfo version shipped in 21.02.3. Support has been backported but must be manually installed. Please retry after `opkg update; opkg upgrade iwinfo libiwinfo libiwinfo-lua`

kevinsumner, 2022-08-01:
Updating the iwinfo packages appears to have resolved it. Is this update already integrated into the upcoming 22.03.0? I'm going to update the wiki entry for gl-mt1300_v1.

kevinsumner, 2022-08-01:
Note added to the wiki: https://openwrt.org/toh/gl.inet/gl-mt1300_v1#iwinfo_update_for_openwrt_2102

jow-, 2022-08-01:
Yes, 22.03 will include the relevant changes. Closing this as it is already solved.

closed 2022-08-01 by jow-

66schoollane, 2022-08-16:
hi jow
how do i update to 22.03
just flashed from 
OpenWrt 21.02.3 r16554-1d4dea6d4f / LuCI openwrt-21.02 branch git-22.083.69138-0a0ce2a

66schoollane, 2022-08-16:
> Yes, 22.03 will include the relevant changes. Closing this as it is already solved.

hi jow
how do i update to 22.03
just flashed from
OpenWrt 21.02.3 r16554-1d4dea6d4f / LuCI openwrt-21.02 branch git-22.083.69138-0a0ce2a