openwrt/luci #5911: luci-led-trigger: Allow hartbeat to be inverted

oliv3r, 2022-07-31:
## Steps to reproduce:
1. go to: System → LED configuration → Add LED action  → Pick heartbeat as trigger

## Actual behavior:
1. Unlike the custom timer, the hearbeat trigger does not pop out additional options, such as (the only one) a checkbox to `invert`

## Expected behavior:
```
When selecting the heartbeat trigger, I wish to be able to set the `invert` flag.
```

## Additional Information:
OpenWrt version information from system `/etc/openwrt_release`

```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='SNAPSHOT'
DISTRIB_REVISION='r20152+61-a3e4949998'
DISTRIB_TARGET='realtek/rtl930x'
DISTRIB_ARCH='mips_24kc'
DISTRIB_DESCRIPTION='OpenWrt SNAPSHOT r20152+61-a3e4949998'
DISTRIB_TAINTS='no-all busybox'
```

feckert, 2022-08-02:
I don't understand your question. When I select the kernel trigger hartbeat, the trigger is set. Why should I invert that?

oliv3r, 2022-08-02:
Why not? Why does a power led go on during normal operation, but the standby led go on during standby :)
Snarkyness aside,
Try it :D,  the kernel expose the invert flag for a reason of course.

If you look at your LED (with the heartbeat trigger) you'll see something like
```
       __                    __                     __       __       __                      __
___|   |__________|   |__________|   |___|   |___|   |___________|   |
```
E.g. it's dark most of the time, and activity causes the led to briefly light up (depending on load) and then is dark again.

With the inversion flag set, we get the opposite.
```
___    ___________    ___________    ____   ____   ____________
      |_|                      |_|                      |_|        |_|       |_|
```
E.g. the led is usually lid, and on activity it briefly dims (depending on load) and is then lit again.

For one, the inverted heartbeat helps you to see 'hey, the LED is here, it's on, and oh, there was a blink. The non-inverted LED, makes you hmm, where is it, is it actually still on, ah there was a brief blink.

Using any random led for the heartbeat makes this less of an issue maybe, but if there is only one LED, and that one LED is also your power LED, yeah, having the 'normally on' to indicate power, then 'briefly dim' to show load, is of course very useful.

I hope this helps a bit @feckert ?

feckert, 2022-08-03:
I didn't know that. Thank you for the hint and your explanation.
The following changes must be made:
- Update the hearbeat trigger in core openwrt to set invert https://github.com/TDT-AG/openwrt/tree/pr/20220803-base-files
- Add this option to the LuCI https://github.com/TDT-AG/luci/tree/pr/20220803-luci-mod-system

Please test and report back so I could upstream this changes.

oliv3r, 2022-08-03:
@feckert I opened up a MR already here with regard to those changes here https://github.com/openwrt/openwrt/pull/10358 which should be exactly what we want to achieve this. I tested it and it works just fine. So not sure what your PR branch is trying to do if not the same? Since it's being in the process of being reviewed, I'll rather keep that there as is :)

As for the LuCI option, thanks, I'll definitely test that one :)

feckert, 2022-08-03:
Ok fine with me. I will update the LuCI pullrequest and merge this if your suggestion is upstream.

oliv3r, 2022-08-03:
Thanks, the PR's are linked; but I'll ping here once it's merged upstream. Meanwhile, I'll test the LuCI change; but gimme some time, as I have to prepare my setup once again for it :p

PR #5923 (2022-08-08): luci-mod-system: add config option to invert heartbeat trigger
Depends on https://github.com/openwrt/openwrt/pull/10358
Fixes #5911
closed 2022-08-08 by feckert