openwrt/luci #5917: luci-app-mwan3: json-rpc not available

dinana, 2022-08-02:
## Steps to reproduce:
As per `https://github.com/openwrt/luci/wiki/JsonRpcHowTo`
POST request to http://192.168.0.1/cgi-bin/luci/rpc/mwan3 

## Actual behavior:
HTML 404 response received

## Expected behavior:
JSON response with actual results

documentation for methods and parameters needed as well

## Additional Information:
OpenWrt version information from system `/etc/openwrt_release`

```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='21.02.1'
DISTRIB_REVISION='r16325-88151b8303'
DISTRIB_TARGET='x86/64'
DISTRIB_ARCH='x86_64'
DISTRIB_DESCRIPTION='OpenWrt 21.02.1 r16325-88151b8303'
DISTRIB_TAINTS=''
```

jow-, 2022-08-02:
The version in 21.02 does not provide an RPC interface.

closed 2022-08-02 by jow-

dinana, 2022-08-02:
Would that be available in a later release? Was it available in previous releases and lo longer supported? Can you please provide some additional info, thanks.

jow-, 2022-08-02:
It is available in 22.03 onwards. The API is for internal consumption of the mwan3 ui btw, it is not meant for public consumption (no compatibility guarantees, methods and arguments can change without notice etc.)