openwrt/luci #5919: luci-app-advanced-reboot: linksys EA4500 not recognized

danorf69, 2022-08-02:
## Steps to reproduce:
1. on Linksys EA4500 go to: System → Advanced-Reboot
.
## Actual behavior:
web page says that linksys,ea4500 is not a recognized device.

## Expected behavior:
web page offers to reboot from one of the two firmware (actual or backup).

## Additional Information:
Modifying /usr/share/advanced-reboot/devices/linksys-ea4500.json to change line 4
-- "boardNames": [ "linksys-e4500", "linksys,e4500" ],
++ "boardNames": [ "linksys-ea4500", "linksys,ea4500" ],
Solve the problem.

hnyman, 2022-08-02:
Already fixed in master three weeks ago with commit 5d7446e1b from PR #5885

Not sure if @stangri has already backported the fix to 22.03 and 21.02 ...

stangri, 2022-08-02:
Thanks @hnyman for the tag, completely forgot to backport the fix. Resolved via 71017a530473bf52d5fbf507c62be096975f2bfb and 7b701aeea124cf778ced6bb5894fabc40fc17d85.

closed 2022-08-02 by stangri

danorf69, 2022-08-03:
Thank you for all the work !