openwrt/luci #5937: luci-dashboard TypeError chan.options[0] is undefined on Raspberry PI 3B+

DN2048, 2022-08-17:
## Steps to reproduce:

1. go to: Network → Wireless
2.  Edit radio0 wireless network

## Actual behavior:

Get an error saying: TypeError chan.options[0] is undefined

## Expected behavior:

Edit the wireless network

## Additional Information:
```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='22.03.0-rc6'
DISTRIB_REVISION='r19590-042d558536'
DISTRIB_TARGET='bcm27xx/bcm2709'
DISTRIB_ARCH='arm_cortex-a7_neon-vfpv4'
DISTRIB_DESCRIPTION='OpenWrt 22.03.0-rc6 r19590-042d558536'
DISTRIB_TAINTS=''
```

The device is a Raspberry PI 3B+ with an external wireless USB device. The problem is editing the internal wifi (radio0). Editing the wireless USB device (radio1) works fine.

[image]

jow-, 2022-08-22:
Please provide the contents of `/etc/config/wireless` and the outputs of `ubus call iwinfo info '{ "device": "radio0" }'` as well as `ubus call iwinfo freqlist '{ "device": "radio0" }'`

DN2048, 2022-08-23:
**/etc/config/wireless:**
```
config wifi-device 'radio0'
        option type 'mac80211'
        option path 'platform/soc/3f300000.mmcnr/mmc_host/mmc1/mmc1:0001/mmc1:0001:1'
        option htmode 'HT20'
        option disabled '0'
        option short_gi_40 '0'

config wifi-device 'radio1'
        option type 'mac80211'
        option path 'platform/soc/3f980000.usb/usb1/1-1/1-1.3/1-1.3:1.0'
        option channel '1'
        option band '2g'
        option htmode 'HT20'
        option disabled '0'

config wifi-iface 'wifinet1'
        option device 'radio1'
        option mode 'sta'
        option network 'wwan'
        option ssid '---'
        option encryption 'psk2'
        option key '---'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option network 'lan'
        option mode 'ap'
        option ssid '---'
        option encryption 'psk2'
        option key '---'
        option channel '8'
```

**ubus call iwinfo info '{ "device": "radio0" }'**
```
{
	"phy": "phy0",
	"bssid": "B8:27:EB:D6:E8:86",
	"country": "99",
	"mode": "Master",
	"channel": 6,
	"center_chan1": 6,
	"frequency": 2437,
	"frequency_offset": 0,
	"txpower": 31,
	"txpower_offset": 0,
	"quality_max": 70,
	"noise": 0,
	"htmodes": [
		"HT20",
		"HT40",
		"VHT20",
		"VHT40",
		"VHT80"
	],
	"hwmodes": [
		"ac",
		"b",
		"g",
		"n"
	],
	"hwmode": "n",
	"htmode": "HT20",
	"hardware": {
		"id": [
			720,
			43430,
			0,
			0
		],
		"name": "Cypress CYW43455"
	}
}
```

**ubus call iwinfo freqlist '{ "device": "radio0" }'**
```
{
	"results": [
		{
			"channel": 1,
			"mhz": 2412,
			"restricted": false,
			"active": false
		},
		{
			"channel": 2,
			"mhz": 2417,
			"restricted": false,
			"active": false
		},
		{
			"channel": 3,
			"mhz": 2422,
			"restricted": false,
			"active": false
		},
		{
			"channel": 4,
			"mhz": 2427,
			"restricted": false,
			"active": false
		},
		{
			"channel": 5,
			"mhz": 2432,
			"restricted": false,
			"active": false
		},
		{
			"channel": 6,
			"mhz": 2437,
			"restricted": false,
			"active": true
		},
		{
			"channel": 7,
			"mhz": 2442,
			"restricted": false,
			"active": false
		},
		{
			"channel": 8,
			"mhz": 2447,
			"restricted": false,
			"active": false
		},
		{
			"channel": 9,
			"mhz": 2452,
			"restricted": false,
			"active": false
		},
		{
			"channel": 10,
			"mhz": 2457,
			"restricted": false,
			"active": false
		},
		{
			"channel": 11,
			"mhz": 2462,
			"restricted": false,
			"active": false
		},
		{
			"channel": 36,
			"mhz": 5180,
			"restricted": false,
			"active": false
		},
		{
			"channel": 40,
			"mhz": 5200,
			"restricted": false,
			"active": false
		},
		{
			"channel": 44,
			"mhz": 5220,
			"restricted": false,
			"active": false
		},
		{
			"channel": 48,
			"mhz": 5240,
			"restricted": false,
			"active": false
		},
		{
			"channel": 149,
			"mhz": 5745,
			"restricted": false,
			"active": false
		},
		{
			"channel": 153,
			"mhz": 5765,
			"restricted": false,
			"active": false
		},
		{
			"channel": 157,
			"mhz": 5785,
			"restricted": false,
			"active": false
		},
		{
			"channel": 161,
			"mhz": 5805,
			"restricted": false,
			"active": false
		},
		{
			"channel": 165,
			"mhz": 5825,
			"restricted": false,
			"active": false
		}
	]
}
```

commit 9af84865 (2022-10-27): luci-mod-network: gracefully handle empty channel lists

Avoid undefined value access when the channel list is empty.

Fixes: #5937, #5951
commit e0760712 (2022-10-27): luci-mod-network: gracefully handle empty channel lists

Avoid undefined value access when the channel list is empty.

Fixes: #5937, #5951

(cherry picked from commit 9af8486517e87eea837fbce0ce787fc89ab46627)