openwrt/luci #5948: Luci - Won't set vlan filtering when expected

ora726, 2022-08-25:
## Steps to reproduce:
1) Goto Network -> Interface -> Devices
2) Click Configure on br-lan
3) Click on Add button WITHOUT selecting "Enable VLAN filtering"
4) When Clicking the Add Button the "Enable VLAN Filtering" is automatically SELECTED by Luci
5) Click Save

## Actual behavior:
In Unsaved Changes Luci show the following commandes to be executed when clicking on Save & Apply
...
uci add network bridge-vlan # =cfg0da1b0
uci set network.@bridge-vlan[-1].device='br-lan'
uci set network.@bridge-vlan[-1].vlan='1'
...

Luci does NOT enable vlan filtering, following line is missing :
...
uci set network.cfg030f15.vlan_filtering='1'
...

When Clicking on Save & Apply, /etc/config/network file is missing the following line in the config device section
...
	option vlan_filtering '1'
...

This result in any vlan setup is not functioning
This is a simplistic config to explain the issue but problem remain also when VLAN details are entered after pressing the Add button.

## Expected behavior:
Command 
...
uci set network.cfg030f15.vlan_filtering='1'
...
Should be added in unsaved changes
Command should be executed when Clicking on Save & Apply
/etc/config/network file should include
...
	option vlan_filtering '1'
...
in the config device section
vlan filtering should be enabled

Workaround is to save and apply then run the command
...
uci set network.cfg030f15.vlan_filtering='1'
...
in a shell and apply the change with 
...
   /etc/init.d/network restart
...

## Additional Information:
OpenWrt version information from system `/etc/openwrt_release`
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='22.03.0-rc6'
DISTRIB_REVISION='r19590-042d558536'
DISTRIB_TARGET='mediatek/mt7622'
DISTRIB_ARCH='aarch64_cortex-a53'
DISTRIB_DESCRIPTION='OpenWrt 22.03.0-rc6 r19590-042d558536'
DISTRIB_TAINTS=''

jow-, 2022-08-25:
This is intentional. netifd will automatically assume `vlan_filtering 1` if at least one bridge VLAN is defined (see https://git.openwrt.org/?p=project/netifd.git;a=blob;f=config.c;h=9bbda39d3fb5259499cc50c10309920b39537364;hb=bfa039c414bc6613cc5361aaf5fa6baf5fa0264b#l124)

LuCI's default logic is to not write flag (checkbox) values which match their defaults, hence no `option vlan_filtering 1` is written since it implicitly defaults to true when at least one bridge-vlan section is present.

closed 2022-08-25 by jow-

ora726, 2022-08-26:
Thanks for highlighting the expected behaviour of Luci, I was not aware.
My comments :
1) Setup is confusing for casual users that don't know of the default behaviour of LUCI and might believe that by not selecting this option they might predefine VLANS but not turn vlan filtering on.
2) Counter intuitive vis a vis documentation at [https://openwrt.org/docs/guide-user/base-system/basic-networking?s=vlan_filtering#options_for_bridges](url) Section **Options for Bridges** where the default for VLAN-Filtering is 0
3) If this is the preferred default behaviour (Assume as VLAN Filtering On if at least one VLAN defined) would it make sense to remove the Checkbox for vlan filtering in Luci, Behaviour would be as expected, confusion removed for casual users, and interface is simplified

jow-, 2022-08-26:
1. Casual users should not be concerned with the uci options rendered by the ui
2. This documentation can be updated to reflect the actual implementation
3. This would remove the ability to temporarily disable VLANs while keeping the VLAN definitions in place