openwrt/luci #5951: chan.options[0] is undefined (OpenWrt 22.03.0-rc6 r19590-042d558536 / LuCI openwrt-22.03 branch git-22.204.42822-9a18337)

blackicewr, 2022-08-27:
OpenWrt 22.03.0-rc6 r19590-042d558536 / LuCI openwrt-22.03 branch git-22.204.42822-9a18337
Netgear 6220 flashed from version 21.02

some day i lost connection to my wifi 2.4
i wanted to provide new network i have:

chan.options[0] is undefined

my /etc/config/wireless:

config wifi-iface 'default_radio1'
	option device 'radio1'
	option network 'lan'
	option mode 'ap'
	option key 'mywifipassword'
	option ssid 'OpenWrt'
	option encryption 'none'

config wifi-device 'radio0'
	option type 'mac80211'
	option path '1e140000.pcie/pci0000:00/0000:00:01.0/0000:02:00.0'
	option cell_density '0'
	option hwmode '11a'
	
config wifi-device 'radio1'
	option type 'mac80211'
	option path '1e140000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0'
	option channel '36'
	option band '5g'
	option htmode 'VHT80'

Ansuel, 2022-08-27:
can you give more info? Is this related to the chan analysis ?

hnyman, 2022-08-27:
To me it looks like you have half-new half-old WiFi options. The definition for radio0 and radio1 differ in style.

I would delete the /etc/config/wireless and reboot, so that your default  WiFi config gets re-created at he reboot in the current style. Then you can modify the config starting from  the current style. (But note that you need wired connection at that time)

blackicewr, 2022-08-27:
Guys i will do what proposed hnyman and let you know. @Ansuel @hnyman

blackicewr, 2022-08-27:
@hnyman @hnyman 
Ok guys. Works after deleting /etc/config/wireless and reboot
Now i can add both configuration to 2.4 and 5.0 networks.
Hope it will solve my problem if not i will back to this forum
greetz