openwrt/luci #5956: luci-proto-wireguard: Wrong generated AllowedIPs

felinira, 2022-09-03:
## Steps to reproduce:
The generated peer configuration in luci-proto-wireguard will put the *peer* IPs in `AllowedIPs`. As this is meant to be the configuration for the peer, those IPs should be configured with `Address` in the `[Interface]` section instead. The `AllowedIPs` configuration should instead include the configured host IP addresses of the interface.

1. go to: Network → Interfaces → Wireguard interface → Peers tab → Edit peer → Generate configuration

## Actual behavior:
1. The peers `AllowedIPs` are set to the peer IPs, not the host IPs
2. The peer `Address` in the `[Interface]` section is not set at all

## Expected behavior:
1. `AllowedIPs` should contain the host (openwrt) IPs set in the interface configuration
2. `Address` should be set to the peer IPs

## Additional Information:
OpenWrt version information from system `/etc/openwrt_release`

```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='22.03.0-rc4'
DISTRIB_REVISION='r19426-2b1941e47d'
DISTRIB_TARGET='ramips/mt7621'
DISTRIB_ARCH='mipsel_24kc'
DISTRIB_DESCRIPTION='OpenWrt 22.03.0-rc4 r19426-2b1941e47d'
DISTRIB_TAINTS=''
```

jow-, 2022-09-08:
Please provide the relevant UCI settings of /etc/config/network as well as the generated peer configuration and the peer configuration as it *should* look like. Your description is not clear enough to me.

felinira, 2022-09-13:
Those are the relevant settings including my LAN interface configuration to better understand where some of those IPs are coming from in my modified configuration. I can't say if it is a good default to include all lan IP ranges in the `AllowedIPs` or not, but at least the interface IP of wg0 (192.168.43.1) should be included to be able to reach the router from the VPN.

**UCI settings**

```
config globals 'globals'
	option ula_prefix 'fd97:a231:665f::/48'

config interface 'lan'
	option proto 'static'
	option netmask '255.255.255.0'
	option ip6assign '60'
	list dns '192.168.42.1'
	option device 'br-lan.1'
	option ipaddr '192.168.42.1'

config interface 'wg0'
	option proto 'wireguard'
	option private_key 'xxx'
	list addresses '192.168.43.1'
	option listen_port '51820'
	option ip6assign '64'
	list ip6class 'local'
	option ip6hint '43'

config wireguard_wg0
	option description 'peer1'
	option public_key 'xxx'
	option private_key 'xxx'
	option preshared_key 'xxx'
	option persistent_keepalive '25'
	option route_allowed_ips '1'
	list allowed_ips '192.168.43.2/32'
	list allowed_ips 'fd97:a231:665f:43::2/128'
```

**Generated configuration that doesn't connect correctly for me**

```
[Interface]
PrivateKey = xxx
# ListenPort not defined

[Peer]
PublicKey = xxx
PresharedKey = xxx
AllowedIPs = 192.168.43.2/32, fd97:a231:665f:43::2/128
Endpoint = my.ddns:51820
PersistentKeepAlive = 25
```

**Modified configuration that works for me**
```
[Interface]
PrivateKey = xxx
Address = 192.168.43.2/32,fd97:a231:665f:43::2/32

[Peer]
PublicKey = xxx
PresharedKey = xxx
AllowedIPs = 192.168.43.1,192.168.42.0/24,fd97:a231:665f::/48
Endpoint = my.ddns:51820
PersistentKeepAlive = 25
```

jow-, 2022-09-13:
Is the `Address = ` part actually needed or is adding the `192.168.43.1` entry to AllowedIPs enough? By default, if you don't specfiy any allowed IPs for the peer, the generated client config will contain AllowedIPs = 0.0.0.0/0 which works for the majority of use cases I suppose (simple road warrior scenario). If you override that by specifying allowed IPs yourself, you would need to manually include the IP of the OpenWrt wg0 interface too, I suppose that part could be automated.

However using given allowed IPs of a peer section ad adding them verbatim as Address entries for the peer does not seem right to me. Not all allowed IP entries are local peer interface IPs, there might also be address ranges referring to remote subnets available on the remote end of the tunnel etc.

Could you confirm that the following configuration also works for you?

```
[Interface]
PrivateKey = xxx
# ListenPort not defined

[Peer]
PublicKey = xxx
PresharedKey = xxx
AllowedIPs = 192.168.43.1, 192.168.42.0/24, fd97:a231:665f::/48
Endpoint = my.ddns:51820
PersistentKeepAlive = 25
```

felinira, 2022-09-13:
> However using given allowed IPs of a peer section and adding them verbatim as Address entries for the peer does not seem right to me

True. That was an oversight.

> Could you confirm that the following configuration also works for you?

This doesn't work for me (at least I can't reach any peers or they can't reach me). I don't know if there is some server side configuration that makes this work, but for me it doesn't. I don't really understand how it is supposed to work either - if you don't set an IP address to the interface, how is it supposed to communicate? Does this require an active DHCP server?

commit 73aca68b (2022-11-09): luci-proto-wireguard: fix generated AllowedIPs

This corrects the option `AllowedIPs` in generated peer configurations,
and allows to customize it via a dropdown list.

Fixes: #5956

[correct fixes tag, slightly adjust option description]
commit 14403feb (2022-11-23): luci-proto-wireguard: fix generated AllowedIPs

This corrects the option `AllowedIPs` in generated peer configurations,
and allows to customize it via a dropdown list.

Fixes: #5956

[correct fixes tag, slightly adjust option description]

(cherry picked from commit 73aca68b9e35857ec155f33ca1b1fb5358d69e52)
closed 2022-11-09 by jow-