openwrt/luci #5959: luci-app-upnp appears incompatible with OpenWrt 22.03

wagnerck, 2022-09-06:
## Steps to reproduce:

1) Upgrade to OpenWrt 22.03
2) Install the `luci-app-upnp` package and its dependencies from the System -> Software panel
3) Enable the service and configure it

## Actual behavior:

Applications on the local network are unable to create a uPnP redirect. No redirects show up on the Services -> UPnP panel, and applications like Plex and Transmission show network ports are closed. uPnP services are inaccessible from outside the local network.

## Expected behavior:

It should functional largely the same as it did under OpenWrt 21.02.

## Additional Information:
OpenWrt version information from system `/etc/openwrt_release`

```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='22.03.0'
DISTRIB_REVISION='r19685-512e76967f'
DISTRIB_TARGET='mvebu/cortexa9'
DISTRIB_ARCH='arm_cortex-a9_vfpv3-d16'
DISTRIB_DESCRIPTION='OpenWrt 22.03.0 r19685-512e76967f'
DISTRIB_TAINTS=''
```

This is apparently a known problem (per https://forum.openwrt.org/t/22-03-0-netgear-r7800-upnp-not-working/136306) but I didn't seen any GitHub issues opened up about it. Here's the package installation confirmation:

[image]

And the actual results of the install:

```
Installing luci-app-upnp (git-20.332.74200-03c77da) to root...
Downloading https://downloads.openwrt.org/releases/22.03.0/packages/arm_cortex-a9_vfpv3-d16/luci/luci-app-upnp_git-20.332.74200-03c77da_all.ipk
Installing libcap-ng (0.8.2-1) to root...
Downloading https://downloads.openwrt.org/releases/22.03.0/packages/arm_cortex-a9_vfpv3-d16/packages/libcap-ng_0.8.2-1_arm_cortex-a9_vfpv3-d16.ipk
Installing libuuid1 (2.37.3-3) to root...
Downloading https://downloads.openwrt.org/releases/22.03.0/packages/arm_cortex-a9_vfpv3-d16/base/libuuid1_2.37.3-3_arm_cortex-a9_vfpv3-d16.ipk
Installing miniupnpd-nftables (2.2.3-4) to root...
Downloading https://downloads.openwrt.org/releases/22.03.0/packages/arm_cortex-a9_vfpv3-d16/packages/miniupnpd-nftables_2.2.3-4_arm_cortex-a9_vfpv3-d16.ipk
Configuring libcap-ng.
Configuring libuuid1.
Configuring miniupnpd-nftables.
Configuring luci-app-upnp.
```

These are the contents of `/etc/config/upnpd` (which I deleted prior to reinstalling the packages to make sure it's a clean install):

```
config upnpd 'config'
        option download '1024'
        option upload '512'
        option internal_iface 'lan'
        option port '5000'
        option upnp_lease_file '/var/run/miniupnpd.leases'
        option igdv1 '1'
        option enabled '1'
        option uuid '94c30bb7-6571-4ff7-adb3-5c79eda38efc'

config perm_rule
        option action 'allow'
        option ext_ports '1024-65535'
        option int_addr '0.0.0.0/0'
        option int_ports '1024-65535'
        option comment 'Allow high ports'

config perm_rule
        option action 'deny'
        option ext_ports '0-65535'
        option int_addr '0.0.0.0/0'
        option int_ports '0-65535'
        option comment 'Default deny'
```

Oddly, `/var/run/miniupnpd.leases` shows the appropriate uPnP redirects:

```
TCP:19787:192.168.0.184:32400:1663085781:Plex Media Server
TCP:51413:192.168.0.184:51413:1663085782:Transmission at 51413
UDP:51413:192.168.0.184:51413:1663085782:Transmission at 51413
```

but nothing shows up in the Services -> UPnP panel:

[image]

It's not clear if this is a problem with `luci-app-upnp` or one of its dependencies, but I think this is the right place to start with this issue. If it needs to be moved elsewhere, that's fine.

jow-, 2022-09-07:
LuCI fix: https://github.com/openwrt/luci/pull/5961
Package fix: https://github.com/openwrt/packages/pull/19320

wagnerck, 2022-09-07:
Thank you for getting those PRs going so quickly. I presume these fixes be available through the package system once the PRs are merged, and will not require waiting for a OpenWRT update?

jow-, 2022-09-07:
Correct.

elliotclee, 2022-09-07:
This also should fix #5678, yay.

wagnerck, 2022-09-08:
The System -> Software panel now lists `luci-app-upnp` as version `git-22.250.64651-a61c041` and `miniupnpd-nftables` as version `2022-08-31-68c8ec50-1`. (The previous versions were `git-20.332.74200-03c77da` and `2.2.3-4`.) The dates and commit IDs don't seem to match the GitHub PRs that Jow did, but I suspect they're being pulled from a different system that the Software system is talking to?

In any case, I just tested these new versions, and uPnP seems to be working correctly now. The Services -> UPnP panel shows the redirects:

[image]

and the services are visible outside the local network. 

For other folks who want to test this, I recommend uninstalling the `luci-app-upnp` package and dependencies, deleting `/etc/config/upnpd` and `/var/run/miniupnpd.leases`, and then reinstalling the package. This will ensure a clean install without any leftover bits.

I'll leave it up to Jow or other OpenWRT folks to close out this issue, and thanks again for the quick response.

jow-, 2022-09-08:
Fixed by https://github.com/openwrt/luci/pull/5961 and https://github.com/openwrt/packages/pull/19320

closed 2022-09-08 by jow-

dropwhile, 2022-09-08:
I do not see the package built for x86_64 yet, but I do see it build for other architectures. 
Is x86_64 just slower to get packages updates?

hnyman, 2022-09-09:
> I do not see the package built for x86_64 yet, but I do see it built for other architectures.
> Is x86_64 just slower to get packages updates?

The packages are now there...
`luci-app-upnp_git-22.250.64651-a61c041_all.ipk | 5.0 KB | Fri Sep  9 00:11:44 2022`

Buildbot builds packages for all ~30 targets in round-robin fashion. It is pure luck in which order your target is in the queue. (It takes 6-9 hours per target, and currently there seem to be 8 concurrent builds...
See https://buildbot.openwrt.org/openwrt-22.03/packages/#/