openwrt/luci #5962: luci - Chrome 105 and luci compatibility issues

MrBridgeSix, 2022-09-06:
When changing settings using the luci interface with Chrome 105 as the browser there's a lot of weirdness now, sometimes luci throws "XHR request aborted by browser" errors and often it fails to refresh the page after a setting change.

Firefox is unaffected.

Official OpenWRT 22.03.0 build for the AX6S target.

jow-, 2022-09-07:
A quick test with "Version 105.0.5195.102 (Official Build) (64-bit)" on Debian shows no unexpected issues. I'm afraid without more detailed information I can't help. Maybe check your browser debug console for clues. Also try HTTP access instead of HTTPS and see if the issue persists there.

MrBridgeSix, 2022-09-07:
[image]

Seems to only happen with HTTPS access and only after the page has been open for a bit (maybe past the 20s keep alive from uhhtpd?).

jow-, 2022-09-08:
It might be related to uhttpd or more specifically ustream-ssl on the OpenWrt side then. The issue does not appear to be due to the LuCI ui itself. I suppose SSL related changes in Chrome 105 and derivates now trigger issues in the uhttpd HTTPS support

ynezz, 2022-09-08:
I've here `Chromium Version 106.0.5249.21 (Official Build) snap (64-bit)` and 22.03 on ZyXEL GS1900-8 (RTL8380, MIPS 4KEc) and I'm not able to reproduce this.

kien-truong, 2022-09-19:
I have been seeing similar issue with Luci + HTTPS in version 22.03. 

It turns out that `uhttpd` seems to crash a lot in this version, which leads to errors such as `XHR request aborted by browser` and `Connection refused`.

ynezz, 2022-09-19:
> I have been seeing similar issue with Luci + HTTPS in version 22.03.

@kien-truong what board/target is that? Would it be possible to enable coredumps `ulimit -c unlimited; /etc/init.d/uhttpd restart` and provide coredumps from /tmp/*.core once it crashes?

kien-truong, 2022-09-19:
It's a Nano Pi R2S, I believe the CPU is a Rockchip RK3328.

I followed your instruction, but could not get any coredumps in `/tmp/`

kien-truong, 2022-09-19:
After reading `procd` docs, I manage to get the coredumps by editing the service file, and even a stack trace. Considering the function name, something is wrong with TLS 1.3

```
Core was generated by `/usr/sbin/uhttpd -f -h /www -r OpenWrt -x /cgi-bin -l /cgi-bin/luci -L /usr/lib'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x0000ffffab361bec in a_crash () at ./src/internal/atomic.h:250
250     ./src/internal/atomic.h: No such file or directory.
(gdb) bt
#0  0x0000ffffab361bec in a_crash () at ./src/internal/atomic.h:250
#1  __stack_chk_fail () at src/env/__stack_chk_fail.c:26
#2  0x0000ffffab0f1fac in DoTls13HandShakeMsgType (ssl=<optimized out>, input=<optimized out>, inOutIdx=0xffffaae3c4a4, type=<optimized out>, size=<optimized out>, totalSz=<optimized out>) at src/tls13.c:9589
#3  0x0000ffffab0fca70 in DoTls13HandShakeMsg (totalSz=<optimized out>, inOutIdx=0xffffaae3c4a4, input=<optimized out>, ssl=0xffffaae3c200) at src/tls13.c:9658
#4  ProcessReplyEx.constprop.0 (ssl=0xffffaae3c200, allowSocketErr=0) at src/internal.c:18770
#5  0x0213021302130213 in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
```

ynezz, 2022-09-19:
> at src/tls13.c:9658

I can see a lot of fixes introduced in wolfSSL 5.5.0-stable into this part, this version is already present in master, would it be possible to reproduce the issue with master snapshot builds? You would need to install LuCI manually as it's not available in the firmware images.

kien-truong, 2022-09-19:
I will try doing that during next weekend.

hauke, 2022-09-19:
This is maybe a security problem.

Could you create a pcap with wiresahrk or tcpdump of the ssl communication. Probably chrome does something strange in the handshake.

kien-truong, 2022-09-21:
@hauke Here you go, there're 3 crashes in this dump, around these points

```
No           Time
2101	48.113583267
4842	216.646136468
10226	284.404956441
```
[uhttpd_crash.zip](https://github.com/openwrt/luci/files/9615717/uhttpd_crash.zip)

kien-truong, 2022-09-24:
No dice, I tried OpenWRT master branch, using wolfssl-5.5.0, and my uhttpd is still crashing with the same error

```
Core was generated by `/usr/sbin/uhttpd -f -h /www -r OpenWrt -x /cgi-bin -l /cgi-bin/luci -L /usr/lib'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x0000ffff8f0c72a0 in a_crash () at ./src/internal/atomic.h:250
250     ./src/internal/atomic.h: No such file or directory.
(gdb) bt
#0  0x0000ffff8f0c72a0 in a_crash () at ./src/internal/atomic.h:250
#1  __stack_chk_fail () at src/env/__stack_chk_fail.c:26
#2  0x0000ffff8ee76744 in DoTls13HandShakeMsgType (ssl=0xffff8ebec170, input=0xffff8ebd46e0 "\001", inOutIdx=<optimized out>, type=<optimized out>, size=<optimized out>, totalSz=<optimized out>) at src/tls13.c:9783
#3  0x0000ffff8ee1ce18 in DoTls13HandShakeMsg (totalSz=529, inOutIdx=0xffff8ebec414, input=0xffff8ebd46e0 "\001", ssl=0xffff8ebec170) at src/tls13.c:10074
#4  ProcessReplyEx (allowSocketErr=0, ssl=0xffff8ebec170) at src/internal.c:19337
#5  ProcessReply (ssl=0xffff8ebec170) at src/internal.c:18593
#6  0x0213021302130213 in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
```

ynezz, 2022-09-24:
> No dice, I tried OpenWRT master branch, using wolfssl-5.5.0, and my uhttpd is still crashing with the same error

Thanks a lot!

BTW do you've some exact steps I could use to reproduce this? I've tried to replay those TLS handshakes, no dice. I've tried to run 22.03.0 on mt7622/aarch64 target, used latest Edge/Chromium/Firefox from Windows 11 for more then 8 hours, no dice.

I've [reported it upstream](https://github.com/wolfSSL/wolfssl/issues/5629).

kien-truong, 2022-09-24:
There's no steps to reproduce I'm sorry, it seems to happen randomly but frequently while browsing the web UI.

ynezz, 2022-09-24:
> There's no steps to reproduce I'm sorry, it seems to happen randomly but frequently while browsing the web UI.

Are there any specific pages you're accessing? Are you doing some refresh, applying certain settings or such? How quickly are you able to reproduce it?

hauke, 2022-09-24:
I did a diff between a working and a failing ClientHello:
````
 Transport Layer Security
-    TLSv1.3 Record Layer: Handshake Protocol: Client Hello
+    TLSv1 Record Layer: Handshake Protocol: Client Hello
         Content Type: Handshake (22)
         Version: TLS 1.0 (0x0301)
         Length: 512
@@ -107,12 +107,14 @@
             Handshake Type: Client Hello (1)
             Length: 508
             Version: TLS 1.2 (0x0303)
-            Random: e166326df109e626b9f724ba7bf79097d3c79de45581ceaac636ee8b81af52f7
+            Random: ab1bda3384fb2375affd7eca766b5e3a9298c23d884c58c2c5cba12b2abbf457
+                GMT Unix Time: Dec 20, 2060 00:47:47.000000000 UTC
+                Random Bytes: 84fb2375affd7eca766b5e3a9298c23d884c58c2c5cba12b2abbf457
             Session ID Length: 32
-            Session ID: 642981579b9246f7abbae8a31a2414189d6bdfe359d66c4b3ff64fb8aef48eff
+            Session ID: 943180631de772a1e6f96d623d141b536c20abe8938c1217eb22e859da56014a
             Cipher Suites Length: 34
             Cipher Suites (17 suites)
-                Cipher Suite: Reserved (GREASE) (0x5a5a)
+                Cipher Suite: Reserved (GREASE) (0xbaba)
                 Cipher Suite: TLS_AES_128_GCM_SHA256 (0x1301)
                 Cipher Suite: TLS_AES_256_GCM_SHA384 (0x1302)
                 Cipher Suite: TLS_AES_256_GCM_SHA384 (0x1302)
@@ -134,7 +136,7 @@
                 Compression Method: null (0)
             Extensions Length: 401
             Extension: Reserved (GREASE) (len=0)
-                Type: Reserved (GREASE) (19018)
+                Type: Reserved (GREASE) (31354)
                 Length: 0
                 Data: <MISSING>
             Extension: server_name (len=22)
@@ -158,7 +160,7 @@
                 Length: 10
                 Supported Groups List Length: 8
                 Supported Groups (4 groups)
-                    Supported Group: Reserved (GREASE) (0x9a9a)
+                    Supported Group: Reserved (GREASE) (0xcaca)
                     Supported Group: x25519 (0x001d)
                     Supported Group: secp256r1 (0x0017)
                     Supported Group: secp384r1 (0x0018)
@@ -225,13 +227,13 @@
                 Key Share extension
                     Client Key Share Length: 41
                     Key Share Entry: Group: Reserved (GREASE), Key Exchange length: 1
-                        Group: Reserved (GREASE) (39578)
+                        Group: Reserved (GREASE) (51914)
                         Key Exchange Length: 1
                         Key Exchange: 00
                     Key Share Entry: Group: x25519, Key Exchange length: 32
                         Group: x25519 (29)
                         Key Exchange Length: 32
-                        Key Exchange: 3a5aa7bd7437bf8e1c36bd30013936bf3e7e02e1d52da5d0cafcd8529ca50c2b
+                        Key Exchange: c68d97081cfeffc2513e899cb62e365c7a20632394f3cad26063dd80d8150d0d
             Extension: psk_key_exchange_modes (len=2)
                 Type: psk_key_exchange_modes (45)
                 Length: 2
@@ -241,7 +243,7 @@
                 Type: supported_versions (43)
                 Length: 7
                 Supported Versions length: 6
-                Supported Version: Unknown (0x3a3a)
+                Supported Version: Unknown (0xeaea)
                 Supported Version: TLS 1.3 (0x0304)
                 Supported Version: TLS 1.2 (0x0303)
             Extension: compress_certificate (len=3)
@@ -254,7 +256,7 @@
                 Length: 5
                 Data: 0003026832
             Extension: Reserved (GREASE) (len=1)
-                Type: Reserved (GREASE) (31354)
+                Type: Reserved (GREASE) (27242)
                 Length: 1
                 Data: 00
             Extension: padding (len=196)
````
The plus is the broken one, minus is the working one.

The different GREASE extension should be fine, see https://www.rfc-editor.org/rfc/rfc8701.html
I do not understand why one is detected as TLSv1 and the other as TLSv1.3, this probably also changes the of decoding of the random value.

ynezz, 2022-09-24:
>  do not understand why one is detected as TLSv1 and the other as TLSv1.3

RFC states following:

> In order to maximize backward compatibility, a record containing an initial ClientHello SHOULD have version 0x0301 (reflecting TLS 1.0) and a record containing a second ClientHello or a ServerHello MUST have version 0x0303 (reflecting TLS 1.2).

Client sends `Version: TLS 1.2 (0x0303)` in both cases (IIRC?), so I've simply assumed, that it's just some wireshark related issue as wolfSSL then chokes in TLSv1.3 handling.

IIRC server is advertising TLS 1.2 as minimum version.

hauke, 2022-09-24:
I think wireshark does the detection of the TLS version when it gets the ServerHello. This is not send in the failing case and therefore it stays at TLS 1.0. The TLS version is then changes for the complete session. 

The first failing TLS ClientHello has a `pre_shared_key` TLS extension. 
````
            Extension: pre_shared_key (len=215)
                Type: pre_shared_key (41)
                Length: 215
                Pre-Shared Key extension
                    Identities Length: 178
                    PSK Identity (length: 172)
                        Identity Length: 172
                        Identity: 2293410d00b14ef0c1984b599e2616cab9abbaf3a4c7de1802c4ae8fe0a00b5e00000068…
                        Obfuscated Ticket Age: 1123217821
                    PSK Binders length: 33
                    PSK Binders
````
Some of the ClientHello messages with this extension where also passing. I do not know if this is broken.

ynezz, 2022-09-26:
> I've [reported it upstream](https://github.com/wolfSSL/wolfssl/issues/5629).

@kien-truong sorry to bother you more, but since you're so far the only one able to reproduce it, I would ask you for more help. We've been [asked by upstream](https://github.com/wolfSSL/wolfssl/issues/5629#issuecomment-1257232415) to check if the issue still exists in the latest wolfSSL master. I've prepared [wolfSSL](https://gitlab.com/ynezz/openwrt/-/commit/b54bb8377e8bc20d53b279138be43855d717dee8.patch) version bump if you would like to compile the image yourself and I've just also [build images/packages for you](https://gitlab.com/ynezz/openwrt/-/jobs/3082589717/artifacts/browse/bin/). I didn't runtime tested this version bump yet.

ynezz, 2022-09-28:
@kien-truong could you please test https://github.com/ynezz/openwrt/commit/b311208fc45c682c7d41942d73d3bd7c4467a357  and report back? I would like to give you street credits for this amazing help of yours, so if you're interested, could you give me your `Tested-by: Real Name <email>`?

closed 2022-09-29 by openwrt-bot