openwrt/luci #5963: luci-app-dawn: 500 Internal Server Error from time to time

MoetaYuko, 2022-09-07:
## Steps to reproduce:

1. go to: DAWN -> View Hearing Map

## Actual behavior:

**Sometimes** observe 500 Internal Server Error when rendering the list:
```
/usr/lib/lua/luci/template.lua:97: Failed to execute template 'cbi/map'.
A runtime error occurred: /usr/lib/lua/luci/template.lua:97: Failed to execute template '[string]'.
A runtime error occurred: [string "[string]"]:16: bad argument #2 to '?' (integer expected, got number)
stack traceback:
	[C]: in function 'error'
	/usr/lib/lua/luci/util.lua:49: in function 
	[string "[string]"]:16: in main chunk
stack traceback:
	[C]: in function 'error'
	/usr/lib/lua/luci/template.lua:97: in function 
	(tail call): ?
	...usr/lib/lua/luci/model/cbi/dawn/dawn_hearing_map.lua:8: in function 'render'
	/usr/lib/lua/luci/cbi.lua:266: in function 'render_children'
	[string "/usr/lib/lua/luci/view/cbi/map.htm"]:6: in main chunk
stack traceback:
	[C]: in function 'error'
	/usr/lib/lua/luci/template.lua:97: in function 
	(tail call): ?
	/usr/lib/lua/luci/cbi.lua:257: in function 'render'
	/usr/lib/lua/luci/cbi.lua:440: in function 'render'
	/usr/lib/lua/luci/dispatcher.lua:1483: in function '_cbi'
	/usr/lib/lua/luci/dispatcher.lua:1056: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:483: in function
```
[image]

However, this does not always happen. I have no idea how it gets triggered.

## Expected behavior:

Always rendering correctly

## Additional Information:
OpenWrt version information from system `/etc/openwrt_release`

```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='22.03.0'
DISTRIB_REVISION='r19685-512e76967f'
DISTRIB_TARGET='mediatek/mt7622'
DISTRIB_ARCH='aarch64_cortex-a53'
DISTRIB_DESCRIPTION='OpenWrt 22.03.0 r19685-512e76967f'
DISTRIB_TAINTS=''
```

loopy321, 2022-09-26:
getting the same thing on my system

```500 Internal Server Error
Sorry, the server encountered an unexpected error.

/usr/lib/lua/luci/template.lua:97: Failed to execute template 'cbi/map'.
A runtime error occurred: /usr/lib/lua/luci/template.lua:97: Failed to execute template '[string]'.
A runtime error occurred: [string "[string]"]:16: bad argument #2 to '?' (integer expected, got number)
stack traceback:
	[C]: in function 'error'
	/usr/lib/lua/luci/util.lua:49: in function 
	[string "[string]"]:16: in main chunk
stack traceback:
	[C]: in function 'error'
	/usr/lib/lua/luci/template.lua:97: in function 
	(tail call): ?
	...usr/lib/lua/luci/model/cbi/dawn/dawn_hearing_map.lua:8: in function 'render'
	/usr/lib/lua/luci/cbi.lua:266: in function 'render_children'
	[string "/usr/lib/lua/luci/view/cbi/map.htm"]:6: in main chunk
stack traceback:
	[C]: in function 'error'
	/usr/lib/lua/luci/template.lua:97: in function 
	(tail call): ?
	/usr/lib/lua/luci/cbi.lua:257: in function 'render'
	/usr/lib/lua/luci/cbi.lua:440: in function 'render'
	/usr/lib/lua/luci/dispatcher.lua:1483: in function '_cbi'
	/usr/lib/lua/luci/dispatcher.lua:1056: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:483: in function 
```

```
# cat /etc/openwrt_release
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='22.03.0'
DISTRIB_REVISION='r19685-512e76967f'
DISTRIB_TARGET='mediatek/mt7622'
DISTRIB_ARCH='aarch64_cortex-a53'
DISTRIB_DESCRIPTION='OpenWrt 22.03.0 r19685-512e76967f'
DISTRIB_TAINTS='' 

# opkg list-installed |grep dawn
dawn - 2022-07-24-9e8060ea-1
luci-app-dawn - git-21.282.29102-4a78122
```

hnyman, 2022-09-26:
cc @PolynomialDivision

PolynomialDivision, 2022-09-27:
Please test this: https://github.com/openwrt/luci/pull/5992

raenye, 2023-11-05:
Irrelevant due to https://github.com/openwrt/luci/pull/6656. Please close.

closed 2023-11-06 by MoetaYuko