openwrt/luci #5973: luci: Static Leases not pulled from all files

arrmo, 2022-09-15:
## Steps to reproduce:
1. create static DHCP leases - in [`/etc/dnsmasq.conf`](https://openwrt.org/docs/guide-user/base-system/dhcp.dnsmasq#configuration), or [`/etc/ethers`](https://openwrt.org/docs/guide-user/base-system/dhcp#static_leases) (as noted at the links provided ... and confirmed, these do work for DHCP).
2. go to: Network → DHCP and DNS → Static Leases
3. the static leases defined in these two files are not showing => only pulling from `/etc/config/dhcp` it seems?

## Actual behavior:
As above - the static leases are not shown in LuCI

## Expected behavior:
Expect that the static leases (in LuCI) are pulled from all three options.

## Additional Information:
OpenWrt version information from system `/etc/openwrt_release`
```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='22.03.0'
DISTRIB_REVISION='r19685-512e76967f'
DISTRIB_TARGET='mvebu/cortexa53'
DISTRIB_ARCH='aarch64_cortex-a53'
DISTRIB_DESCRIPTION='OpenWrt 22.03.0 r19685-512e76967f'
DISTRIB_TAINTS='no-all busybox'
```

hnyman, 2022-09-15:
> only pulling from /etc/config/dhcp it seems?

LuCI is mostly about configuring via "uci" with the files in /etc/config. The whole LuCI syntax is based on reading/manipulating uci config files.

LuCI provides support for some external upstream configuration files, but rarely for all features / optional files.

jow-, 2022-09-15:
Static leases outside of uci are unsupported.

closed 2022-09-15 by jow-

arrmo, 2022-09-15:
OK, understand - thanks. I made the (bad) assumption that all 3 sources were OK. Appreciate the feedback!

arrmo, 2022-09-15:
Sorry, hope you don't mind - but with my static leases all in the correct file, I'm getting an error in LuCI now 😞. I'm trying to dig into the code, but perhaps you know ... the Static Leases page, it seems to call an RPC (ubus?) - do you know what function it's calling?

This is the error I receive,
[image]

FYI, I think it's caused by the number of static leases I have (~ 130), if I try just a few it works fine.

Thanks!!