openwrt/luci #5975: luci-app-firewall: The language in this module is not changed to Polish, although it has been set and downloaded

Zaptyp, 2022-09-15:
## Steps to reproduce:

1. Go to: System -> Software 
2. Download and install the package: luci-i18n-firewall-pl
3. After installation go to: System -> System and go to language settings options and changed to Polish
4. After launguage change go to: Sieć -> Zapora Sieciowa 
5. And you now can see something like this:
[image]

## Actual behavior:
The language in this module is not changed to Polish, although it has been set and downloaded

## Expected behavior:
The language should change to Polish when it is set, but this does not happen.

## Additional Information:
OpenWrt version information from system `/etc/openwrt_release`
```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='22.03.0'
DISTRIB_REVISION='r19685-512e76967f'
DISTRIB_TARGET='ramips/mt7621'
DISTRIB_ARCH='mipsel_24kc'
DISTRIB_DESCRIPTION='OpenWrt 22.03.0 r19685-512e76967f'
DISTRIB_TAINTS=''
```

feckert, 2022-09-16:
I cannot confirm this. I use the German translation and it works there.
Have you ever logged on and off again?

hnyman, 2022-09-16:
I would suggest rebooting the router, if you have not done that.

jow-, 2022-10-13:
Cannot reproduce it either.

closed 2022-10-13 by jow-