openwrt/luci #5978: luci-app-sqm

JCasanova2022, 2022-09-18:
Buenas noches Estimados,
Tengo un inconveniente con un router Tplink Archer C60 V3, tiene instalada la versión, OpenWrt 22.03.0 r19685-512e76967f, pero no me permite instalar la función de QSM, si me pueden orientar como se hace, cuando entro en sistemas>software>Update list y sale el siguiente error

Downloading https://downloads.openwrt.org/releases/22.03.0/targets/ath79/generic/packages/Packages.gz
*** Failed to download the package list from https://downloads.openwrt.org/releases/22.03.0/targets/ath79/generic/packages/Packages.gz

Downloading https://downloads.openwrt.org/releases/22.03.0/packages/mips_24kc/base/Packages.gz
*** Failed to download the package list from https://downloads.openwrt.org/releases/22.03.0/packages/mips_24kc/base/Packages.gz

Downloading https://downloads.openwrt.org/releases/22.03.0/packages/mips_24kc/luci/Packages.gz
*** Failed to download the package list from https://downloads.openwrt.org/releases/22.03.0/packages/mips_24kc/luci/Packages.gz

Downloading https://downloads.openwrt.org/releases/22.03.0/packages/mips_24kc/packages/Packages.gz
*** Failed to download the package list from https://downloads.openwrt.org/releases/22.03.0/packages/mips_24kc/packages/Packages.gz

Downloading https://downloads.openwrt.org/releases/22.03.0/packages/mips_24kc/routing/Packages.gz
*** Failed to download the package list from https://downloads.openwrt.org/releases/22.03.0/packages/mips_24kc/routing/Packages.gz

Downloading https://downloads.openwrt.org/releases/22.03.0/packages/mips_24kc/telephony/Packages.gz
*** Failed to download the package list from https://downloads.openwrt.org/releases/22.03.0/packages/mips_24kc/telephony/Packages.gz
Errors
Failed to send request: Operation not permitted
Failed to send request: Operation not permitted
Failed to send request: Operation not permitted
Failed to send request: Operation not permitted
Failed to send request: Operation not permitted
Failed to send request: Operation not permitted
Collected errors:
 * opkg_download: Failed to download https://downloads.openwrt.org/releases/22.03.0/targets/ath79/generic/packages/Packages.gz, wget returned 4.
 * opkg_download: Check your network settings and connectivity.

 * opkg_download: Failed to download https://downloads.openwrt.org/releases/22.03.0/packages/mips_24kc/base/Packages.gz, wget returned 4.
 * opkg_download: Check your network settings and connectivity.

 * opkg_download: Failed to download https://downloads.openwrt.org/releases/22.03.0/packages/mips_24kc/luci/Packages.gz, wget returned 4.
 * opkg_download: Check your network settings and connectivity.

 * opkg_download: Failed to download https://downloads.openwrt.org/releases/22.03.0/packages/mips_24kc/packages/Packages.gz, wget returned 4.
 * opkg_download: Check your network settings and connectivity.

 * opkg_download: Failed to download https://downloads.openwrt.org/releases/22.03.0/packages/mips_24kc/routing/Packages.gz, wget returned 4.
 * opkg_download: Check your network settings and connectivity.

 * opkg_download: Failed to download https://downloads.openwrt.org/releases/22.03.0/packages/mips_24kc/telephony/Packages.gz, wget returned 4.
 * opkg_download: Check your network settings and connectivity.
The opkg update command failed with code 6.

Luego, No information available, si hay alguna manera de instalar este paquete por favor orientarme, ya que lo necesito de manera urgente

hnyman, 2022-09-18:
> wget returned 4.
>
> opkg_download: Check your network settings and connectivity.

You have somehow invalid basic network settings.
Not a LuCI problem.

closed 2022-09-18 by hnyman

JCasanova2022, 2022-09-18:
Hay alguna forma de bajar ese paquete adicional y instalarlo manual en el router?

hnyman, 2022-09-18:
Valitan, enpä ymmärrä espanjaa.

English, please.

JCasanova2022, 2022-09-18:
Is there a way to download that additional package and install it manually on the router?
Since I tried several ways and it does not allow me to add the sqm.

hnyman, 2022-09-18:
The error message  "wget returned 4., opkg_download: Check your network settings and connectivity." means that your network settings in the router are wrong. It can't use wget/curl/whatever to download the packages.

You need to fix the network settings. Typical error on a dumb secondary router is that you have forgotten to specify the gateway (main router lan address) and/or DNS server (main router lan IP address).

Alternatively, use e.g. USB stick to transfer the downloaded packages from your PC.