openwrt/luci #5984: <luci-app-statistics>,luci/admin/statistics/graphs,No RRD data found

chuyueye, 2022-09-21:
# Issue template:
1. Please make sure that the issue subject starts with `<package-name>:`
2. Issues related to end of life (EOL) releases are not supported or maintained and will be closed
3. Remove lines from top till here and fill in the following questions
***

## Steps to reproduce:
**Example: Replace the following lines and remove this**

1. go to: Network → Interfaces → LAN (an interface with a static address) → DHCP server tab → IPv6 RA settings tab
2. set RA MTU and RA Hop Limit to arbitrary non-default value
3. Save & Apply
4. go to: IPv6 RA settings tab again

## Actual behavior:
**Example: Replace the following lines and remove this**

1. option ra_mtu '1500' is set correctly in /etc/config/dhcp
2. In LuCI though, RA MTU and RA Hop Limit are still displaying the default values in light gray font color.

## Expected behavior:
**Example: Replace the following lines and remove this**

```
RA MTU and RA Hop Limit are displaying the correct values found in /etc/config/dhcp in black font color.
```

## Additional Information:
OpenWrt version information from system `/etc/openwrt_release`

**Example: Replace the following lines and remove this**
```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='21.02.0-rc4'
DISTRIB_REVISION='r16256-2d5ee43dc6'
DISTRIB_TARGET='x86/64'
DISTRIB_ARCH='x86_64'
DISTRIB_DESCRIPTION='OpenWrt 21.02.0-rc4 r16256-2d5ee43dc6'
DISTRIB_TAINTS=''
```

No RRD data found
There is no RRD data available yet to render graphs.

You need to configure collectd to gather data into .rrd files.

hnyman, 2022-09-21:
Sounds like you have not yet configured LuCI statistics properly. Either you have a non-standard data location configured, or something else is wrong in your settings.

(The statistics package itself works normally ok in 21.02, 22.03 and master.)

Note that you are using an ancient the pre-release version of the previous stable release.
You should use either 21.02.3 or 22.03.0.

closed 2022-09-21 by hnyman

chuyueye, 2022-10-11:
OpenWrt 22.03.0, r19685-512e76967f

i am using this version,and the problem still going on

chuyueye, 2022-10-11:
of course i have configured ,and have the collectd.conf file

chuyueye, 2022-10-11:
BaseDir "/var/run/collectd"
Include "/etc/collectd/conf.d/*.conf"
PIDFile "/var/run/collectd.pid"
PluginDir "/usr/lib/collectd"
TypesDB "/usr/share/collectd/types.db"
Interval 30
ReadThreads 2

LoadPlugin iwinfo

LoadPlugin ping
<Plugin ping>
	TTL 127
	Interval 30
	AddressFamily any
	Host "223.5.5.5"
	Host "8.8.8.8"
	Host "1.1.1.1"
	Host "127.0.0.1"
	Host "www.bilibili.com"
	Host "www.github.com"
</Plugin>

LoadPlugin memory
<Plugin memory>
	ValuesPercentage false
	ValuesAbsolute true
</Plugin>

LoadPlugin cpu
<Plugin cpu>
	ValuesPercentage true
	ReportByCpu true
	ReportByState true
</Plugin>

LoadPlugin rrdtool
<Plugin rrdtool>
	DataDir "/tmp/rrd"
	RRARows 288
	RRASingle true
	RRATimespan 3600
	RRATimespan 86400
	RRATimespan 604800
	RRATimespan 2678400
	RRATimespan 31622400
</Plugin>

LoadPlugin dns
<Plugin dns>
	IgnoreSource "127.0.0.1"
</Plugin>

LoadPlugin load

LoadPlugin interface
<Plugin interface>
	Interface "br-lan"
	Interface lan1
	Interface lan2
	Interface lan3
	Interface lan4
</Plugin>

zeitisen, 2023-05-18:
I have the same problem with OpenWrt 22.03.3 r20028-43d71ad93e

I checked the installation. I checked the configuration. I checked the documentation.
I cannot find any mistake.
@chuyueye  If you have found the solution, could you please leave this here?

zeitisen, 2023-05-18:
Here my solution:
In the config file there were entries of plugins which were not installed.
You can not see these entries in Luci.
You can only see these errors in systemlog after restarting collectd.

After installing the missing plugins collectd is running and is displaying  graphs.
Could it be possible, that there were plugins as a dependency and always installed with the packet and these dependencies changed, but the configfile not?