openwrt/luci #5987: luci-app-ddns: error when trying to create a service

castillofrancodamian, 2022-09-23:
# Issue template:
1. Please make sure that the issue subject starts with `<package-name>:`
2. Issues related to end of life (EOL) releases are not supported or maintained and will be closed
3. Remove lines from top till here and fill in the following questions
***

## Steps to reproduce:
**Example: Replace the following lines and remove this**

1. go to: Services → Dynamic DNS→ Add new services...
2. In the pop-up enter any name and choose any provider.
3. Click on "Create service"

Actual behavior:
Error that prevents the creation of a service.

Expected behavior:
Create a DDNS service and make the corresponding configuration.

Additional Information:
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='SNAPSHOT'
DISTRIB_REVISION='r20728-a24e07700d'
DISTRIB_TARGET='ath79/generic'
DISTRIB_ARCH='mips_24kc'
DISTRIB_DESCRIPTION='OpenWrt SNAPSHOT r20728-a24e07700d'
DISTRIB_TAINTS='no-all'

csjtl, 2022-10-01:
me too

IP don't Update

 214812       : ************ ************** ************** **************
 214812  note : PID '3375' started at 2022-10-01 21:48
 214812       : ddns version  : 2.8.2-27
 214812       : uci configuration:
ddns.proxypoolCheck.check_unit='minutes'
ddns.proxypoolCheck.domain='ddns.xxxxxx.com'
ddns.proxypoolCheck.enabled='1'
ddns.proxypoolCheck.force_unit='minutes'
ddns.proxypoolCheck.interface='wan'
ddns.proxypoolCheck.ip_network='wan'
ddns.proxypoolCheck.ip_source='network'
ddns.proxypoolCheck.lookup_host='ddns.xxxxxx.com'
ddns.proxypoolCheck.password='***PW***'
ddns.proxypoolCheck.retry_unit='seconds'
ddns.proxypoolCheck.service_name='google.com'
ddns.proxypoolCheck.use_ipv6='0'
ddns.proxypoolCheck.use_syslog='2'
ddns.proxypoolCheck.username='xxxxxxxxxxxxxx'
ddns.proxypoolCheck=service
 214813       : verbose mode  : 0 - run normal, NO console output
 214813       : check interval: 600 seconds
 214813       : force interval: 4320 seconds
 214813       : retry interval: 60 seconds
 214813       : retry max count : 0 times
 214813       : No old process
 214813       : last update: never
 214813       : Detect registered/public IP
 214813       : #> /usr/bin/nslookup ddns.xxxxxx.com  >/var/run/ddns/proxypoolCheck.dat 2>/var/run/ddns/proxypoolCheck.err
 214813       : Registered IP 'xx.xx.215.41' detected
 214813  info : Starting main loop at 2022-10-01 21:48
 214814       : Detect current IP on 'network'
 214814       : Current IP 'xx.xx.92.242' detected on network 'wan'
 214814       : Update needed - L: 'xx.xx.92.242' <> R: 'xx.xx.215.41'
 214814       : Force communication via device 'pppoe-wan'
 214814       : #> /usr/bin/curl -RsS -o /var/run/ddns/proxypoolCheck.dat --stderr /var/run/ddns/proxypoolCheck.err --interface pppoe-wan --noproxy '*' 'https://xxxxxxxxxxxxxx:***PW***@domains.google.com/nic/update?hostname=ddns.xxxxxx.com&myip=xx.xx.92.242'
 215112 ERROR : cURL Error: '28'
 215112       : curl: (28) Failed to connect to domains.google.com port 443 after 132154 ms: Operation timed out
 215112  WARN : Transfer failed - retry 1/0 in 60 seconds
 215212       : #> /usr/bin/curl -RsS -o /var/run/ddns/proxypoolCheck.dat --stderr /var/run/ddns/proxypoolCheck.err --interface pppoe-wan --noproxy '*' 'https://xxxxxxxxxxxxxx:***PW***@domains.google.com/nic/update?hostname=ddns.xxxxxx.com&myip=xx.xx.92.242'
 215421 ERROR : cURL Error: '28'
 215421       : curl: (28) Failed to connect to domains.google.com port 443 after 129270 ms: Operation timed out
 215421  WARN : Transfer failed - retry 2/0 in 60 seconds
 215521       : #> /usr/bin/curl -RsS -o /var/run/ddns/proxypoolCheck.dat --stderr /var/run/ddns/proxypoolCheck.err --interface pppoe-wan --noproxy '*' 'https://xxxxxxxxxxxxxx:***PW***@domains.google.com/nic/update?hostname=ddns.xxxxxx.com&myip=xx.xx.92.242'
 215731 ERROR : cURL Error: '28'
 215731       : curl: (28) Failed to connect to domains.google.com port 443 after 129281 ms: Operation timed out
 215731  WARN : Transfer failed - retry 3/0 in 60 seconds
 215831       : #> /usr/bin/curl -RsS -o /var/run/ddns/proxypoolCheck.dat --stderr /var/run/ddns/proxypoolCheck.err --interface pppoe-wan --noproxy '*' 'https://xxxxxxxxxxxxxx:***PW***@domains.google.com/nic/update?hostname=ddns.xxxxxx.com&myip=xx.xx.92.242'

adelton, 2025-01-11:
On OpenWrt 23.05.5 (r24106-10cc5fcd00) with luci-app-ddns git-24.364.71483-75d2b84, clicking the `Create service` button seemingly does nothing, the dialog is still there. But the service actually got created and there is an error message

> TypeError
> Option "Name" contains an invalid input value. The service name is already used

On OpenWrt 24.10.0-rc5 (r28304-6dacba30a7) with luci-app-ddns 25.009.71487~a88d4af there is no longer any issue creating the service.

systemcrash, 2025-01-13:
This was likely fixed in the bug-crushing set committed recently. For a while it was impossible even to define a new service from the GUI (which went unnoticed - evidently two default services seemed to suffice)