openwrt/luci #5993: OpenVPN and Wireguard does not support multiple VPN servers

Che15ea, 2022-09-27:
Hi
I am using wireguard and VPN on luci. I always add dozens of servers to create some randomness and backup when one of them is down. However, both of them always stick to the server on the bottom of the list of servers in my config, instead of randomly selecting a server or jumping to another server when the bottom server is down.
Please check with the issue.
Thank you!

stangri, 2022-09-28:
Are you using `remote-random` option for OpenVPN?

I don't think wireguard supports selecting a server randomly, but there are helpful suggestions for achieving random wg connections in [this OpenWrt forum thread](https://forum.openwrt.org/t/how-to-choose-a-random-wireguard-server/14750).

adelton, 2025-01-11:
@Che15ea Is it fair to say that both OpenVPN and WireGuard support multiple server in OpenWrt and you are looking for the randomization? Can you answer the question about `remote-random` and also whether you tried the linked approach for WireGuard?

Having that information would increase the chance that someone might be interested in implementing UI in LuCI for the functionality.