openwrt/luci #5997: LuCI parses wrongly version-specific dependencies

hnyman, 2022-09-29:
LuCI master parses wrongly at least some package dependencies with a version specification.
First seen on forum discussion, then reproduced with up-to-date master
(LuCI Master git-22.260.19132-34dd31a / OpenWrt SNAPSHOT r20779-c5e167e0d6)

nginx-ssl-util is shown twice in the dependency tree.
The first line with a version requirement fails, while the second line without the version req shows the package available.
The end result is that LuCI claims that it can't find the package.

[image]

SSH console opkg shows the package:

```
root@router1:~# opkg list | grep nginx-ssl
nginx-ssl - 1.21.3-2 - nginx is an HTTP and reverse proxy server, as well as a mail proxy server, written by Igor Sysoev. This variant is compiled with SSL support enabled. To enable additional module select them in the nginx default configuration menu.
nginx-ssl-util - 1.6-14 - Utility that builds dynamically LAN listen directives for Nginx. Furthermore, it manages SSL directives for its server parts and can create corresponding (self-signed) certificates. It uses the PCRE library for performance.
nginx-ssl-util-nopcre - 1.6-14 - Utility that builds dynamically LAN listen directives for Nginx. Furthermore, it manages SSL directives for its server parts and can create corresponding (self-signed) certificates. It uses the standard regex library of C++.
uwsgi - 2.0.20-1 - The uWSGI project (to be used together with nginx or nginx-ssl).
...
root@router1:~# opkg install --noaction nginx-ssl
Installing nginx-ssl (1.21.3-2) to root...
Downloading https://downloads.openwrt.org/snapshots/packages/arm_cortex-a15_neon-vfpv4/packages/nginx-ssl_1.21.3-2_arm_cortex-a15_neon-vfpv4.ipk
Installing libstdcpp6 (11.3.0-4) to root...
Downloading https://downloads.openwrt.org/snapshots/targets/ipq806x/generic/packages/libstdcpp6_11.3.0-4_arm_cortex-a15_neon-vfpv4.ipk
Installing nginx-util (1.6-14) to root...
Downloading https://downloads.openwrt.org/snapshots/packages/arm_cortex-a15_neon-vfpv4/packages/nginx-util_1.6-14_all.ipk
Installing nginx-ssl-util (1.6-14) to root...
Downloading https://downloads.openwrt.org/snapshots/packages/arm_cortex-a15_neon-vfpv4/packages/nginx-ssl-util_1.6-14_arm_cortex-a15_neon-vfpv4.ipk
```

Possibly related to  43cea79a8 and ac47bc9dd

Forum: 
https://forum.openwrt.org/t/cant-install-nginx-ssl-on-master-missing-package-luci-bug/138328

jow-, 2022-09-30:
The upstream metadata looks broken. The depends line currently reads:

`Depends: nginx-ssl-util (>=1.5-1) (<2), libc, libopenssl1.1, libpthread, libpcre, nginx-ssl-util, zlib, liblua5.1.5, libxml2, libubus20220601, libblobmsg-json20220515, libjson-c5`

Note the superfluous `(<2)` after the `>=1.5-1` constrain. LuCI currently parses this as:
 [`nginx-ssl-util (>=1.5-1)`] [`<`] [`2`] instead of [`nginx-ssl-util`] [`>=`] [`1.5-1`].

I can fix the parsing to be more forgiving in this case.

closed 2022-09-30 by jow-

hnyman, 2022-09-30:
Thanks fo figuring it out.
I wonder if that is a recent change on package metadata reading/parsing/compilation, as this has not surfaced earlier.

commit 84ba6a5c (2022-09-30): luci-app-opkg: relax dependency parsing

Align depdendency parse logic with opkg to properly process certain
malformed dependency specifications. In particular do not require a
depdendency specification to end after the closing paren of a version
constraint and perform non-greedy matching in order to extract the
first version constraint only.

Fixes: #5997
Ref: https://forum.openwrt.org/t/x/138328
commit e01e38cf (2022-09-30): luci-app-opkg: relax dependency parsing

Align depdendency parse logic with opkg to properly process certain
malformed dependency specifications. In particular do not require a
depdendency specification to end after the closing paren of a version
constraint and perform non-greedy matching in order to extract the
first version constraint only.

Fixes: #5997
Ref: https://forum.openwrt.org/t/x/138328

(cherry picked from commit 84ba6a5c36072f74721d535104735757b005c8d6)