openwrt/luci #5998: luci-app-nlbwmon, exporting previous accounting periods broken.

4k3or3et, 2022-10-01:
Hi All

I am using Netlink Bandwidth Monitor for more than a month now so i have two accounting periods to choose from.

The problem i have is that i am not able to export previous accounting period. When i select previous period and perform export, it always exports current accounting period.

I am on 22.03.1.

Step to reproduce:

Select previous accounting period and try to export to CSV.

Expected behaviour:

Selected accounting period exported.

Acutal behaviour:

Current accounting period exported.

Can anyone confirm that this not only my issue?

ruralroots, 2022-10-15:
I have confirmed this behaviour, but it only occurs within `luci-app-nlbwmon`.

Selecting different accounting periods works using CLI notation ie.`nlbw -c csv -t 2021-07-17 `

jow-, 2022-10-15:
It simply never was implemented in the ui, it always exports the latest period. This isn't strictly a bug, just nbot implemented functionality and misleading ui interface.

ruralroots, 2022-10-15:
Makes sense. I've always used `nlbw` for that functionality. Thanks.