openwrt/luci #6007: Luci small visual/functional bug

Imwithstupid007, 2022-10-06:
When in Luci (snapshot OpenWrt SNAPSHOT r20813-3968529285 / LuCI Master git-22.260.19132-34dd31a; Belkin RT3200):

1. Goto "System-Software"
2. Select tab "Installed"
3. Scroll down to the bottom of the page
4. The next page button (with >>) doesn't work

[image]

It looks like its active but doesn't react to clicks.
Clicking the equivalent button on the top of the page works and shows the next page. After going back to the bottom of first page the button works. 
So it looks like it only malfunctions when the page is first opened.

The equivalent button on the tab "Available" seems to work fine. The tab "Updates" i didn't test.

hnyman, 2022-11-05:
Seems to work ok for me on
OpenWrt SNAPSHOT r21170-6198eb3e64 / LuCI Master git-22.299.72136-a98e2ea

Still broken for you? (with current LuCI)

Imwithstupid007, 2022-11-05:
Seems to work fine now (on  SNAPSHOT r21164).

closed 2022-11-05 by hnyman

Imwithstupid007, 2022-11-23:
Hmm.....
Seems to be broken again on OpenWrt SNAPSHOT r21335-e410833bdd / LuCI Master git-22.319.67745-a1ee78f.

hnyman, 2023-01-09:
Adding also here the explanation from #6156:

The bottom button wants to jump over page 2 and go directly to page 3 with packages 201-300. 
And if you have less than 200 packages (like @Imwithstupid007 screenshot above with only 166 packages), the button probably thinks that there is no possibility to change page and gets disabled.

commit ec3aac47 (2023-01-09): luci-app-opkg: fix bottom pager

Fixes: #6007, #6156