openwrt/luci #6029: luci-base: handleReset function has no functionality

feckert, 2022-10-17:
## Steps to reproduce:

1. Goto System → System (or any other page changing config option via uci)
2. Change an option in the system ( in my example Language)
3. Press save on this page (changes gets stage in 'Unsaved changes')
4. Press reset on this page
[image]

## Actual behavior

```
Nothing happens. The staged uci change in 'Unsaved changes' gets not deleted!
```

## Expected behavior:

```
The staged uci changes should get deleted on button press
```

@jow- There is something wrong with [this](https://github.com/openwrt/luci/blob/master/modules/luci-base/htdocs/luci-static/resources/luci.js#L2120) implementation. If, on the other hand, using the `Unsaved change` button and pressing `Revert` everything is working as expected, and the staged uci changes get delete.

## Additional Information:

```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='SNAPSHOT'
DISTRIB_REVISION='r20961-689cfaeb7c'
DISTRIB_TARGET='x86/64'
DISTRIB_ARCH='x86_64'
DISTRIB_DESCRIPTION='OpenWrt SNAPSHOT r20961-689cfaeb7c'
DISTRIB_TAINTS='no-all'
```

jow-, 2022-10-17:
The reset button is meant to reset the form to its initial state, discarding any values which have not been saved yet (it essentially is the same as realoding the page without submitting the changes). It is not meant to revert saved, staged changes.

closed 2022-10-17 by jow-

feckert, 2022-10-17:
I must confess, I have never used the button before.
Thanks for the explanation!