openwrt/luci #6044: 22.03.2 page load of most pages in LuCI gives traceback (ppp and pppoe)

systemcrash, 2022-10-20:
```logs
NetworkError: HTTP error 404 while loading class file "/luci-static/resources/protocol/ppp.js?v=git-22.288.45147-96ec0cd"
  at compileClass (http://192.0.2.1/luci-static/resources/luci.js?v=git-22.288.45147-96ec0cd:168:16)
  at promise callback*require (http://192.0.2.1/luci-static/resources/luci.js?v=git-22.288.45147-96ec0cd:184:112)
  at getProtocolHandlers/</< (http://192.0.2.1/luci-static/resources/network.js?v=git-22.288.45147-96ec0cd:7:162)
  at getProtocolHandlers/< (http://192.0.2.1/luci-static/resources/network.js?v=git-22.288.45147-96ec0cd:7:121)
  at promise callback*getProtocolHandlers (http://192.0.2.1/luci-static/resources/network.js?v=git-22.288.45147-96ec0cd:5:71)
  at initNetworkState (http://192.0.2.1/luci-static/resources/network.js?v=git-22.288.45147-96ec0cd:59:311)
  at getNetworks (http://192.0.2.1/luci-static/resources/network.js?v=git-22.288.45147-96ec0cd:135:54)
  at load (http://192.0.2.1/luci-static/resources/tools/widgets.js?v=git-22.288.45147-96ec0cd:34:369)
  at loadChildren (http://192.0.2.1/luci-static/resources/form.js?v=git-22.288.45147-96ec0cd:30:34)
  at load (http://192.0.2.1/luci-static/resources/form.js?v=git-22.288.45147-96ec0cd:67:17)
  at loadChildren (http://192.0.2.1/luci-static/resources/form.js?v=git-22.288.45147-96ec0cd:30:34)
  at load/< (http://192.0.2.1/luci-static/resources/form.js?v=git-22.288.45147-96ec0cd:45:32)
  at promise callback*load (http://192.0.2.1/luci-static/resources/form.js?v=git-22.288.45147-96ec0cd:44:542)
  at render (http://192.0.2.1/luci-static/resources/form.js?v=git-22.288.45147-96ec0cd:48:146)
  at render (http://192.0.2.1/luci-static/resources/view/system/system.js?v=git-22.288.45147-96ec0cd:47:10)
  at promise callback*__init__ (http://192.0.2.1/luci-static/resources/luci.js?v=git-22.288.45147-96ec0cd:147:232)
  at super (http://192.0.2.1/luci-static/resources/luci.js?v=git-22.288.45147-96ec0cd:22:42)
  at ClassConstructor (http://192.0.2.1/luci-static/resources/luci.js?v=git-22.288.45147-96ec0cd:12:11)
  at require/compileClass/< (http://192.0.2.1/luci-static/resources/luci.js?v=git-22.288.45147-96ec0cd:182:111)
  at promise callback*compileClass (http://192.0.2.1/luci-static/resources/luci.js?v=git-22.288.45147-96ec0cd:178:29)
  at promise callback*require (http://192.0.2.1/luci-static/resources/luci.js?v=git-22.288.45147-96ec0cd:184:112)
  at instantiateView (http://192.0.2.1/luci-static/resources/ui.js?v=git-22.288.45147-96ec0cd:352:252)
  at  (http://192.0.2.1/cgi-bin/luci/admin/system/system:54:7)
  at promise callback* (http://192.0.2.1/cgi-bin/luci/admin/system/system:53:19)
raise http://192.0.2.1/luci-static/resources/luci.js?v=git-22.288.45147-96ec0cd:155
    compileClass http://192.0.2.1/luci-static/resources/luci.js?v=git-22.288.45147-96ec0cd:168
    promise callback*require http://192.0.2.1/luci-static/resources/luci.js?v=git-22.288.45147-96ec0cd:184
    getProtocolHandlers http://192.0.2.1/luci-static/resources/network.js?v=git-22.288.45147-96ec0cd:7
    getProtocolHandlers http://192.0.2.1/luci-static/resources/network.js?v=git-22.288.45147-96ec0cd:7
    promise callback*getProtocolHandlers http://192.0.2.1/luci-static/resources/network.js?v=git-22.288.45147-96ec0cd:5
    initNetworkState http://192.0.2.1/luci-static/resources/network.js?v=git-22.288.45147-96ec0cd:59
    getNetworks http://192.0.2.1/luci-static/resources/network.js?v=git-22.288.45147-96ec0cd:135
    load http://192.0.2.1/luci-static/resources/tools/widgets.js?v=git-22.288.45147-96ec0cd:34
    loadChildren http://192.0.2.1/luci-static/resources/form.js?v=git-22.288.45147-96ec0cd:30
    load http://192.0.2.1/luci-static/resources/form.js?v=git-22.288.45147-96ec0cd:67
    loadChildren http://192.0.2.1/luci-static/resources/form.js?v=git-22.288.45147-96ec0cd:30
    load http://192.0.2.1/luci-static/resources/form.js?v=git-22.288.45147-96ec0cd:45
    promise callback*load http://192.0.2.1/luci-static/resources/form.js?v=git-22.288.45147-96ec0cd:44
    render http://192.0.2.1/luci-static/resources/form.js?v=git-22.288.45147-96ec0cd:48
    render http://192.0.2.1/luci-static/resources/view/system/system.js?v=git-22.288.45147-96ec0cd:47
    promise callback*__init__ http://192.0.2.1/luci-static/resources/luci.js?v=git-22.288.45147-96ec0cd:147
    super http://192.0.2.1/luci-static/resources/luci.js?v=git-22.288.45147-96ec0cd:22
    ClassConstructor http://192.0.2.1/luci-static/resources/luci.js?v=git-22.288.45147-96ec0cd:12
    compileClass http://192.0.2.1/luci-static/resources/luci.js?v=git-22.288.45147-96ec0cd:182
    promise callback*compileClass http://192.0.2.1/luci-static/resources/luci.js?v=git-22.288.45147-96ec0cd:178
    promise callback*require http://192.0.2.1/luci-static/resources/luci.js?v=git-22.288.45147-96ec0cd:184
    instantiateView http://192.0.2.1/luci-static/resources/ui.js?v=git-22.288.45147-96ec0cd:352
    <anonymous> http://192.0.2.1/cgi-bin/luci/admin/system/system:54
    promise callback* http://192.0.2.1/cgi-bin/luci/admin/system/system:53
...
NetworkError: HTTP error 404 while loading class file "/luci-static/resources/protocol/pppoe.js?v=git-22.288.45147-96ec0cd"
  at compileClass (http://192.0.2.1/luci-static/resources/luci.js?v=git-22.288.45147-96ec0cd:168:16)
  at promise callback*require (http://192.0.2.1/luci-static/resources/luci.js?v=git-22.288.45147-96ec0cd:184:112)
  at getProtocolHandlers/</< (http://192.0.2.1/luci-static/resources/network.js?v=git-22.288.45147-96ec0cd:7:162)
  at getProtocolHandlers/< (http://192.0.2.1/luci-static/resources/network.js?v=git-22.288.45147-96ec0cd:7:121)
  at promise callback*getProtocolHandlers (http://192.0.2.1/luci-static/resources/network.js?v=git-22.288.45147-96ec0cd:5:71)
  at initNetworkState (http://192.0.2.1/luci-static/resources/network.js?v=git-22.288.45147-96ec0cd:59:311)
  at getNetworks (http://192.0.2.1/luci-static/resources/network.js?v=git-22.288.45147-96ec0cd:135:54)
  at load (http://192.0.2.1/luci-static/resources/tools/widgets.js?v=git-22.288.45147-96ec0cd:34:369)
  at loadChildren (http://192.0.2.1/luci-static/resources/form.js?v=git-22.288.45147-96ec0cd:30:34)
  at load (http://192.0.2.1/luci-static/resources/form.js?v=git-22.288.45147-96ec0cd:67:17)
  at loadChildren (http://192.0.2.1/luci-static/resources/form.js?v=git-22.288.45147-96ec0cd:30:34)
  at load/< (http://192.0.2.1/luci-static/resources/form.js?v=git-22.288.45147-96ec0cd:45:32)
  at promise callback*load (http://192.0.2.1/luci-static/resources/form.js?v=git-22.288.45147-96ec0cd:44:542)
  at render (http://192.0.2.1/luci-static/resources/form.js?v=git-22.288.45147-96ec0cd:48:146)
  at render (http://192.0.2.1/luci-static/resources/view/system/system.js?v=git-22.288.45147-96ec0cd:47:10)
  at promise callback*__init__ (http://192.0.2.1/luci-static/resources/luci.js?v=git-22.288.45147-96ec0cd:147:232)
  at super (http://192.0.2.1/luci-static/resources/luci.js?v=git-22.288.45147-96ec0cd:22:42)
[... 9 lines cut ...]
    promise callback*require http://192.0.2.1/luci-static/resources/luci.js?v=git-22.288.45147-96ec0cd:184
    getProtocolHandlers http://192.0.2.1/luci-static/resources/network.js?v=git-22.288.45147-96ec0cd:7
    getProtocolHandlers http://192.0.2.1/luci-static/resources/network.js?v=git-22.288.45147-96ec0cd:7
    promise callback*getProtocolHandlers http://192.0.2.1/luci-static/resources/network.js?v=git-22.288.45147-96ec0cd:5
    initNetworkState http://192.0.2.1/luci-static/resources/network.js?v=git-22.288.45147-96ec0cd:59
    getNetworks http://192.0.2.1/luci-static/resources/network.js?v=git-22.288.45147-96ec0cd:135
    load http://192.0.2.1/luci-static/resources/tools/widgets.js?v=git-22.288.45147-96ec0cd:34
    loadChildren http://192.0.2.1/luci-static/resources/form.js?v=git-22.288.45147-96ec0cd:30
    load http://192.0.2.1/luci-static/resources/form.js?v=git-22.288.45147-96ec0cd:67
    loadChildren http://192.0.2.1/luci-static/resources/form.js?v=git-22.288.45147-96ec0cd:30
    load http://192.0.2.1/luci-static/resources/form.js?v=git-22.288.45147-96ec0cd:45
    promise callback*load http://192.0.2.1/luci-static/resources/form.js?v=git-22.288.45147-96ec0cd:44
    render http://192.0.2.1/luci-static/resources/form.js?v=git-22.288.45147-96ec0cd:48
    render http://192.0.2.1/luci-static/resources/view/system/system.js?v=git-22.288.45147-96ec0cd:47
    promise callback*__init__ http://192.0.2.1/luci-static/resources/luci.js?v=git-22.288.45147-96ec0cd:147
[... cut ...]
    promise callback*compileClass http://192.0.2.1/luci-static/resources/luci.js?v=git-22.288.45147-96ec0cd:178
    promise callback*require http://192.0.2.1/luci-static/resources/luci.js?v=git-22.288.45147-96ec0cd:184
    instantiateView http://192.0.2.1/luci-static/resources/ui.js?v=git-22.288.45147-96ec0cd:352
    <anonymous> http://192.0.2.1/cgi-bin/luci/admin/system/system:54
    promise callback* http://192.0.2.1/cgi-bin/luci/admin/system/system:53
[luci.js:161:9](http://192.0.2.1/luci-static/resources/luci.js?v=git-22.288.45147-96ec0cd)
```

jow-, 2022-10-20:
Yes, this is expected. LuCI attempts to load a matching class file for every network protocol reported by netifd. If such a file is not found, a traceback is emitted.

closed 2022-10-20 by jow-

systemcrash, 2023-12-07:
Evidently that's not the case from a design perspective: all UI modules are optional. I don't install ppp because I don't use it. It probably wins me about 1Kb in my final image 😆 

Does it make sense then to enumerate which UI modules are available? As a ppp user, I would expect to poke around in menus and status windows when running it. Its absence for me is fine. 

Maybe some stub handler for proto UIs, and requesting ppp gives a fake '200 OK' answer, whose content is 'go install ppp'?

jow-, 2023-12-07:
I don't install ppp because I don't use it. It probably wins me about 1Kb in my final image

But why is pppd installed then?

systemcrash, 2023-12-07:
Actually, you're right - those modules are included in the default selection from imagebuilder. I assume some are mandatory and generally leave that selection alone. I'd almost forgotten that ppp is there. I can remove those and trim them out from the image, but the overall 'traceback' remains. Probably unimportant if one can remove the underlying ppp*