openwrt/luci #6050: luci-app-wireguard: Missing option to configure peer address

manuelsteiner, 2022-10-21:
The Wireguard peer configuration section is missing the address field. I am not aware of configurations omitting the address parameter. It would be handy to set that in the peer configuration so the exported configuration as well as QR code will automatically include the parameter.

At least on Android, the Wireguard app will complain if no address is set and one has to be entered manually after scanning the QR code from the OpenWrt peer config generation page.

In case this is desired behaviour (i.e. to export configs to peers without an address and manually set it on the respective devices), please consider this issue void.
 
## Steps to reproduce:
1. Go to: Network → Interfaces → Wireguard → Edit → Peers tab → Edit a peer

## Actual behavior:
1. The option to set the `address` parameter is not present
2. As a result, the generated peer config does not contain any `address` parameter

## Expected behavior:
1. An option to set the `address` parameter for a peer in the peer configuration settings
2. The `address` parameter is included in the generated peer configs (text and QR code)

## Additional Information:
```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='22.03.1'
DISTRIB_REVISION='r19777-2853b6d652'
DISTRIB_TARGET='rockchip/armv8'
DISTRIB_ARCH='aarch64_generic'
DISTRIB_DESCRIPTION='OpenWrt 22.03.1 r19777-2853b6d652'
DISTRIB_TAINTS=''
```

castillofrancodamian, 2022-10-21:
I think the option is "Allowed IPs" in the peer settings.

manuelsteiner, 2022-10-21:
Hm but that generates the parameter `AllowedIps` for the `Peer` section and not `Address` for the `Interface section`. So that setting would indicate which packet sources are accepted from the connected endpoint. Which is how it is supposed to be used and apparently mapped correctly by the Luci interface.

castillofrancodamian, 2022-10-21:
You have to enter an IP in the `Allowed IPs` section. For example `192.168.34.2/32`.

manuelsteiner, 2022-10-21:
Well but what would I do if I want to use `Allowed IPs` for its intended purpose? Say I want to configure the peer with `Address 10.0.0.2/32` in the `Interface` section and want to set `AllowedIps 192.168.0.0/24` so the device can receive packets from that specific range?

In that case the generated config and QR Code will contain the `AllowedIps` parameter, as to its intended usage. Still, there is no `address` parameter for the `Interface` section and I need to configure that manually on the target device, for which the configuration is used. Otherwise (at least on Android), it won't even try to establish a connection.

castillofrancodamian, 2022-10-21:
Take a screenshot.

manuelsteiner, 2022-10-21:
Ok,

here are two scenarios.

- [config1](https://user-images.githubusercontent.com/5830180/197298656-6338346c-60f2-4f65-a958-fff14b90324c.png) contains a single IP address in the `Allowed IPs` field. Yet in [export1](https://user-images.githubusercontent.com/5830180/197298733-0866f7bc-ec04-4f8a-a6ee-786a433e1db0.png) it can be found in the `AllowediPs` parameter of the `Interface` section. As to your logic, you would expect to see an `Address 10.0.0.32` entry in the `Interface` section, where an empty red box is drawn now? That would not make sense because it would use the `AllowedIPs` parameter of Luci incorrectly and in two different ways depending on if a single IP or a range is entered.
- [config1](https://user-images.githubusercontent.com/5830180/197298784-528d675a-8ab6-4554-91e5-6257dd1b8fa8.png) contains an IP range in the `Allowed IPs` field. As expected, in [export2](https://user-images.githubusercontent.com/5830180/197298829-864cac6c-da13-46c5-82c7-4997d994a21e.png) the range can be found in the `AllowedIPs` parameter of the `Interface` section.

So in summary, there is no way to enter an IP address in the peer config section in Luci in order for it to show up as `Address` parameter in the `Interface` section of the exported configuration.

I hope this makes my issue clear now?

castillofrancodamian, 2022-10-21:
To configure an IP address on the interface, you must enter the corresponding IP in `IP Addresses`. That is in case the device in question is a "client".

jow-, 2022-10-21:
The reason why support for that was not implemented is that it does not appear to be an official option according to [`man 8 wg`](https://www.man7.org/linux/man-pages/man8/wg.8.html). We could add it - but what are the requirements? Is it a single address or can it be multiple (IPv4 + IPv6)? Must the addresses lie within the subnets of the router side wireguard interface addresses? Shall we enforce uniqueness among all peers or allow duplicates? What is the user changes the subnets (addresses) of the wireguard interface, shall we clear out all peer addresses or keep them?

manuelsteiner, 2022-10-23:
Ah, thank you for clarifying. That at least makes sense. So i guess the current path to success (at least on Android) is to scan the QR code and add the IP Address parameter in the app itself.

Zburatorul, 2023-01-08:
> So i guess the current path to success (at least on Android) is to scan the QR code and add the IP Address parameter in the app itself.
That is poor user experience because it requires whoever is doing the scanning to magically come up with an IP that's not already in use. That is why this is best handled by Luci. It could at least check whatever `[Interface] Address` is manually assigned to the given peer against the other peers.

jow-, 2023-01-09:
> imo, that is poor user experience because it requires whoever is doing the scanning to magically come up with an IP that's not already in use. That is why this is best handled by Luci. 

Nobody disagrees, but then LuCI has to "magically come up with an IP that's not already in use" and this has to be better defined before we can implement it. See considerations three comments above.

damadmai, 2023-02-03:
> Is it a single address or can it be multiple (IPv4 + IPv6)?

It can be none, one of that or both at the same time and even multiple ones.
I would suggest the same input field as used in AllowedIPs.

> Must the addresses lie within the subnets of the router side wireguard interface addresses?

The router side wireguard interface does not need to have an interface adress at all, I would not enforce anything other than it should be a valid IPv4 or IPv6 adresse.

> Shall we enforce uniqueness among all peers or allow duplicates?

In the beginning it should be sufficent to omit any check for that in order to keep it simple.
For example with static DHCP/DNS entries there is no check and dnsmasq does not start if you input duplicates, but I would also keep it simple by omitting the check because I could imagine a scenario where somebody would do 

> Shall we automatically allocate a new IP when adding a peer and prefill the field or leave it empty?

No, just keep it optional like the private key field.

> What if the user changes the subnets (addresses) of the wireguard interface, shall we clear out all peer addresses or keep them?

Just keep it simple, as said before the wireguard interface does not need to have an IP and even if it has one it does not need to be in the same subnet.

damadmai, 2023-02-03:
In summary I think it would be very helpful to add an optional field like "Private Key" name "Adresses" with the same input type as "Allowed IPs" field and just add the input from there to the generated config.

damadmai, 2023-02-03:
The `Address` Field is not part of `wg` but of `wg-quick` see also [man 8 wg-quick](https://man7.org/linux/man-pages/man8/wg-quick.8.html)

damadmai, 2023-02-03:
It may be desirable to also make the possibility for setting DNS and MTU

nevumx, 2023-04-19:
Hi folks, one PR for for the `Addresses` field is [here](https://github.com/openwrt/luci/pull/6352), and there is also one for the DNS setting [here](https://github.com/openwrt/luci/pull/6351). I didn't do MTU yet, but if someone really wants it, I can add it. First time contributing so let me know if anything is amiss.

nevumx, 2023-04-21:
@jow- and @castillofrancodamian, thoughts on the above PRs?

nevumx, 2023-07-08:
Any updates here, folks? @jow- / @castillofrancodamian ?

castillofrancodamian, 2023-07-08:
@nevumx I'm not a developer 😢

nevumx, 2023-07-17:
@castillofrancodamian Oh sorry, do you know who I should tag, then? 😅

castillofrancodamian, 2023-07-17:
@nevumx The other person you tagged is a developer.

nevumx, 2023-08-24:
@jow- I updated my PRs above(https://github.com/openwrt/luci/pull/6352 and https://github.com/openwrt/luci/pull/6351) to address feedback and merge conflicts, let me know your thoughts.

ralfbergs, 2025-07-05:
TBH I don't fully understand the outcome of this issue, but in any case I have a similar issue that may be identical or at least very strongly related to this one.

The entry field where you can define the client address when you create a new peer is still called "Allowed IPs" in OpenWrt 24.10, see the screenshot I'm attaching to this issue:

[image]

It should be called "Address," however.

(When I generate a config file from what I entered into the field, it will show up as follows:
```
[Interface]
...
Address = 192.168.9.6/32, fd00:9::6/128
```

systemcrash, 2025-07-05:
> TBH I don't fully understand the outcome of this issue, but in any case I have a similar issue that may be identical or at least very strongly related to this one.
> 
> The entry field where you can define the client address when you create a new peer is still called "Allowed IPs" in OpenWrt 24.10, see the screenshot I'm attaching to this issue:
> <img alt="Image" width="603" height="162" src="https://private-user-images.githubusercontent.com/7198745/462801886-b80aeef8-929a-4f96-85b3-70ac966069fb.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTE3MzQ0OTMsIm5iZiI6MTc1MTczNDE5MywicGF0aCI6Ii83MTk4NzQ1LzQ2MjgwMTg4Ni1iODBhZWVmOC05MjlhLTRmOTYtODViMy03MGFjOTY2MDY5ZmIucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MDcwNSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTA3MDVUMTY0OTUzWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NTIyMTc0MjMyOWYwYzU1NWEyMzllOGUwYzRkYzIzYjRmNWJkZTc0YWEyMTI3ODAzOGRiMzdhMGVhYTM4M2Y3ZCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.3kD_UKy3JYy2TyyitLMsq55-nRx5doDFABj4OrgLPKo">
> 
> It should be called "Address," however.
> 
> (When I generate a config file from what I entered into the field, it will show up as follows:
> 
> ```
> [Interface]
> ...
> Address = 192.168.9.6/32, fd00:9::6/128
> ```

This is not a problem: when you configure for a peer, you also configure for the host: from the host perspective, it needs to know what `allowed_ips` are allowed through the tunnel for the client. From the client perspective, it assumes that the `allowed_ips` are its interface IPs. This field has a dual purpose. The client config assumes the `allowed_ip` as its interface IP.

This is why `AllowedIPs = 0.0.0.0/0, ::/0` is added for the client peers - so the remote client can route traffic to any destination.

ralfbergs, 2025-07-06:
@systemcrash Thanks, Paul.

I must admit I'm even more confused now. :-(

According to the man page you quoted, `Address` would not be an allowed field in the `Interface` section. So, I thought "Maybe the man page he quoted was outdated?" Therefore, I looked at [the offical, latest man page](https://git.zx2c4.com/wireguard-tools/about/src/man/wg.8), and even according to *that* doc such a config item would be invalid.

And what makes it even _more_ confusing is that all clients I tested (Android, Mac) accept said config item as valid and follow it.

Would be great to hear your clarification, I must be missing something?

(The other explanations you gave make sense, I can understand them. Thanks a lot.)