openwrt/luci #6054: Legacy lua luci application runtime error / Missing views in luci-lua-runtime?

xsm1997, 2022-10-25:
## Steps to reproduce:

1. Goto some tabs (e.g. logs) of legacy lua luci applications.
2. Runtime error.
```
Runtime error
Unhandled exception during request dispatching
/usr/lib/lua/luci/ucodebridge.lua:22: /usr/lib/lua/luci/template.lua:158: Failed to load template 'header'.
Error while parsing template '/usr/lib/lua/luci/view/header.htm':
No such file or directory

In error(), file [C]
called from function [anonymous function] (/usr/lib/lua/luci/ucodebridge.lua:22)
called from function ((tail call))
In [anonymous function](), file /usr/share/ucode/luci/runtime.uc, line 126, byte 45:
  called from function [arrow function] (/usr/share/ucode/luci/dispatcher.uc:767:4)
  called from function render ([C])
  called from function run_action (/usr/share/ucode/luci/dispatcher.uc:768:4)
  called from function [anonymous function] (/usr/share/ucode/luci/dispatcher.uc:959:48)
  called from anonymous function (/www/cgi-bin/luci:39:12)

 `        return lcall.call(modname, method, ...args);`
  Near here ----------------------------------------^
```

Missing header.htm in luci-lua-runtime? I cannot find views which are originally in luci-base/luasrc/view.

## Actual behavior:

1. Runtime error.

## Expected behavior:

Pages load normally.

## Additional Information:
OpenWrt version information from system `/etc/openwrt_release`

**Example: Replace the following lines and remove this**
```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='SNAPSHOT'
DISTRIB_REVISION='r21108-a31b598590'
DISTRIB_TARGET='x86/64'
DISTRIB_ARCH='x86_64'
DISTRIB_DESCRIPTION='OpenWrt SNAPSHOT r21108-a31b598590'
DISTRIB_TAINTS='no-all'
```

vbzc, 2022-10-26:
I also found this problem, my solution is to copy the header.htm file from the original version of the router to the same location

jow-, 2022-10-26:
Please mention a specific page / affected application so I can reproduce the issue.

vbzc, 2022-10-26:
My affected program is https://github.com/lisaac/luci-app-diskman

1715173329, 2022-10-26:
extedit is also broken. you can reproduce this with luci-app-unbound.
visit http://openwrt.lan/cgi-bin/luci/admin/services/unbound/zones and click `edit` on right side.

jow-, 2022-10-26:
@1715173329 - extedit/arcombine is fixed with ca57291bdd3afbeae073ee315305d1ecc57f7f76

1715173329, 2022-10-26:
> @1715173329 - extedit/arcombine is fixed with [ca57291](https://github.com/openwrt/luci/commit/ca57291bdd3afbeae073ee315305d1ecc57f7f76)

Thank you! Now it works fine.

jow-, 2022-10-26:
@vbzc - should be fixed with 81f204d38f089ce6114bd67659c72e07ffbd57b4

closed 2022-10-26 by jow-