openwrt/luci #6058: luci-theme-openwrt-2020 can not fit！

zxlhhyccc, 2022-10-26:
At present, there are still some lua plug-ins, which are invalid for lua plug-ins...

jow-, 2022-10-26:
Please be more specific. What means "invalid for lua plug-ins..." ?

zxlhhyccc, 2022-10-27:
> Please be more specific. What means "invalid for lua plug-ins..." ?

Plug ins in lua format cannot be saved and applied under this theme。

1715173329, 2022-10-27:
UCI changes made with legacy lua applications cannot be applied. `Save and Apply` equals to `Save`, `Reset` doesn't work.
`Unsaved Changes` tip is also missing, have to execute `uci commit xxx` via cli.

Tested on luci-app-unbound with luci-theme-openwrt-2020.

jow-, 2022-10-27:
The theme has been converted to ucode and the compatibility issue should be addressed as well after https://github.com/openwrt/luci/commit/767acf32d4c568103bfb74746886050095cf90ef

closed 2022-10-27 by jow-