openwrt/luci #6068: attendedsysupgrade: Server response: Unsupported package(s): kmod-crypto-lib-blake2s

mmokrejs, 2022-11-02:
Hi,
  the attendedsysupgrade does not offer me 22.03.2 at all but only 21.02.5. Why?

```
Attended Sysupgrade
The attended sysupgrade service allows to easily upgrade vanilla and custom firmware images.
This is done by building a new firmware on demand via an online service.
Currently running: 21.02.3 - r16554-1d4dea6d4f
```
  I picked the only choice 21.02.5 but it failed with:

```
Error building the firmware image
Server response: Unsupported package(s): kmod-crypto-lib-blake2s
[Please report the error message and request](https://github.com/openwrt/asu/issues)

Request Data:ror building the firmware image

Server response: Unsupported package(s): `kmod-crypto-lib-blake2s`
[Please report the error message and request](https://github.com/openwrt/asu/issues)
```

Request Data:
```
{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r16554-1d4dea6d4f",
    "advanced_mode": "0",
    "branch": "21.02",
    "efi": null,
    "profile": "netgear,r7800",
    "target": "ipq806x/generic",
    "version": "21.02.5",
    "packages": [
        "asterisk-app-attended-transfer",
        "ath10k-firmware-qca9984-ct",
        "attendedsysupgrade-common",
        "auc",
        "base-files",
        "bash",
        "busybox",
        "ca-bundle",
        "cgi-io",
        "cryptsetup",
        "diffutils",
        "dnsmasq",
        "dropbear",
        "firewall",
        "fstools",
        "fwtool",
        "getrandom",
        "gnutls-utils",
        "hostapd-common",
        "ip-bridge",
        "ip6tables",
        "iptables",
        "iputils-ping",
        "iw",
        "iwinfo",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-ata-ahci",
        "kmod-ata-ahci-platform",
        "kmod-ata-core",
        "kmod-ath",
        "kmod-ath10k-ct",
        "kmod-cfg80211",
        "kmod-crypto-acompress",
        "kmod-crypto-aead",
        "kmod-crypto-arc4",
        "kmod-crypto-authenc",
        "kmod-crypto-cbc",
        "kmod-crypto-ccm",
        "kmod-crypto-cmac",
        "kmod-crypto-crc32",
        "kmod-crypto-crc32c",
        "kmod-crypto-cts",
        "kmod-crypto-deflate",
        "kmod-crypto-des",
        "kmod-crypto-ecb",
        "kmod-crypto-ecdh",
        "kmod-crypto-echainiv",
        "kmod-crypto-fcrypt",
        "kmod-crypto-gcm",
        "kmod-crypto-hw-geode",
        "kmod-crypto-hw-hifn-795x",
        "kmod-crypto-hw-padlock",
        "kmod-crypto-hw-talitos",
        "kmod-crypto-lib-blake2s",
        "kmod-crypto-lib-chacha20",
        "kmod-crypto-lib-chacha20poly1305",
        "kmod-crypto-lib-curve25519",
        "kmod-crypto-md4",
        "kmod-crypto-md5",
        "kmod-crypto-michael-mic",
        "kmod-crypto-misc",
        "kmod-crypto-pcbc",
        "kmod-crypto-rmd160",
        "kmod-crypto-sha1",
        "kmod-crypto-sha512",
        "kmod-crypto-test",
        "kmod-crypto-user",
        "kmod-crypto-wq",
        "kmod-crypto-xcbc",
        "kmod-fs-exfat",
        "kmod-fs-ext4",
        "kmod-fuse",
        "kmod-gpio-button-hotplug",
        "kmod-hid",
        "kmod-hwmon-core",
        "kmod-ip6tables",
        "kmod-ipt-conntrack",
        "kmod-ipt-core",
        "kmod-ipt-nat",
        "kmod-ipt-offload",
        "kmod-leds-gpio",
        "kmod-lib-crc-ccitt",
        "kmod-lib-lz4",
        "kmod-lib-lzo",
        "kmod-mac80211",
        "kmod-nf-conntrack",
        "kmod-nf-conntrack6",
        "kmod-nf-flow",
        "kmod-nf-ipt",
        "kmod-nf-ipt6",
        "kmod-nf-nat",
        "kmod-nf-reject",
        "kmod-nf-reject6",
        "kmod-nls-base",
        "kmod-phy-qcom-ipq806x-usb",
        "kmod-ppp",
        "kmod-pppoe",
        "kmod-pppox",
        "kmod-scsi-core",
        "kmod-scsi-generic",
        "kmod-slhc",
        "kmod-tcp-bbr",
        "kmod-usb-core",
        "kmod-usb-dwc3",
        "kmod-usb-dwc3-qcom",
        "kmod-usb-ehci",
        "kmod-usb-ledtrig-usbport",
        "kmod-usb-ohci",
        "kmod-usb-storage",
        "kmod-usb-storage-uas",
        "kmod-usb-uhci",
        "kmod-usb2",
        "kmod-usb3",
        "libblobmsg-json",
        "libdevmapper",
        "libgpg-error",
        "libip4tc",
        "libip6tc",
        "libiwinfo",
        "libiwinfo-data",
        "libiwinfo-lua",
        "libjson-c",
        "libjson-script",
        "liblua",
        "liblucihttp",
        "liblucihttp-lua",
        "libnl-tiny",
        "libopenssl",
        "libopenssl-conf",
        "libopenssl-devcrypto",
        "libprotobuf-c",
        "libsysfs",
        "libubox",
        "libubus",
        "libubus-lua",
        "libuci",
        "libuclient",
        "libunbound",
        "libustream-wolfssl",
        "libwolfssl",
        "libxml2",
        "libxtables",
        "logd",
        "lua",
        "luci",
        "luci-app-attendedsysupgrade",
        "luci-app-firewall",
        "luci-app-ocserv",
        "luci-app-openvpn",
        "luci-app-opkg",
        "luci-app-squid",
        "luci-base",
        "luci-i18n-attendedsysupgrade-en",
        "luci-lib-base",
        "luci-lib-ip",
        "luci-lib-jsonc",
        "luci-lib-nixio",
        "luci-mod-admin-full",
        "luci-mod-network",
        "luci-mod-status",
        "luci-mod-system",
        "luci-proto-3g",
        "luci-proto-bonding",
        "luci-proto-ipip",
        "luci-proto-ipv6",
        "luci-proto-modemmanager",
        "luci-proto-ncm",
        "luci-proto-openconnect",
        "luci-proto-ppp",
        "luci-proto-pppossh",
        "luci-proto-qmi",
        "luci-proto-vpnc",
        "luci-ssl",
        "luci-theme-bootstrap",
        "lvm2",
        "mbim-utils",
        "mc",
        "mtd",
        "netifd",
        "nmap-ssl",
        "ocserv",
        "odhcp6c",
        "odhcpd-ipv6only",
        "openconnect",
        "openssh-client",
        "openssh-client-utils",
        "openssh-moduli",
        "openssh-sftp-client",
        "openssh-sftp-server",
        "openssl-util",
        "openvpn-easy-rsa",
        "openvpn-openssl",
        "openwrt-keyring",
        "opkg",
        "ppp",
        "ppp-mod-pppoe",
        "procd",
        "px5g-wolfssl",
        "qmi-utils",
        "rpcd",
        "rpcd-mod-file",
        "rpcd-mod-iwinfo",
        "rpcd-mod-luci",
        "rpcd-mod-rrdns",
        "squid",
        "squid-mod-cachemgr",
        "sshfs",
        "sshtunnel",
        "strace",
        "swconfig",
        "syncthing",
        "sysfsutils",
        "tcpdump",
        "ubi-utils",
        "uboot-envtools",
        "ubox",
        "ubus",
        "ubusd",
        "uci",
        "uclient-fetch",
        "uhttpd",
        "uhttpd-mod-ubus",
        "urandom-seed",
        "urngd",
        "usign",
        "vpn-policy-routing",
        "vpnc",
        "vsftpd-tls",
        "wireless-regdb",
        "wpad-basic-wolfssl",
        "zlib"
    ],
    "diff_packages": true,
    "filesystem": "squashfs",
    "client": "luci/git-22.210.54423-3b0b325"
}
```

## Additional Information:
```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='21.02.3'
DISTRIB_REVISION='r16554-1d4dea6d4f'
DISTRIB_TARGET='ipq806x/generic'
DISTRIB_ARCH='arm_cortex-a15_neon-vfpv4'
DISTRIB_DESCRIPTION='OpenWrt 21.02.3 r16554-1d4dea6d4f'
DISTRIB_TAINTS=''
```

hnyman, 2022-11-02:
This is unrelated to the LuCI GUI itself.

But the underlying reason is likely that the blake2s kmod is always "built-in" into kernel since a few months ago and is not a selectable package any more. You should remove/deselect it from the build.

closed 2022-11-02 by hnyman

mmokrejs, 2022-11-02:
Thank you @hnyman but please re-assign it then to the releavnt person to fix the attendedsysupgrade package. I don't think I was even offered to continue.