openwrt/luci #6071: Broken install in latest nightly 1-2th November 2022

bittin, 2022-11-03:
root@rainbowdash:~# opkg update
Downloading https://downloads.openwrt.org/snapshots/targets/mvebu/cortexa9/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_core
Downloading https://downloads.openwrt.org/snapshots/targets/mvebu/cortexa9/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/packages/arm_cortex-a9_vfpv3-d16/base/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_base
Downloading https://downloads.openwrt.org/snapshots/packages/arm_cortex-a9_vfpv3-d16/base/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/targets/mvebu/cortexa9/kmods/5.10.152-1-73d6c4a291e73b18aeb040875a4c9940/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_kmods
Downloading https://downloads.openwrt.org/snapshots/targets/mvebu/cortexa9/kmods/5.10.152-1-73d6c4a291e73b18aeb040875a4c9940/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/packages/arm_cortex-a9_vfpv3-d16/luci/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_luci
Downloading https://downloads.openwrt.org/snapshots/packages/arm_cortex-a9_vfpv3-d16/luci/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/packages/arm_cortex-a9_vfpv3-d16/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_packages
Downloading https://downloads.openwrt.org/snapshots/packages/arm_cortex-a9_vfpv3-d16/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/packages/arm_cortex-a9_vfpv3-d16/routing/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_routing
Downloading https://downloads.openwrt.org/snapshots/packages/arm_cortex-a9_vfpv3-d16/routing/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/packages/arm_cortex-a9_vfpv3-d16/telephony/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_telephony
Downloading https://downloads.openwrt.org/snapshots/packages/arm_cortex-a9_vfpv3-d16/telephony/Packages.sig
Signature check passed.
root@rainbowdash:~# opkg install luci
Installing luci (git-22.297.83017-0143ef2) to root...
Downloading https://downloads.openwrt.org/snapshots/packages/arm_cortex-a9_vfpv3-d16/luci/luci_git-22.297.83017-0143ef2_all.ipk
Installing luci-proto-ipv6 (git-21.148.48881-79947af) to root...
Downloading https://downloads.openwrt.org/snapshots/packages/arm_cortex-a9_vfpv3-d16/luci/luci-proto-ipv6_git-21.148.48881-79947af_all.ipk
Installing rpcd (2022-09-21-8c852b65-1) to root...
Downloading https://downloads.openwrt.org/snapshots/packages/arm_cortex-a9_vfpv3-d16/base/rpcd_2022-09-21-8c852b65-1_arm_cortex-a9_vfpv3-d16.ipk
Installing rpcd-mod-file (2022-09-21-8c852b65-1) to root...
Downloading https://downloads.openwrt.org/snapshots/packages/arm_cortex-a9_vfpv3-d16/base/rpcd-mod-file_2022-09-21-8c852b65-1_arm_cortex-a9_vfpv3-d16.ipk
Installing libnl-tiny2022-05-23 (2022-05-23-28c44ca9-1) to root...
Downloading https://downloads.openwrt.org/snapshots/packages/arm_cortex-a9_vfpv3-d16/base/libnl-tiny2022-05-23_2022-05-23-28c44ca9-1_arm_cortex-a9_vfpv3-d16.ipk
Configuring rpcd.
Configuring rpcd-mod-file.
Configuring luci-proto-ipv6.
Collected errors:
 * check_data_file_clashes: Package libnl-tiny2022-05-23 wants to install file /usr/lib/libnl-tiny.so
	But that file is already provided by package  * libnl-tiny2022-11-01
 * opkg_install_cmd: Cannot install package luci.

hnyman, 2022-11-03:
Due to the 2-phase builsbot (kernel, core packages, images) vs. vanilla pacvkages,  it may take 2-3 days after the libnl-tiny version bump before all packages have been rebuilt against the new version.
Try again in 1-2 days.

(and not related to LuCI GUI at all)

closed 2022-11-03 by hnyman

bittin, 2022-11-05:
this is now solved