openwrt/luci #6074: luci-base: file saving (`fs.write`) saves the file, but fails on unexpected response

oldium, 2022-11-03:
## Steps to reproduce:

Tested on master build from yesterday (commit [Merge pull request](https://github.com/openwrt/luci/commit/9ea26a75ed44cde3def5d061a867fa513d365c27) https://github.com/openwrt/luci/pull/6064 [from weblate/weblate-openwrt-luci](https://github.com/openwrt/luci/commit/9ea26a75ed44cde3def5d061a867fa513d365c27)). No LUA installed, just plain ucode.

1. go to: System → Backup/Flash Firmware → Configuration tab
2. click Save

## Actual behavior:

[image]

## Expected behavior:

Success.

## Additional Information:

The JavaScript `fs.write` function gets response from RPC call:

```json
{
    "jsonrpc": "2.0",
    "id": 9,
    "result": [
        0,
        null
    ]
}
```

This does not fit the expected response. Older few months old firmware returns the following:

```json
{
    "id": 12,
    "jsonrpc": "2.0",
    "result": [
        0
    ]
}
```

There was no change to the `fs.js` nor to `rpc.js` JavaScript code for 2 years, so I guess this could be related to the ucode switch.

CC @jow-

oldium, 2022-11-03:
Another difference is that the response now is nicely formatted, old RPC response was received all in one line.

jow-, 2022-11-03:
@oldium - fix pused. It only affected installations without uhttpd or uhttpd-mod-ubus.

closed 2022-11-03 by jow-

oldium, 2022-11-03:
Thanks, I will test it. You are right, I am using lighttpd, so no uhttpd.

oldium, 2022-11-03:
Works fine, thanks for a quick fix!

commit 0dd0114a (2022-11-03): luci-base: fix empty reply format in ubus gateway fallback code

The ubus gateway fallback code incorrectly formatted ubus replies
containing no payload data when forwarding them via HTTP, leading
to `TypeError: Unexpected reply data format` errors in at least
the `luci.fs` class when receiving replies without payload.

Fix this issue by ensuring that the result array never contains
a `null` value for the payload, send an one-element array containing
just the status code in this case instead.

Fixes: #6074