openwrt/luci #6078: RPCError

snakwu, 2022-11-06:
RPC call to luci/getFeatures failed with error -32000: Object not found

snakwu, 2022-11-06:
[image]

M95D, 2022-11-09:
It might be the same as: https://github.com/openwrt/luci/issues/4555

jow-, 2022-11-09:
Run `service rpcd restart` and check `logread` for errors. Also run `ucode /usr/share/rpcd/ucode/luci` and see if this returns without output.

snakwu, 2022-11-10:
> Run `service rpcd restart` and check `logread` for errors. Also run `ucode /usr/share/rpcd/ucode/luci` and see if this returns without output.

[image]
[image]

nothing is output

rendiix, 2022-11-10:
Sampe problem
[image]

snakwu, 2022-11-10:
> 

[image]

jow-, 2022-11-10:
Please provide the output of `ubus -v list`

ptpt52, 2022-11-10:
@jow- 

my lagency luci code cannot run:
https://github.com/x-wrt/com.x-wrt/blob/201a40c0f1eca3f0d68fdc08e286fac4eb736b37/luci-app-natcap/files/luci/controller/natcap.lua#L175-L211
```lua
function get_openvpn_client_udp()
	local reader = ltn12_popen("sh /usr/share/natcapd/natcapd.openvpn.sh gen_client_udp")

	luci.http.header('Content-Disposition', 'attachment; filename="natcap-client-udp.ovpn"')
	luci.http.prepare_content("application/x-openvpn-profile")
	luci.ltn12.pump.all(reader, luci.http.write)
end

function ltn12_popen(command)

	local fdi, fdo = nixio.pipe()
	local pid = nixio.fork()

	if pid > 0 then
		fdo:close()
		local close
		return function()
			local buffer = fdi:read(2048)
			local wpid, stat = nixio.waitpid(pid, "nohang")
			if not close and wpid and stat == "exited" then
				close = true
			end

			if buffer and #buffer > 0 then
				return buffer
			elseif close then
				fdi:close()
				return nil
			end
		end
	elseif pid == 0 then
		nixio.dup(fdo, nixio.stdout)
		fdi:close()
		fdo:close()
		nixio.exec("/bin/sh", "-c", command)
	end
end
```

error in logread:
```
Fri Nov 11 06:04:36 2022 daemon.info uwsgi-luci: sed: write error
Fri Nov 11 06:04:36 2022 daemon.info uwsgi-luci: sh: write error: Broken pipe
Fri Nov 11 06:04:36 2022 daemon.info uwsgi-luci: cat: write error: Broken pipe
Fri Nov 11 06:04:36 2022 daemon.info uwsgi-luci: sh: write error: Broken pipe
Fri Nov 11 06:04:36 2022 daemon.info uwsgi-luci: cat: write error: Broken pipe
Fri Nov 11 06:04:36 2022 daemon.info uwsgi-luci: sh: write error: Broken pipe
Fri Nov 11 06:04:36 2022 daemon.info uwsgi-luci: cat: write error: Broken pipe
Fri Nov 11 06:04:36 2022 daemon.info uwsgi-luci: sh: write error: Broken pipe
Fri Nov 11 06:04:36 2022 daemon.info uwsgi-luci: cat: write error: Broken pipe
Fri Nov 11 06:04:36 2022 daemon.info uwsgi-luci: sh: write error: Broken pipe
Fri Nov 11 06:05:06 2022 daemon.info uwsgi-luci: sed: write error
Fri Nov 11 06:05:06 2022 daemon.info uwsgi-luci: sh: write error: Broken pipe
Fri Nov 11 06:05:06 2022 daemon.info uwsgi-luci: cat: write error: Broken pipe
Fri Nov 11 06:05:06 2022 daemon.info uwsgi-luci: sh: write error: Broken pipe
Fri Nov 11 06:05:06 2022 daemon.info uwsgi-luci: cat: write error: Broken pipe
Fri Nov 11 06:05:06 2022 daemon.info uwsgi-luci: sh: write error: Broken pipe
Fri Nov 11 06:05:06 2022 daemon.info uwsgi-luci: cat: write error: Broken pipe
Fri Nov 11 06:05:06 2022 daemon.info uwsgi-luci: sh: write error: Broken pipe
Fri Nov 11 06:05:06 2022 daemon.info uwsgi-luci: cat: write error: Broken pipe
Fri Nov 11 06:05:06 2022 daemon.info uwsgi-luci: sh: write error: Broken pipe
```

ptpt52, 2022-11-10:
hmm, I should open an new issue?

snakwu, 2022-11-10:
> Please provide the output of `ubus -v list`. Also mention the OpenWrt target

root@SnakWu-OpenWrt:~# ubus -v list
'dhcp' @ad719314
	"ipv4leases":{}
	"ipv6leases":{}
	"add_lease":{"ip":"String","mac":"String","duid":"String","hostid":"String","leasetime":"String","name":"String"}
'dnsmasq' @5d120a16
	"metrics":{}
	"set_connmark_allowlist":{"mark":"Integer","mask":"Integer","patterns":"Array"}
'dnsmasq.dns' @36aeb29b
'file' @9a3e035e
	"read":{"path":"String","base64":"Boolean","ubus_rpc_session":"String"}
	"write":{"path":"String","data":"String","append":"Boolean","mode":"Integer","base64":"Boolean","ubus_rpc_session":"String"}
	"list":{"path":"String","ubus_rpc_session":"String"}
	"stat":{"path":"String","ubus_rpc_session":"String"}
	"md5":{"path":"String","ubus_rpc_session":"String"}
	"remove":{"path":"String","ubus_rpc_session":"String"}
	"exec":{"command":"String","params":"Array","env":"Table","ubus_rpc_session":"String"}
'hotplug.dhcp' @07d8d6ed
	"call":{"env":"Array"}
'hotplug.iface' @dfef3157
	"call":{"env":"Array"}
'hotplug.neigh' @9f4f5410
	"call":{"env":"Array"}
'hotplug.net' @215495a6
	"call":{"env":"Array"}
'hotplug.ntp' @818c934d
	"call":{"env":"Array"}
'hotplug.tftp' @c6f62bda
	"call":{"env":"Array"}
'iwinfo' @babb5594
	"devices":{}
	"info":{"device":"String"}
	"scan":{"device":"String"}
	"assoclist":{"device":"String","mac":"String"}
	"freqlist":{"device":"String"}
	"txpowerlist":{"device":"String"}
	"countrylist":{"device":"String"}
	"survey":{"device":"String"}
	"phyname":{"section":"String"}
'log' @1a2c5c3b
	"read":{"lines":"Integer","stream":"Boolean","oneshot":"Boolean"}
	"write":{"event":"String"}
'luci-rpc' @952c05c2
	"getNetworkDevices":{}
	"getWirelessDevices":{}
	"getHostHints":{}
	"getDUIDHints":{}
	"getBoardJSON":{}
	"getDHCPLeases":{"family":"Integer"}
'luci.ddns' @763185ec
	"get_services_log":{"service_name":"String"}
	"get_env":{}
	"get_ddns_state":{}
	"get_services_status":{}
'network' @6d7ab89e
	"restart":{}
	"reload":{}
	"add_host_route":{"target":"String","v6":"Boolean","interface":"String","exclude":"Boolean"}
	"get_proto_handlers":{}
	"add_dynamic":{"name":"String"}
	"netns_updown":{"jail":"String","start":"Boolean"}
'network.device' @df4af68e
	"status":{"name":"String"}
	"set_alias":{"alias":"Array","device":"String"}
	"set_state":{"name":"String","defer":"Boolean","auth_status":"Boolean"}
	"stp_init":{}
'network.interface' @679ca21c
	"up":{}
[... 126 lines cut ...]
'system' @cd51e573
	"board":{}
	"info":{}
	"reboot":{}
	"watchdog":{"frequency":"Integer","timeout":"Integer","magicclose":"Boolean","stop":"Boolean"}
	"signal":{"pid":"Integer","signum":"Integer"}
	"validate_firmware_image":{"path":"String"}
	"sysupgrade":{"path":"String","force":"Boolean","backup":"String","prefix":"String","command":"String","options":"Table"}
'uci' @6e8b5fce
	"configs":{}
	"get":{"config":"String","section":"String","option":"String","type":"String","match":"Table","ubus_rpc_session":"String"}
	"state":{"config":"String","section":"String","option":"String","type":"String","match":"Table","ubus_rpc_session":"String"}
	"add":{"config":"String","type":"String","name":"String","values":"Table","ubus_rpc_session":"String"}
	"set":{"config":"String","section":"String","type":"String","match":"Table","values":"Table","ubus_rpc_session":"String"}
	"delete":{"config":"String","section":"String","type":"String","match":"Table","option":"String","options":"Array","ubus_rpc_session":"String"}
	"rename":{"config":"String","section":"String","option":"String","name":"String","ubus_rpc_session":"String"}
	"order":{"config":"String","sections":"Array","ubus_rpc_session":"String"}
	"changes":{"config":"String","ubus_rpc_session":"String"}
	"revert":{"config":"String","ubus_rpc_session":"String"}
	"commit":{"config":"String","ubus_rpc_session":"String"}
	"apply":{"rollback":"Boolean","timeout":"Integer","ubus_rpc_session":"String"}
	"confirm":{"ubus_rpc_session":"String"}
	"rollback":{"ubus_rpc_session":"String"}
	"reload_config":{}
root@SnakWu-OpenWrt:~#

jow-, 2022-11-10:
Please install strace and provide the output of `service rpcd stop; strace /sbin/rpcd`

snakwu, 2022-11-11:
i can't install strace and provide，I can't install the mission software right now

> Please install strace and provide the output of `service rpcd stop; strace /sbin/rpcd`
> 
> Also: `ubus call system board`

i can't install strace and provide，I can't install the mission software right now

wangzan200808, 2022-11-13:
the same problem

jow-, 2022-11-13:
the same questions

naluckstar, 2023-01-03:
the same questions

naluckstar, 2023-01-03:
{
        "kernel": "5.10.159",
        "hostname": "OpenWrt",
        "system": "MediaTek MT7628AN ver:1 eco:2",
        "model": "Easywayphy DTU03_S",
        "board_name": "Easywayphy,DTU03_S",
        "rootfs_type": "squashfs",
        "release": {
                "distribution": "OpenWrt",
                "version": "SNAPSHOT",
                "revision": "unknown",
                "target": "ramips/mt76x8",
                "description": "OpenWrt SNAPSHOT unknown"
        }
}

naluckstar, 2023-01-03:
"set_data":{}
'network.interface.wifi6' @c2e635f2
        "up":{}
        "down":{}
        "renew":{}
        "status":{}
        "prepare":{}
        "dump":{}
        "add_device":{"name":"String","link-ext":"Boolean","vlan":"Array"}
        "remove_device":{"name":"String","link-ext":"Boolean","vlan":"Array"}
        "notify_proto":{}
        "remove":{}
        "set_data":{}
'network.wireless' @fd64f66e
        "up":{}
        "down":{}
        "reconf":{}
        "status":{}
        "notify":{}
        "get_validate":{}
'service' @56b7f3ba
        "set":{"name":"String","script":"String","instances":"Table","triggers":"Array","validate":"Array","autostart":"Boolean","data":"Table"}
        "add":{"name":"String","script":"String","instances":"Table","triggers":"Array","validate":"Array","autostart":"Boolean","data":"Table"}
        "list":{"name":"String","verbose":"Boolean"}
        "delete":{"name":"String","instance":"String"}
        "signal":{"name":"String","instance":"String","signal":"Integer"}
        "update_start":{"name":"String"}
        "update_complete":{"name":"String"}
        "event":{"type":"String","data":"Table"}
        "validate":{"package":"String","type":"String","service":"String"}
        "get_data":{"name":"String","instance":"String","type":"String"}
        "state":{"spawn":"Boolean","name":"String"}
        "watchdog":{"mode":"Integer","timeout":"Integer","name":"String","instance":"String"}
'system' @e5a800c0
        "board":{}
        "info":{}
        "reboot":{}
        "watchdog":{"frequency":"Integer","timeout":"Integer","magicclose":"Boolean","stop":"Boolean"}
        "signal":{"pid":"Integer","signum":"Integer"}
        "validate_firmware_image":{"path":"String"}
        "sysupgrade":{"path":"String","force":"Boolean","backup":"String","prefix":"String","command":"String","options":"Table"}
'wpa_supplicant' @5a54d93e
        "config_add":{"driver":"String","iface":"String","bridge":"String","hostapd_ctrl":"String","ctrl":"String","config":"String"}
        "config_remove":{"iface":"String"}

XiaoliChan, 2023-01-10:
Got same problem
```
RPC call to luci/getFeatures failed with error -32000: Object not found
  at handleCallReply (http://192.168.1.102/luci-static/resources/rpc.js?v=git-22.361.69865-deed682:11:3)
```

jow-, 2023-01-10:
And it is still the same questions. All people reporting this issue seem to lack the `luci` object in ubus. This object is hosted by rpcd and implemented using an ucode script. Maybe your builds don't have rpcd-mod-ucode installed. I requested an strace of the rpcd startup sequence a while back, haven't received it yet.

Also asked for the OpenWrt version of affected devices and the output presented contained `"revision": "unknown"` so I guess this is proprietary, non-official OpenWrt builds exhibiting this behavior.

XiaoliChan, 2023-01-11:
@jow- I can confirm the `rpcd-mod-ucode` have already installed, still don't know why get this exception.

jow-, 2023-01-11:
> Another thing you can try is executing the following command, it should return without error and no output:
> `ucode /usr/share/rpcd/ucode/luci`

XiaoliChan, 2023-01-11:
> 

Yep, no output with this command.

jow-, 2023-01-11:
The please run `service rpcd stop; service rcpd start; ubus call luci getFeatures`

XiaoliChan, 2023-01-11:
> The please run `service rpcd stop; service rcpd start; ubus call luci getFeatures`

I will try this in weekend :)

XiaoliChan, 2023-01-15:
@jow- Really weird, I recompile the firmware with latest commit version of OP & Luci, the exception has been gone.

mis1042, 2023-01-16:
> 请跑`service rpcd stop; service rcpd start; ubus call luci getFeatures`
Not Work
[image]

mis1042, 2023-01-16:
> The please run `service rpcd stop; service rcpd start; ubus call luci getFeatures`

[image]

As you see,Not work

mis1042, 2023-01-16:
Maybe we can save this error for the Spring Festival : )

mis1042, 2023-01-16:
> > Run `service rpcd restart` and check `logread` for errors. Also run `ucode /usr/share/rpcd/ucode/luci` and see if this returns without output.
> 
> [image] [image]
> 
> nothing is output

I founf something wrong...

Sun Jan 15 12:43:53 2023 daemon.notice netifd: wan (2087): Command failed: ubus call network.interface notify_proto { "action": 0, "link-up": false, "keep": false, "interface": "wan" }

mis1042, 2023-01-16:
> The please run `service rpcd stop; service rcpd start; ubus call luci getFeatures`

ubus -v list:
'container' @d5d957db
        "set":{"name":"String","script":"String","instances":"Table","triggers":"Array","validate":"Array","autostart":"Boolean","data":"Table"}
        "add":{"name":"String","script":"String","instances":"Table","triggers":"Array","validate":"Array","autostart":"Boolean","data":"Table"}
        "list":{"name":"String","verbose":"Boolean"}
        "delete":{"name":"String","instance":"String"}
        "state":{"spawn":"Boolean","name":"String"}
        "get_features":{}
        "console_set":{"name":"String","instance":"String"}
        "console_attach":{"name":"String","instance":"String"}
'dawn' @ab324f6f
        "add_mac":{"addrs":"Array"}
        "get_hearing_map":{}
        "get_network":{}
        "reload_config":{}
'dhcp' @be0181c9
        "ipv4leases":{}
        "ipv6leases":{}
        "add_lease":{"ip":"String","mac":"String","duid":"String","hostid":"String","leasetime":"String","name":"String"}
'dnsmasq' @51589253
        "metrics":{}
'dnsmasq.dns' @5b9b2a49
'file' @e9879f6d
        "read":{"path":"String","base64":"Boolean","ubus_rpc_session":"String"}
        "write":{"path":"String","data":"String","append":"Boolean","mode":"Integer","base64":"Boolean","ubus_rpc_session":"String"}
        "list":{"path":"String","ubus_rpc_session":"String"}
        "stat":{"path":"String","ubus_rpc_session":"String"}
        "md5":{"path":"String","ubus_rpc_session":"String"}
        "remove":{"path":"String","ubus_rpc_session":"String"}
        "exec":{"command":"String","params":"Array","env":"Table","ubus_rpc_session":"String"}
'hostapd' @b12e0815
        "config_add":{"iface":"String","config":"String"}
        "config_remove":{"iface":"String"}
'hotplug.dhcp' @c42d4717
        "call":{"env":"Array"}
'hotplug.ieee80211' @7e98961e
        "call":{"env":"Array"}
'hotplug.iface' @7228ecb1
        "call":{"env":"Array"}
'hotplug.neigh' @e9af9f37
        "call":{"env":"Array"}
'hotplug.net' @7dc29955
        "call":{"env":"Array"}
'hotplug.ntp' @2a35069a
        "call":{"env":"Array"}
'hotplug.tftp' @ef21e244
        "call":{"env":"Array"}
'hotplug.usb' @ffd114d6
        "call":{"env":"Array"}
'iwinfo' @d2e4b6fc
        "devices":{}
        "info":{"device":"String"}
        "scan":{"device":"String"}
        "assoclist":{"device":"String","mac":"String"}
        "freqlist":{"device":"String"}
        "txpowerlist":{"device":"String"}
        "countrylist":{"device":"String"}
        "survey":{"device":"String"}
        "phyname":{"section":"String"}
'log' @bc26a7f5
        "read":{"lines":"Integer","stream":"Boolean","oneshot":"Boolean"}
        "write":{"event":"String"}
'luci-rpc' @1025f597
        "getNetworkDevices":{}
        "getWirelessDevices":{}
        "getHostHints":{}
        "getDUIDHints":{}
        "getBoardJSON":{}
[... 440 lines cut ...]
        "add":{"config":"String","type":"String","name":"String","values":"Table","ubus_rpc_session":"String"}
        "set":{"config":"String","section":"String","type":"String","match":"Table","values":"Table","ubus_rpc_session":"String"}
        "delete":{"config":"String","section":"String","type":"String","match":"Table","option":"String","options":"Array","ubus_rpc_session":"String"}
        "rename":{"config":"String","section":"String","option":"String","name":"String","ubus_rpc_session":"String"}
        "order":{"config":"String","sections":"Array","ubus_rpc_session":"String"}
        "changes":{"config":"String","ubus_rpc_session":"String"}
        "revert":{"config":"String","ubus_rpc_session":"String"}
        "commit":{"config":"String","ubus_rpc_session":"String"}
        "apply":{"rollback":"Boolean","timeout":"Integer","ubus_rpc_session":"String"}
        "confirm":{"ubus_rpc_session":"String"}
        "rollback":{"ubus_rpc_session":"String"}
        "reload_config":{}
'umdns' @55f47a51
        "set_config":{"interfaces":"Array","keep":"Boolean"}
        "query":{"question":"String","interface":"String","type":"Integer"}
        "fetch":{"question":"String","interface":"String","type":"Integer"}
        "update":{}
        "browse":{}
        "hosts":{}
        "reload":{}
'wpa_supplicant' @9f548ce3
        "config_add":{"driver":"String","iface":"String","bridge":"String","hostapd_ctrl":"String","ctrl":"String","config":"String"}
        "config_remove":{"iface":"String"}

**But there is no luci getFeatures**

jow-, 2023-01-16:
@mis1042 - What's the output of `ls -lh /usr/lib/rpcd/ucode.so` ?

mis1042, 2023-01-16:
-rwxr-xr-x    1 root     root       16.4K Nov 17 01:55 /usr/lib/rpcd/ucode.so

jow-, 2023-01-16:
Then please `opkg update strace; service rpcd stop; strace rpcd` and provide the output here.

mis1042, 2023-01-17:
> Then please `opkg update strace; service rpcd stop; strace rpcd` and provide the output here.

root@OpenWrt:~# strace rpcd
execve("/sbin/rpcd", ["rpcd"], 0x7fc76ee0 /* 13 vars */) = 0
set_thread_area(0x77eb7df0)             = 0
set_tid_address(0x77eaf69c)             = 27806
brk(NULL)                               = 0x557ee000
brk(0x557f0000)                         = 0x557f0000
mmap2(0x557ee000, 4096, PROT_NONE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x557ee000
open("/etc/ld-musl-mipsel-sf.path", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/lib/libubox.so.20220927", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fcntl64(3, F_SETFD, FD_CLOEXEC)         = 0
statx(3, "", AT_STATX_SYNC_AS_STAT|AT_EMPTY_PATH, STATX_BASIC_STATS, {stx_mask=STATX_BASIC_STATS|STATX_MNT_ID, stx_attributes=0, stx_mode=S_IFREG|0644, stx_size=37451, ...}) = 0
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0\10\0\1\0\0\0\0\37\0\0004\0\0\0"..., 936) = 936
mmap2(NULL, 106496, PROT_READ|PROT_EXEC, MAP_PRIVATE, 3, 0) = 0x77ddb000
mmap2(0x77df3000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0x8000) = 0x77df3000
close(3)                                = 0
open("/lib/libubus.so.20220601", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fcntl64(3, F_SETFD, FD_CLOEXEC)         = 0
statx(3, "", AT_STATX_SYNC_AS_STAT|AT_EMPTY_PATH, STATX_BASIC_STATS, {stx_mask=STATX_BASIC_STATS|STATX_MNT_ID, stx_attributes=0, stx_mode=S_IFREG|0644, stx_size=16763, ...}) = 0
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0\10\0\1\0\0\0`\20\0\0004\0\0\0"..., 936) = 936
mmap2(NULL, 86016, PROT_READ|PROT_EXEC, MAP_PRIVATE, 3, 0) = 0x77dc6000
mmap2(0x77dd9000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0x3000) = 0x77dd9000
close(3)                                = 0
open("/lib/libuci.so", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fcntl64(3, F_SETFD, FD_CLOEXEC)         = 0
statx(3, "", AT_STATX_SYNC_AS_STAT|AT_EMPTY_PATH, STATX_BASIC_STATS, {stx_mask=STATX_BASIC_STATS|STATX_MNT_ID, stx_attributes=0, stx_mode=S_IFREG|0777, stx_size=29199, ...}) = 0
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0\10\0\1\0\0\0 \22\0\0004\0\0\0"..., 936) = 936
mmap2(NULL, 98304, PROT_READ|PROT_EXEC, MAP_PRIVATE, 3, 0) = 0x77dae000
mmap2(0x77dc4000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0x6000) = 0x77dc4000
close(3)                                = 0
open("/lib/libblobmsg_json.so.20220927", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fcntl64(3, F_SETFD, FD_CLOEXEC)         = 0
statx(3, "", AT_STATX_SYNC_AS_STAT|AT_EMPTY_PATH, STATX_BASIC_STATS, {stx_mask=STATX_BASIC_STATS|STATX_MNT_ID, stx_attributes=0, stx_mode=S_IFREG|0644, stx_size=8351, ...}) = 0
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0\10\0\1\0\0\0\260\t\0\0004\0\0\0"..., 936) = 936
mmap2(NULL, 77824, PROT_READ|PROT_EXEC, MAP_PRIVATE, 3, 0) = 0x77d9b000
mmap2(0x77dac000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0x1000) = 0x77dac000
close(3)                                = 0
open("/lib/libjson-c.so.5", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/usr/local/lib/libjson-c.so.5", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/usr/lib/libjson-c.so.5", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fcntl64(3, F_SETFD, FD_CLOEXEC)         = 0
statx(3, "", AT_STATX_SYNC_AS_STAT|AT_EMPTY_PATH, STATX_BASIC_STATS, {stx_mask=STATX_BASIC_STATS|STATX_MNT_ID, stx_attributes=0, stx_mode=S_IFREG|0644, stx_size=45847, ...}) = 0
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0\10\0\1\0\0\0`)\0\0004\0\0\0"..., 936) = 936
mmap2(NULL, 114688, PROT_READ|PROT_EXEC, MAP_PRIVATE, 3, 0) = 0x77d7f000
mmap2(0x77d99000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0xa000) = 0x77d99000
close(3)                                = 0
open("/lib/libgcc_s.so.1", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fcntl64(3, F_SETFD, FD_CLOEXEC)         = 0
statx(3, "", AT_STATX_SYNC_AS_STAT|AT_EMPTY_PATH, STATX_BASIC_STATS, {stx_mask=STATX_BASIC_STATS|STATX_MNT_ID, stx_attributes=0, stx_mode=S_IFREG|0644, stx_size=78099, ...}) = 0
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0\10\0\1\0\0\0\260'\0\0004\0\0\0"..., 936) = 936
mmap2(NULL, 147456, PROT_READ|PROT_EXEC, MAP_PRIVATE, 3, 0) = 0x77d5b000
mmap2(0x77d7d000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0x12000) = 0x77d7d000
close(3)                                = 0
mprotect(0x77df3000, 4096, PROT_READ)   = 0
mprotect(0x77dd9000, 4096, PROT_READ)   = 0
mprotect(0x77dc4000, 4096, PROT_READ)   = 0
mprotect(0x77dac000, 4096, PROT_READ)   = 0
mprotect(0x77d99000, 4096, PROT_READ)   = 0
mprotect(0x77d7d000, 4096, PROT_READ)   = 0
mprotect(0x5559a000, 4096, PROT_READ)   = 0
statx(AT_FDCWD, "/var/run/rpcd", AT_STATX_SYNC_AS_STAT, STATX_BASIC_STATS, {stx_mask=STATX_BASIC_STATS|STATX_MNT_ID, stx_attributes=0, stx_mode=S_IFDIR|0700, stx_size=60, ...}) = 0
umask(077)                              = 022
rt_sigaction(SIGPIPE, {sa_handler=SIG_IGN, sa_mask=[RT_68 RT_70 RT_72 RT_73 RT_80 RT_86 RT_87 RT_88 RT_89 RT_90 RT_91 RT_93 RT_94 RT_95], sa_flags=SA_RESTORER|SA_RESTART, sa_restorer=NULL}, {sa_handler=SIG_DFL, sa_mask=[RT_65 RT_67 RT_72 RT_74 RT_76 RT_79 RT_82 RT_83 RT_87 RT_88 RT_89 RT_90 RT_94 RT_95], sa_flags=0}, 16) = 0
rt_sigprocmask(SIG_UNBLOCK, [RT_1 RT_2], NULL, 16) = 0
rt_sigaction(SIGHUP, {sa_handler=0x555825e1, sa_mask=[RT_68 RT_70 RT_72 RT_73 RT_80 RT_86 RT_87 RT_88 RT_89 RT_90 RT_91 RT_93 RT_94 RT_95], sa_flags=SA_RESTORER|SA_RESTART, sa_restorer=NULL}, {sa_handler=SIG_DFL, sa_mask=[RT_65 RT_67 RT_72 RT_74 R [... cut ...]
getdents64(7, 0x77eaa268 /* 3 entries */, 2048) = 80
getdents64(7, 0x77eaa268 /* 0 entries */, 2048) = 0
close(7)                                = 0
munmap(0x77eaa000, 8192)                = 0
wait4(-1, 0x7f885adc, WNOHANG, NULL)    = -1 ECHILD (No child process)
epoll_pwait(3, [{events=EPOLLIN, data={u32=2011119632, u64=2011119632}}], 10, -1, NULL, 16) = 1
read(4, "w", 4)                         = 1
read(4, 0x7f885a84, 4)                  = -1 EAGAIN (Resource temporarily unavailable)
epoll_pwait(3,

jow-, 2023-01-17:
> Ignoring ucode script /usr/share/rpcd/ucode/luci because it is world writable
> Ignoring ucode script /usr/share/rpcd/ucode/luci.upnp because it is world writable

Seems your rootfs file permissions are broken. Try the following:
```
chmod 0644 /usr/share/rpcd/ucode/luci
chmod 0644 /usr/share/rpcd/ucode/luci.upnp
service rpcd restart
```

mis1042, 2023-01-17:
> Seems your rootfs file permissions are broken. Try the following:
> ```
> chmod 0644 /usr/share/rpcd/ucode/luci
> chmod 0644 /usr/share/rpcd/ucode/luci.upnp
> service rpcd restart
> ```

Fixed
But I didn't change the permission settings, this might be a new problem

jow-, 2023-01-17:
I does not to be a problem in the official builds. I assume it is an issue with your build environment, unusual umask settings or filesystem limitations.

mis1042, 2023-01-17:
> It does not appear to be a problem in the official builds. I assume it is an issue with your build environment, unusual umask settings or filesystem limitations.

Ok, maybe I need to test more, my compilation environment is GitHub CodeSpace

naluckstar, 2023-10-21:
chmod 0644 /usr/share/rpcd/ucode/luci
chmod 0644 /usr/share/rpcd/ucode/luci.upnp
service rpcd restart

The above can solve the problem, but I am using version 23.0.5 of openwrt and still have this issue with version 23.05.0. Is this question not included?

thezedt, 2024-03-03:
> The above can solve the problem, but I am using version 23.0.5 of openwrt and still have this issue with version 23.05.0. Is this question not included?

[image]

I've had the same `RPC call to luci/getFeatures failed with error -32000: Object not found` issue after building a slightly customized OpenWrt 23.05.2 from a work folder on which I messed with file/folder permissions (in an attempt to make them easily editable through a network share).

Running 
`chmod 0644 /usr/share/rpcd/ucode/luci`
`service rpcd restart`
on the device running the broken build as suggested above by **jow-** fixed the issue.

After [restarting from scratch](https://openwrt.org/docs/guide-developer/toolchain/use-buildsystem) and rebuilding without messing up with permissions/ownership, the generated image works perfectly.

systemcrash, 2024-05-02:
@marek22k OK but does the 'fix' resolve the problem for you? Or does it persist? It seems like something else is messing around with that file.

marek22k, 2024-05-02:
```
chmod 0644 /usr/share/rpcd/ucode/luci
service rpcd restart
```
Didn't fix the problem for me.

However, I have noticed that the problem only seems to occur when I open the graphs of collectd (`/cgi-bin/luci/admin/statistics/graphs`).

I have also restarted the router several times. That didn't help.

systemcrash, 2024-05-02:
Tell you what, just run:
```sh
opkg install -force-reinstall --force-overwrite luci-app-statistics
```
if that doesn't fix it, uninstall `luci-app-statistics` and install anew.
```sh
opkg remove luci-app-statistics
opkg install luci-app-statistics
```

marek22k, 2024-06-03:
Sorry for the late reply. Even though I haven't thought about the problem for some time, it unfortunately still exists.

> run:
> 
> ```shell
> opkg list-installed | grep -i luci-app-statistics
> ```

```
root@OpenWrt:~# opkg list-installed | grep -i luci-app-statistics
luci-app-statistics - git-24.034.36441-feee897
```

> Tell you what, just run:
> 
> ```shell
> opkg install -force-reinstall --force-overwrite luci-app-statistics
> ```

```
root@OpenWrt:~# opkg install -force-reinstall --force-overwrite luci-app-statist
ics
Removing package luci-app-statistics from root...
Not deleting modified conffile /etc/config/luci_statistics.
Installing luci-app-statistics (git-24.034.36441-feee897) to root...
Downloading https://downloads.openwrt.org/releases/23.05.3/packages/aarch64_cortex-a53/luci/luci-app-statistics_git-24.034.36441-feee897_all.ipk
Configuring luci-app-statistics.
Collected errors:
 * resolve_conffiles: Existing conffile /etc/config/luci_statistics is different from the conffile in the new package. The new conffile will be placed at /etc/config/luci_statistics-opkg.
```
Doen't fix it.

> if that doesn't fix it, uninstall `luci-app-statistics` and install anew.
> 
> ```shell
> opkg remove luci-app-statistics
> opkg install luci-app-statistics
> ```

```
root@OpenWrt:~# opkg remove luci-app-statistics
Removing package luci-app-statistics from root...
Not deleting modified conffile /etc/config/luci_statistics.
root@OpenWrt:~# opkg install luci-app-statistics
Installing luci-app-statistics (git-24.034.36441-feee897) to root...
Downloading https://downloads.openwrt.org/releases/23.05.3/packages/aarch64_cortex-a53/luci/luci-app-statistics_git-24.034.36441-feee897_all.ipk
Configuring luci-app-statistics.
Collected errors:
 * resolve_conffiles: Existing conffile /etc/config/luci_statistics is different from the conffile in the new package. The new conffile will be placed at /etc/config/luci_statistics-opkg.
```
Also doesn't fix the bug.

```
root@OpenWrt:~# cat /etc/config/luci_statistics

config statistics 'collectd'
	option BaseDir '/var/run/collectd'
	option PIDFile '/var/run/collectd.pid'
	option PluginDir '/usr/lib/collectd'
	option TypesDB '/usr/share/collectd/types.db'
	option Interval '30'
	option ReadThreads '2'
	option Include '/etc/collectd/conf.d/*.conf'

config statistics 'rrdtool'
	option default_timespan '2hour'
	option image_width '600'
	option image_height '150'
	option image_path '/tmp/rrdimg'

[... 200 lines cut ...]
config statistics 'collectd_splash_leases'
	option enable '0'

config statistics 'collectd_tcpconns'
	option enable '0'
	option ListeningPorts '0'
	list LocalPorts '22'
	option AllPortsSummary '0'

config statistics 'collectd_thermal'
	option enable '1'

config statistics 'collectd_uptime'
	option enable '0'

config collectd_exec_input
	option cmdline '/root/5ghz_temp.sh'
	option cmduser 'nobody'
	option cmdgroup 'nogroup'

config collectd_exec_input
	option cmdline '/root/24ghz_temp.sh'
	option cmduser 'nobody'
	option cmdgroup 'nogroup'
```

marek22k, 2024-06-15:
Mhh, I have now replaced the luci_statistics with the file from opkg. Then I have set up collectd via LuCi again. Now it seems to work.

Abicom-Dave, 2024-08-19:
On my proprietary 23.05 system, it appears to be a permissions problem. 

However, when I try to manually run rpcd, I get the following:

root@OpenWrt:/# rpcd
Ignoring ucode script /usr/share/rpcd/ucode/luci because it is world writable
Ignoring ucode script /usr/share/rpcd/ucode/luci.upnp because it is world writable

Changing the permissions to a:rx fixed the issue and I no longer get the RPC error

root@OpenWrt:/# chmod a-w /usr/share/rpcd/ucode/luci.upnp
root@OpenWrt:/# chmod a-w /usr/share/rpcd/ucode/luci

root@OpenWrt:/# rpcd &
root@OpenWrt:/# 

[image]

marek22k, 2025-01-12:
Mhh, was the issue resolved?

francismanjares, 2025-01-16:
I am also getting this, when someone tries to torrent the router chokes from being flooding by a lot of connections, the webui crashes. 

running _service rpcd restart_ brought it back.