openwrt/luci #6098: luci-app-lxc: "Action failed!" when configuring container

gclawes, 2022-11-15:
## Steps to reproduce:

1. go to: Services -> LXC Containers
2. Create a new container (I'm using "Debian Bullseye" from https://repo.turris.cz/lxc)
3. After container is created, select "more" drop-down for it and select "configure"
4. Change any configuration or leave config unchanged
5. Click "Confirm" 

## Actual behavior:

`Action failed!` response is given and the configuration is not saved.  Configuration window does not close.
[image]

Nothing appears in `/var/log/messages` when this happens.

## Expected behavior:
**Example: Replace the following lines and remove this**

Application is saved and configuration window closes

## Additional Information:
OpenWrt version information from system `/etc/openwrt_release`

```
DISTRIB_ID='TurrisOS'
DISTRIB_RELEASE='6.0.2'
DISTRIB_REVISION='r16699+129-1d12cfdce2'
DISTRIB_TARGET='mvebu/cortexa9'
DISTRIB_ARCH='arm_cortex-a9_vfpv3-d16'
DISTRIB_DESCRIPTION='TurrisOS 6.0.2 1d12cfdce28101ca60fc7afed1e4cf0812814992'
DISTRIB_TAINTS='busybox'
```

`/etc/config/lxc`:
```
config lxc 'lxc'
        option ssl_enabled '1'
        option min_space '100000'
        option min_temp '100000'
        option url 'repo.turris.cz/lxc'
```

Turris storage plugin is configured for `/srv` on an ssd (including `/srv/lxc/`):
`/etc/config/storage`:
```
config srv 'srv'
        option uuid 'ca298dde-abf3-49ca-bc77-f10ba5cc0784'
```

If additional information is required, let me know.

thomas-maurice, 2024-08-23:
I have the same issue, doing the same modification directly on the config file using vim or something works perfectly though

Vinke2, 2024-08-28:
No one cares about this plugin?

eugenio, 2024-10-14:
same problem here!