openwrt/luci #6101: luci-mod-network: Wireless interface name field not validated

crass, 2022-11-16:
## Steps to reproduce:

1. go to: Network → Wireless
2. Click edit button on existing wireless interface (not radio).
3. In modal dialog, go to: Advanced Settings tab in Interface Configuration section.
4. Set Interface name field to "iface678901234567890".
5. Click Save button.
6. Click Save & Apply.

## Actual behavior:
**Example: Replace the following lines and remove this**

1. Save is allowed with validation even though kernel only allows interface names of up to 15 characters.
2. All interfaces on the radio with the invalid interface name fail to come up.
3. Search in logread messages for "iface67890", there should be messages indicating that an interface name "iface6789012345" is being used.
4. There are log messages indicating that the truncated interface name fails to be created.

Example:
```
Tue Nov 15 06:45:25 2022 daemon.err hostapd: nl80211: kernel reports: Attribute failed policy validation
Tue Nov 15 06:45:25 2022 daemon.err hostapd: Failed to create interface iface67890123456: -22 (Invalid argument)
Tue Nov 15 06:45:25 2022 daemon.err hostapd: Failed to add BSS (BSSID=00:23:5c:fc:4d:c0)
Tue Nov 15 06:45:25 2022 daemon.err hostapd: Interface initialization failed
```

## Expected behavior:

LuCI should validate this field and only allow a maximum of 15 characters to be entered.

## Additional Information:
The maximum length of an interface name is 15 characters.
```
root@OpenWRT:~# iw phy phy0 interface add iface67890123456 type monitor
kernel reports: Attribute failed policy validation
command failed: Invalid argument (-22)
root@OpenWRT:~# iw phy phy0 interface add iface6789012345 type monitor
root@OpenWRT:~# iw dev iface6789012345 del
root@OpenWRT:~# 
```

OpenWrt version information from system `/etc/openwrt_release`

```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='22.03.2'
DISTRIB_REVISION='r19803-9a599fee93'
DISTRIB_TARGET='mvebu/cortexa9'
DISTRIB_ARCH='arm_cortex-a9_vfpv3-d16'
DISTRIB_DESCRIPTION='OpenWrt 22.03.2 r19803-9a599fee93'
DISTRIB_TAINTS=''
```

hnyman, 2023-01-14:
Should be fixed in master with 0c1be9ae5
(possible backport to 22.03 later)

closed 2023-01-14 by hnyman

jow-, 2023-01-14:
@hnyman - that fix will prevent ifnames with dashes which are perfectly legal.

hnyman, 2023-01-15:
Better validation?  Just length?

hnyman, 2023-01-15:
I removed the uciname validation and left just the maxlength.

jow-, 2023-01-16:
I wonder about what kind of chars are allowed exactly in linux netdev names from time to time, so I digged it up again:

https://elixir.bootlin.com/linux/latest/source/net/core/dev.c#L995

So the rules are:

 - It must not be an empty string
 - It must not be `.`
 - It must not be `..`
 - It must not contain any `/`, `:` or space character (` `, `\t`, `\n`, ...)
 - It must be less than 16 chars
 - It likely must not contain any `%` either

Expressed in JavaScript, the validator likely should be:

```javascript
o.validate = function(section_id, value) {
    return (/^[^:/%\s]{1,15}$/.exec(value) && value != '.' && value != '..') ? true : _('Must be a valid network device name');
};
```

hnyman, 2023-01-17:
Too complex for me ;-)

Will you define that? We might actually benefit of having that validation in general. Might be useful for other net device names, too.

Ps. I must admit that I am so old-school (started with Apple ][ in the 80s) that I am always scared about pretty much anything outside 7-bit ASCII letters. Too often burned with Scandinavian letter troubles and MS-DOS 8.3 variations of the early days ;-)

crass, 2023-03-02:
I don't think this is currently fixed since the validation described by @jow- isn't implemented, just the length is checked. Having less strict validation is more problematic (for users that use names that shouldn't validate) than a little overly strict as noted in the issue description. I've created pull request #6270 to address this issue.