openwrt/luci #6105: luci-base: LuCI is not detecting properly sysauth.htm and uses default sysauth.ut in 3rd party theme.

okibcn, 2022-11-19:
## Steps to reproduce:
1. On a fresh snapshot, install 3rd party [luci-theme-argon](https://github.com/jerrykuku/luci-theme-argon) package: downloading the packages from that repo or using CLI: 
```sh
opkg update
opkg install luci-compat
opkg install luci-lib-ipkg
wget --no-check-certificate https://github.com/jerrykuku/luci-theme-argon/releases/download/v2.2.9.4/luci-theme-argon-master_2.2.9.4_all.ipk
opkg install luci-theme-argon*.ipk
wget --no-check-certificate https://github.com/jerrykuku/luci-theme-argon/releases/download/v2.2.9/luci-app-argon-config_0.9-20210309_all.ipk
opkg install luci-app-argon-config*.ipk
```

2. In OpenWrt, go to System > System > Language and Style and select the Argon design.
3. Logout OpenWrt to return to the login page

## Actual behavior:
For some reason, LuCI is executing `/usr/share/ucode/luci/template/sysauth.ut` instead of `/usr/lib/lua/luci/view/themes/argon/sysauth.htm`. However, it renders properly the htm headers and the footers provided in the theme, so it seems that the issue only affects the sysauth detection.

[image]

## Expected behavior:
LuCI should render the provided sysauth.htm file  `/usr/lib/lua/luci/view/themes/argon/sysauth.htm`. This is the current behavior in the official release (only fails in the snapshot):

[image]

## Additional Information:
OpenWrt version information from system `/etc/openwrt_release`

```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='SNAPSHOT'
DISTRIB_REVISION='r21314-1996d600bd'
DISTRIB_TARGET='mediatek/mt7622'
DISTRIB_ARCH='aarch64_cortex-a53'
DISTRIB_DESCRIPTION='OpenWrt SNAPSHOT r21314-1996d600bd'
DISTRIB_TAINTS='no-all busybox'
```

jow-, 2022-11-20:
This actually is a kind of bug in the Argon theme itself. The shipped `sysauth.htm` is rendered but throws a runtime exception, which causes LuCI to fall back to it's builtin `sysauth.htm`.

The issue is within this line:

https://github.com/jerrykuku/luci-theme-argon/blob/master/luasrc/view/themes/argon/sysauth.htm#L132

On 22.03, the global `node` variable is not defined which causes the `( (node and node.title) and ' - ' .. translate(node.title) or ''))` part to be skipped. On master, `node` is defined and referring to the dispatcher's `node()` function, leading to a Lua `attempt to index a function value` exception due to the `node.title` expression. After removing the extraneous code which has never been rendered anyway, the theme appears to function normally.

okibcn, 2022-11-20:
Thank you @jow- for tracking the problem. The same code in the theme was working fine until 22.03. I think that now it junps by default upon any error that previously was rendered the "best possible way".

Thanks

jow-, 2022-11-20:
No, the behavior is the same. 22.03 just happens to not trigger the bug/undefined variable access in Argon.

I will push a fix for this later

closed 2022-11-21 by jow-

okibcn, 2022-11-22:
@jow- 

I flashed the new snapshot build r21335-e410833bdd, using the same Argon theme. Now, the login page loads properly, but it can't access the rest of the site. it returns the following error:

[image]

```
Runtime error
Unhandled exception during request dispatching
/usr/lib/lua/luci/ucodebridge.lua:23: /usr/lib/lua/luci/template.lua:158: Failed to load template 'admin_status/index'.
Error while parsing template '/usr/lib/lua/luci/view/admin_status/index.htm':
No such file or directory

In error(), file [C]
called from function [anonymous function] (/usr/lib/lua/luci/ucodebridge.lua:23)
called from function ((tail call))
In [anonymous function](), file /usr/share/ucode/luci/runtime.uc, line 138, byte 45:
  called from function [arrow function] (/usr/share/ucode/luci/dispatcher.uc:778:70)
  called from function render ([C])
  called from function render_action (/usr/share/ucode/luci/dispatcher.uc:768:24)
  called from function run_action (/usr/share/ucode/luci/dispatcher.uc:779:4)
  called from function [anonymous function] (/usr/share/ucode/luci/dispatcher.uc:993:48)
  called from anonymous function (/www/cgi-bin/luci:39:12)

 `        return lcall.call(modname, method, ...args);`
  Near here ----------------------------------------^
```

Should I report it as a new issue or just continue here?