openwrt/luci #6110: Luci login screen shows menu entries when using unofficial packages.

hudra0, 2022-11-20:
## Steps to reproduce:

1. Build custom image with unofficial luci-app-diskman package.
2. flash image
3. login screen looks like this
[image]

## Expected behavior:

```
There should be no menu visible. This was working until a few builds. Probably until the end of october.
```

## Additional Information:
OpenWrt version information from system `/etc/openwrt_release`

```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='SNAPSHOT'
DISTRIB_REVISION='r21326-bca663f20f'
DISTRIB_TARGET='x86/64'
DISTRIB_ARCH='x86_64'
DISTRIB_DESCRIPTION='OpenWrt SNAPSHOT r21326-bca663f20f'
DISTRIB_TAINTS='no-all override'
```

jow-, 2022-11-20:
Looks like a duplicate of / caused by https://github.com/openwrt/luci/issues/6105

closed 2022-11-20 by jow-

hudra0, 2022-11-20:
Thanks for looking into it. Just for your information: this issue wasn't argon theme specific.. also happens with other themes.