openwrt/luci #6111: LuCI index page gives error after login

hnyman, 2022-11-21:
Up-to-date OpenWrt master r21334-9b482ee22f & LuCI (7c3705b)

Login screen comes ok, but after login there is an error and no LuCI:
```
/usr/lib/lua/luci/ucodebridge.lua:23: /usr/lib/lua/luci/template.lua:158: Failed to load template 'admin_status/index'.
Error while parsing template '/usr/lib/lua/luci/view/admin_status/index.htm':
No such file or directory
```

Looking at the directory, there is luaindex.htm instead of index.htm
```
root@router1:~# ls /usr/lib/lua/luci/view/admin_status/
luaindex.htm
```

This should be a vanilla openssl based LuCI, bootstrap theme. (But also the openwrt-2020 theme is installed)

Full error shown:
```
Runtime error
Unhandled exception during request dispatching
/usr/lib/lua/luci/ucodebridge.lua:23: /usr/lib/lua/luci/template.lua:158: Failed to load template 'admin_status/index'.
Error while parsing template '/usr/lib/lua/luci/view/admin_status/index.htm':
No such file or directory

In error(), file [C]
  called from function [anonymous function] (/usr/lib/lua/luci/ucodebridge.lua:23)
  called from function  ((tail call))

In [anonymous function](), file /usr/share/ucode/luci/runtime.uc, line 138, byte 45:
  called from function [arrow function] (/usr/share/ucode/luci/dispatcher.uc:778:70)
  called from function render ([C])
  called from function render_action (/usr/share/ucode/luci/dispatcher.uc:768:24)
  called from function run_action (/usr/share/ucode/luci/dispatcher.uc:779:4)
  called from function [anonymous function] (/usr/share/ucode/luci/dispatcher.uc:993:48)
  called from anonymous function (/www/cgi-bin/luci:39:12)

 `        return lcall.call(modname, method, ...args);`
  Near here ----------------------------------------^
```

okibcn, 2022-11-22:
It also fails with 3rd party Argon theme.

phoenixxie0, 2022-11-22:
The same problem.

openips, 2022-11-22:
The same problem with the material theme

ldir-EDB0, 2022-11-22:
Unfortunately I'm getting the following after the fix:

Runtime error

Unhandled exception during request dispatching
Unable to open source file admin_status/index: No such file or directory
In [anonymous function](), file /usr/share/ucode/luci/runtime.uc, line 69, byte 52:
  called from function run_action (/usr/share/ucode/luci/dispatcher.uc:778:36)
  called from function [anonymous function] (/usr/share/ucode/luci/dispatcher.uc:996:48)
  called from anonymous function (/www/cgi-bin/luci:39:12)

 `        let tmplfunc = loadfile(path, { raw_mode: false });`
  Near here -----------------------------------------------^

ldir-EDB0, 2022-11-22:
And custom commands gives:

Runtime error

Unhandled exception during request dispatching
Unable to open source file commands: No such file or directory
In [anonymous function](), file /usr/share/ucode/luci/runtime.uc, line 69, byte 52:
  called from function run_action (/usr/share/ucode/luci/dispatcher.uc:778:36)
  called from function [anonymous function] (/usr/share/ucode/luci/dispatcher.uc:996:48)
  called from anonymous function (/www/cgi-bin/luci:39:12)

 `        let tmplfunc = loadfile(path, { raw_mode: false });`
  Near here -----------------------------------------------^

northbound1, 2022-11-22:
After fix still can't update software lists in luci. And the above error on overview.

snakwu, 2022-11-22:
@jow- http://192.168.1.1/cgi-bin/luci/admin/system/flash,also have problems

trippleflux, 2022-11-22:
Also cannot input / edit OpenWrt password

jow-, 2022-11-22:
> After fix still can't update software lists in luci. And the above error on overview.

Please show the error

> @jow- http://192.168.1.1/cgi-bin/luci/admin/system/flash,also have problems

Please show the problem

> Also cannot input / edit OpenWrt password

Where?

Please be more specific. I can't act on "something is broken" reports.

jow-, 2022-11-22:
Thanks! Fix pushed.

Neifab, 2022-11-23:
Hello, same problem here, how to reverse or resolve it ?

Neifab, 2022-11-24:
Hello, just installed the new snapshot and the problem is still there, is the fix already pushed on the last one ? Thanks

AmirulAndalib, 2022-11-24:
> Hello, just installed the new snapshot and the problem is still there, is the fix already pushed on the last one ? Thanks

Hi there

Snapshots are set to build on a specific cronjob time so you might wanna wait for few more hours for the latest snapshot build to be available for upgrading.....

I guess they build/refresh snapshots every 48 hours or something

commit 9d10c2e6 (2022-11-22): luci-base: fix (again) rendering ucode templates in `template` target

Fixes: #6111
Fixes: a5d21dadbd ("luci-base: fix rendering ucode templates from `template` target")
commit d8e3b35c (2022-11-22): luci-base: fix passing dispatched node details to templates

Fixes: #6111
Fixes: 08ae8d40a9 ("luci-base, luci-lua-runtime: fix "dispatched" and "requested" properties")