openwrt/luci #6112: Luci Status overview

northbound1, 2022-11-21:
***

## Steps to reproduce:
Open Luci click status overview
```

## Additional Information:
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='SNAPSHOT'
DISTRIB_REVISION='r21334-9b482ee22f'
DISTRIB_TARGET='mvebu/cortexa9'
DISTRIB_ARCH='arm_cortex-a9_vfpv3-d16'
DISTRIB_DESCRIPTION='OpenWrt SNAPSHOT r21334-9b482ee22f'
DISTRIB_TAINTS='no-all'

Error:

Runtime error
Unhandled exception during request dispatching
/usr/lib/lua/luci/ucodebridge.lua:23: /usr/lib/lua/luci/template.lua:158: Failed to load template 'admin_status/index'.
Error while parsing template '/usr/lib/lua/luci/view/admin_status/index.htm':
No such file or directory

In error(), file [C]
called from function [anonymous function] (/usr/lib/lua/luci/ucodebridge.lua:23)
called from function ((tail call))
In [anonymous function](), file /usr/share/ucode/luci/runtime.uc, line 138, byte 45:
  called from function [arrow function] (/usr/share/ucode/luci/dispatcher.uc:778:70)
  called from function render ([C])
  called from function render_action (/usr/share/ucode/luci/dispatcher.uc:768:24)
  called from function run_action (/usr/share/ucode/luci/dispatcher.uc:779:4)
  called from function [anonymous function] (/usr/share/ucode/luci/dispatcher.uc:993:48)
  called from anonymous function (/www/cgi-bin/luci:39:12)

 `        return lcall.call(modname, method, ...args);`
  Near here ----------------------------------------^

Error 2 OPKG also broke

hnyman, 2022-11-21:
Duplicate of #6111

closed 2022-11-21 by hnyman

phoenixxie0, 2022-11-22:
The same problem.