openwrt/luci #6113: make nconfig: selecting luci from the config fails to bring in lua dependencies

graysky2, 2022-11-22:
# Issue template:
## Steps to reproduce:
1. Pull https://github.com/openwrt/openwrt/pull/11115
2. Run `git revert 9b482ee22f3f509880d2420652392971eef3211d` (this breaks the AX6000 currently which is the device I am using)
3. Start with the following `.config`
```
CONFIG_TARGET_mediatek=y
CONFIG_TARGET_mediatek_filogic=y
CONFIG_TARGET_mediatek_filogic_DEVICE_xiaomi_redmi-router-ax6000-stock=y
```
4. run `make nconfig`
5. Go to LuCI --> Collection and enable luci (`<*> luci................... LuCI interface with Uhttpd as Webserver (default)`)
6. Exit and save the config
7. Run `./scripts/diffconfig.sh > diffconfig`

## Actual behavior:

None of the lua dependencies are added to this `diffconfig` which results in a broken luci.

## Expected behavior:
The following seem missing
```
CONFIG_PACKAGE_liblucihttp-lua=y
CONFIG_PACKAGE_luci-lib-base=y
CONFIG_PACKAGE_luci-lib-ip=y 
CONFIG_PACKAGE_luci-lib-jsonc=y
CONFIG_PACKAGE_luci-lib-nixio=y
```

## Additional Information:
OpenWrt version information from system `/etc/openwrt_release`

```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='SNAPSHOT'
DISTRIB_REVISION='r0+21340-e844ad753f'
DISTRIB_TARGET='mediatek/filogic'
DISTRIB_ARCH='aarch64_cortex-a53'
DISTRIB_DESCRIPTION='OpenWrt SNAPSHOT r0+21340-e844ad753f'
DISTRIB_TAINTS='no-all'
```

jow-, 2022-11-22:
Current LuCI does not require Lua for the basic package set. Whatever is broken in your case is either a non-official Lua package lacking a `luci-compat` or `luci-lua-runtime` dependency or you encountered a recently fixed error such as #6111

graysky2, 2022-11-22:
@jow- Is this the expected diffconfig for just:
1.  Those 3 lines of target info
2. Running `make nconfig` and selecting that luci metapackage
3. Running `./scripts/diffconfig.sh`
```
CONFIG_TARGET_mediatek=y
CONFIG_TARGET_mediatek_filogic=y
CONFIG_TARGET_mediatek_filogic_DEVICE_xiaomi_redmi-router-ax6000-stock=y
CONFIG_PACKAGE_attendedsysupgrade-common=y
CONFIG_PACKAGE_cgi-io=y
CONFIG_PACKAGE_liblucihttp=y
CONFIG_PACKAGE_liblucihttp-ucode=y
CONFIG_PACKAGE_luci=y
CONFIG_PACKAGE_luci-app-attendedsysupgrade=y
CONFIG_PACKAGE_luci-app-firewall=y
CONFIG_PACKAGE_luci-app-opkg=y
CONFIG_PACKAGE_luci-base=y
CONFIG_PACKAGE_luci-mod-admin-full=y
CONFIG_PACKAGE_luci-mod-network=y
CONFIG_PACKAGE_luci-mod-status=y
CONFIG_PACKAGE_luci-mod-system=y
CONFIG_PACKAGE_luci-proto-ipv6=y
CONFIG_PACKAGE_luci-proto-ppp=y
CONFIG_PACKAGE_luci-theme-bootstrap=y
CONFIG_PACKAGE_rpcd=y
CONFIG_PACKAGE_rpcd-mod-file=y
CONFIG_PACKAGE_rpcd-mod-iwinfo=y
CONFIG_PACKAGE_rpcd-mod-luci=y
CONFIG_PACKAGE_rpcd-mod-rpcsys=y
CONFIG_PACKAGE_rpcd-mod-rrdns=y
CONFIG_PACKAGE_rpcd-mod-ucode=y
CONFIG_PACKAGE_ucode-mod-html=y
CONFIG_PACKAGE_ucode-mod-math=y
CONFIG_PACKAGE_uhttpd=y
CONFIG_PACKAGE_uhttpd-mod-ubus=y
```

jow-, 2022-11-22:
Yes, looks right