openwrt/luci #6114: luci: /usr/lib/lua/luci/ucodebridge.lua:23: /usr/lib/lua/luci/template.lua:158: Failed to load template 'admin_status/index

mikeporterinmd, 2022-11-23:
## Steps to reproduce:

Login to luci works, but as soon as I am logged in:

Unhandled exception during request dispatching
/usr/lib/lua/luci/ucodebridge.lua:23: /usr/lib/lua/luci/template.lua:158: Failed to load template 'admin_status/index'.
Error while parsing template '/usr/lib/lua/luci/view/admin_status/index.htm':
No such file or directory

In error(), file [C]
called from function [anonymous function] (/usr/lib/lua/luci/ucodebridge.lua:23)
called from function ((tail call))
In [anonymous function](), file /usr/share/ucode/luci/runtime.uc, line 138, byte 45:
  called from function [arrow function] (/usr/share/ucode/luci/dispatcher.uc:778:70)
  called from function render ([C])
  called from function render_action (/usr/share/ucode/luci/dispatcher.uc:768:24)
  called from function run_action (/usr/share/ucode/luci/dispatcher.uc:779:4)
  called from function [anonymous function] (/usr/share/ucode/luci/dispatcher.uc:993:48)
  called from anonymous function (/www/cgi-bin/luci:39:12)

 `        return lcall.call(modname, method, ...args);`
  Near here ----------------------------------------^

## Actual behavior:
Runtime error

## Expected behavior:
Main luci page

## Additional Information:
OpenWrt version information from system `/etc/openwrt_release`

```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='SNAPSHOT'
DISTRIB_REVISION='r21335-e410833bdd'
DISTRIB_TARGET='ipq40xx/generic'
DISTRIB_ARCH='arm_cortex-a7_neon-vfpv4'
DISTRIB_DESCRIPTION='OpenWrt SNAPSHOT r21335-e410833bdd'
DISTRIB_TAINTS=''
```

Installed packages (no errors during the installs):

ath10k-board-qca4019 - 20221109-4
ath10k-firmware-qca4019-ct - 2020-11-08-1
attendedsysupgrade-common - 8
base-files - 1507-r21335-e410833bdd
busybox - 1.35.0-5
ca-bundle - 20211016-1
cgi-io - 2022-08-10-901b0f04-21
collectd - 5.12.0-41
collectd-mod-cpu - 5.12.0-41
collectd-mod-interface - 5.12.0-41
collectd-mod-iwinfo - 5.12.0-41
collectd-mod-load - 5.12.0-41
collectd-mod-memory - 5.12.0-41
collectd-mod-network - 5.12.0-41
collectd-mod-rrdtool - 5.12.0-41
dnsmasq - 2.87-5
dropbear - 2022.82-2
firewall4 - 2022-10-18-7ae5e14b-1
fstools - 2022-11-10-3affe9ea-1
fwtool - 2019-11-12-8f7fe925-1
getrandom - 2022-08-13-4c7b720b-1
hostapd-common - 2022-07-29-b704dc72-6.1
iw - 5.19-1
iwinfo - 2022-11-01-00aab871-1
jansson4 - 2.14-3
jshn - 2022-09-27-ea560134-1
jsonfilter - 2018-02-04-c7e938d6-1
kernel - 5.15.79-1-781b85eccaea38764e0cdb6d74669a8d
kmod-ath - 5.15.79+5.15.74-1-1
kmod-ath10k-ct - 5.15.79+2022-05-13-f808496f-1
kmod-cfg80211 - 5.15.79+5.15.74-1-1
kmod-crypto-acompress - 5.15.79-1
kmod-crypto-aead - 5.15.79-1
kmod-crypto-ccm - 5.15.79-1
kmod-crypto-cmac - 5.15.79-1
kmod-crypto-crc32c - 5.15.79-1
kmod-crypto-ctr - 5.15.79-1
kmod-crypto-gcm - 5.15.79-1
kmod-crypto-gf128 - 5.15.79-1
kmod-crypto-ghash - 5.15.79-1
kmod-crypto-hash - 5.15.79-1
kmod-crypto-hmac - 5.15.79-1
kmod-crypto-manager - 5.15.79-1
kmod-crypto-null - 5.15.79-1
kmod-crypto-rng - 5.15.79-1
kmod-crypto-seqiv - 5.15.79-1
kmod-crypto-sha256 - 5.15.79-1
kmod-gpio-button-hotplug - 5.15.79-3
kmod-hwmon-core - 5.15.79-1
kmod-leds-gpio - 5.15.79-1
kmod-lib-crc-ccitt - 5.15.79-1
kmod-lib-crc32c - 5.15.79-1
kmod-lib-lzo - 5.15.79-1
kmod-mac80211 - 5.15.79+5.15.74-1-1
kmod-nf-conntrack - 5.15.79-1
kmod-nf-conntrack6 - 5.15.79-1
kmod-nf-flow - 5.15.79-1
kmod-nf-log - 5.15.79-1
kmod-nf-log6 - 5.15.79-1
kmod-nf-nat - 5.15.79-1
kmod-nf-reject - 5.15.79-1
kmod-nf-reject6 - 5.15.79-1
kmod-nfnetlink - 5.15.79-1
kmod-nft-core - 5.15.79-1
kmod-nft-fib - 5.15.79-1
kmod-nft-nat - 5.15.79-1
kmod-nft-offload - 5.15.79-1
kmod-nls-base - 5.15.79-1
kmod-ppp - 5.15.79-1
kmod-pppoe - 5.15.79-1
kmod-pppox - 5.15.79-1
kmod-slhc - 5.15.79-1
kmod-usb-core - 5.15.79-1
kmod-usb-dwc3 - 5.15.79-1
kmod-usb-dwc3-qcom - 5.15.79-1
kmod-usb-xhci-hcd - 5.15.79-1
kmod-usb3 - 5.15.79-1
libblobmsg-json20220927 - 2022-09-27-ea560134-1
libc - 1.2.3-4
libgcc1 - 11.3.0-4
libiwinfo-data - 2022-11-01-00aab871-1
libiwinfo20210430 - 2022-11-01-00aab871-1
libjson-c5 - 0.16-2
libjson-script20220927 - 2022-09-27-ea560134-1
libltdl7 - 2.4.7-1
liblua5.1.5 - 5.1.5-10
liblucihttp-lua - 2022-07-08-6e68a106-1
liblucihttp-ucode - 2022-07-08-6e68a106-1
liblucihttp0 - 2022-07-08-6e68a106-1
libmnl0 - 1.0.5-1
libnftnl11 - 1.2.4-1
libnl-tiny2022-11-01 - 2022-11-01-db3b2cdb-1
libopenssl1.1 - 1.1.1s-1
libpthread - 1.2.3-4
librrd1 - 1.0.50-5
libubox20220927 - 2022-09-27-ea560134-1
libubus-lua - 2022-06-01-2bebf93c-1
libubus20220601 - 2022-06-01-2bebf93c-1
libuci-lua - 2021-10-22-f84f49f0-6
libuci20130104 - 2021-10-22-f84f49f0-6
libuclient20201210 - 2021-05-14-6a6011df-1
libucode20220812 - 2022-10-18-00af0650-1
libustream-wolfssl20201210 - 2022-01-16-868fd881-2
libwolfssl5.5.1.e624513f - 5.5.1-stable-8
logd - 2022-08-13-4c7b720b-1
lua - 5.1.5-10
luci - git-22.297.83017-0143ef2
luci-app-attendedsysupgrade - git-22.285.67526-18bfcca
luci-app-firewall - git-22.089.67453-0eb3aeb
luci-app-opkg - git-22.273.28779-84ba6a5
luci-app-statistics - git-22.297.83017-984a9a4
luci-app-uhttpd - git-22.029.54222-e3fbfe9
luci-base - git-22.325.41248-fa17c15
luci-compat - git-22.297.83017-673f382
luci-lib-base - git-22.308.54612-9118452
luci-lib-ip - git-20.250.76529-62505bd
luci-lib-jsonc - git-22.079.54693-45f411b
luci-lib-nixio - git-22.222.71555-88b9088
luci-lua-runtime - git-22.325.41248-fa17c15
luci-mod-admin-full - git-19.253.48496-3f93650
luci-mod-network - git-22.325.41910-7c3705b
luci-mod-status - git-22.297.83017-97da0ba
luci-mod-system - git-22.297.83017-f478fe6
luci-proto-ipv6 - git-21.148.48881-79947af
luci-proto-ppp - git-21.158.38888-88b9d84
luci-ssl - git-20.244.36115-e10f954
luci-theme-bootstrap - git-22.297.83017-2e3282e
mesh11sd - 1.2.0-3
mtd - 26
netifd - 2022-08-25-76d2d41b-1
nftables-json - 1.0.5-2
odhcp6c - 2022-08-05-7d21e8d8-19
odhcpd-ipv6only - 2022-10-31-a92c0a73-1
openwrt-keyring - 2022-03-25-62471e69-2
opkg - 2022-02-24-d038e5b6-1
ppp - 2.4.9.git-2021-01-04-4
ppp-mod-pppoe - 2.4.9.git-2021-01-04-4
procd - 2022-11-10-039b88f7-2
procd-seccomp - 2022-11-10-039b88f7-2
procd-ujail - 2022-11-10-039b88f7-2
px5g-wolfssl - 6.1
rpcd - 2022-09-21-8c852b65-1
rpcd-mod-file - 2022-09-21-8c852b65-1
rpcd-mod-iwinfo - 2022-09-21-8c852b65-1
rpcd-mod-luci - 20210614
rpcd-mod-rpcsys - 2022-09-21-8c852b65-1
rpcd-mod-rrdns - 20170710
rpcd-mod-ucode - 2022-09-21-8c852b65-1
rrdtool1 - 1.0.50-5
ubi-utils - 2.1.5-1
uboot-envtools - 2022.10-9
ubox - 2022-08-13-4c7b720b-1
ubus - 2022-06-01-2bebf93c-1
ubusd - 2022-06-01-2bebf93c-1
uci - 2021-10-22-f84f49f0-6
uclient-fetch - 2021-05-14-6a6011df-1
ucode - 2022-10-18-00af0650-1
ucode-mod-fs - 2022-10-18-00af0650-1
ucode-mod-html - 1
ucode-mod-lua - 1
ucode-mod-math - 2022-10-18-00af0650-1
ucode-mod-ubus - 2022-10-18-00af0650-1
ucode-mod-uci - 2022-10-18-00af0650-1
uhttpd - 2022-08-12-e3395cd9-3
uhttpd-mod-ubus - 2022-08-12-e3395cd9-3
urandom-seed - 3
urngd - 2020-01-21-c7f7b6b6-1
usign - 2020-05-23-f1f65026-1
wireless-regdb - 2022.08.12-1
wpad-mesh-wolfssl - 2022-07-29-b704dc72-6.1
zlib - 1.2.13-1

hnyman, 2022-11-23:
Already known and fixed, duplicate of #6111 and #6112

luci-lua-runtime - git-22.325.41248-fa17c15
Broken version.

Use opkg to upgrade luci-base and luci-lua-runtime

closed 2022-11-23 by jow-

mikeporterinmd, 2022-11-24:
Waiting for the builds to complete inspired me to learn how to build openwrt.  Even on macOS, the very well written instructions made it a snap.  I was able to build the whole thing and then install the two packages I needed.  Thanks for the pointers.  And to those who took the time to document building on macOS.  A Mac Studio is a good build machine.  Also, luci worked fine with the new packages.  -- Mike